armbian/build #4842: Unable to build due to uninstallable package 'python3-rich'

mikedolx, 2023-02-21:
### What happened?

The CI build failed during the build. As it seems, the package python-rich cannot be installed as the required dependency `python3-markdown-it` is not installable.

### How to reproduce?

I am building with the following versions:

U-Boot Version: v2022.07
Kernel version: v5.15.82
Build branch: v22.11

The build branch is used by the CI to checkout the armbian-build repo with

```shell
git checkout v22.11
```

My `config-default.conf` is as follows:

```bash
# Read build script documentation https://docs.armbian.com/Developer-Guide_Build-Options/
# for detailed explanation of these options and for additional options not listed here

# leave empty to select each time, set to "yes" or "no">
BUILD_ONLY=""
# leave empty to select each time, set to "yes" or "no" to skip>
KERNEL_CONFIGURE="no"

# comma-separated list of clean targets: "make" = make clean for select>
# "debs" = delete packages in "./output>
# "alldebs" = delete all packages in ".>
# "cache" = delete "./output/cache", "s>
# "oldcache" = remove old cached rootfs>
CLEAN_LEVEL="make,debs,cache"   

# yes: use kernel config file from previous compilation for the same branch, device family and version
# no: use default or user-provided config file
KERNEL_KEEP_CONFIG="yes"

# comma-separated list of core modules which will be installed >
# "u-boot", "kernel", "bsp", "armbian-config", "armbian-firmwar>
# leave empty to build from sources or use local cache
REPOSITORY_INSTALL=""                   

BUILD_DESKTOP="yes"
DEST_LANG="de_DE.UTF-8"

# advanced
# compile and install or install prebuilt additional packages
EXTERNAL_NEW="prebuilt"

# change to "branchname" to use any branch currently available.
LIB_TAG="master"

# additions to compile 'unattended'
DESKTOP_ENVIRONMENT="xfce"
DESKTOP_APPGROUPS_SELECTED="editors, internet"
DESKTOP_ENVIRONMENT_CONFIG_NAME="config_base"

# additions to compile for the rock-pi4
BOARD="rockpi-4b"
BRANCH="current"
RELEASE="bullseye"

# yes: will stop the script and prompt for adding userpatches (kernel, u-boot)
CREATE_PATCHES="no"

# creates logs in output/debug
PROGRESS_LOG_TO_FILE="yes"

# set to yes, to prevent cloning from git
OFFLINE_WORK="no"
# provides a faster download
USE_MAINLINE_GOOGLE_MIRROR="yes"
USE_GITHUB_UBOOT_MIRROR="yes"
# create compressed archive with image file and GPG signature f>
# yes - compatibility shorcut for sha,gpg,7z
COMPRESS_OUTPUTIMAGE="yes"
```

My `lib.config` is fed by the CI. The resolved file would look like this:

```bash
BOOTBRANCH="tag:v2022.07"
KERNELBRANCH="tag:v5.15.82"
```

### Branch

other

### On which host OS are you observing this problem?

Bullseye

### Relevant log output

```shell
tail outpu/debug/compilation.log

  CLEAN   drivers/scsi
  CLEAN   drivers/tty/vt
  CLEAN   drivers/video/logo
  CLEAN   vmlinux.symvers modules-only.symvers modules.builtin modules.builtin.modinfo
patching file tools/include/tools/be_byteshift.h
patching file tools/include/tools/le_byteshift.h
dpkg-deb: building package 'linux-headers-current-rockchip64' in '../linux-headers-current-rockchip64_22.11.2_arm64.deb'.
 dpkg-genbuildinfo --build=binary -O../linux-5.15.82-rockchip64_22.11.2_arm64.buildinfo
 dpkg-genchanges --build=binary -O../linux-5.15.82-rockchip64_22.11.2_arm64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build .
dpkg-buildpackage: info: binary-only upload (no source included)

---

CI / compile.sh output:

The following information may help to resolve the situation:
The following packages have unmet dependencies:
 python3-rich : Depends: python3-markdown-it but it is not installable
E: Unable to correct problems, you have held broken packages.
[ error ] ERROR in function create_rootfs_cache [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:285 -> functions/main/build-tasks.sh:208 -> functions/main/rootfs-image.sh:66 -> functions/rootfs/create-cache.sh:201 -> functions/logging/traps.sh:0 ]
[ error ] Installation of Armbian main packages for current rockpi-4b bullseye editors, internet xfce no failed 
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:285 -> functions/main/build-tasks.sh:208 -> functions/main/rootfs-image.sh:66 -> functions/rootfs/create-cache.sh:201 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 ]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-02-21:
This problem is produced by Debian and we are mitigating it by providing recent enough dependencies:
https://imola.armbian.com/apt/pool/main/m/

Normally you shouldn't run into this problem. Probably it was due to the network issues / our repository was unreachable and you hit Debian only ... which is broken.

mikedolx, 2023-02-21:
Oh yes, you're right (see attached logs). Just wondering, how that could be possible? The thing, that nags me is, that its seemingly reproducible. But from the current location i am, where also our GitLab runner is stationed, i can access the repos fine. 

OK, maybe it was a temporarily problem. I see, that the last CI job ran 4 days ago. I'll wait for the CI to finish and close that bug then.

Logs:
```
Err:36 http://apt.armbian.com/ bullseye/main arm64 Packages
  File has unexpected size (68237 != 638508). Mirror sync in progress? [IP: ::1 3142]
  Hashes of expected file:
   - Filesize:638508 [weak]
   - SHA512:7215d7213322e7ea7c290ccda6f702ced24e9308edf6f72bef40d1b28324d4941c5b83e5c0827818765f358a2024ea94a4d758b7fe2debe32b4c54fc21a91a42
   - SHA256:de044653d52804811f380c7f4e9a10b9cb89c1c46e103cc03429fba931d155c2
   - SHA1:11c5b6eb6cf78b7611c21f0a8b4ea2171fd11604 [weak]
   - MD5Sum:c0e8571d2c6895edab1e4c7352dabbcc [weak]
  Release file created at: Sun, 12 Feb 2023 09:21:30 +0000
Get:37 http://apt.armbian.com/ bullseye/bullseye-utils arm64 Packages [25.3 kB]
Err:37 http://apt.armbian.com/ bullseye/bullseye-utils arm64 Packages
  
Get:38 http://apt.armbian.com/ bullseye/bullseye-utils all Packages [19.4 kB]
Err:38 http://apt.armbian.com/ bullseye/bullseye-utils all Packages
  
Get:39 http://apt.armbian.com/ bullseye/bullseye-utils armhf Packages [26.0 kB]
Err:39 http://apt.armbian.com/ bullseye/bullseye-utils armhf Packages
  
Get:40 http://apt.armbian.com/ bullseye/bullseye-desktop armhf Packages [19.8 kB]
Get:41 http://apt.armbian.com/ bullseye/bullseye-desktop all Packages [16.6 kB]
Ign:41 http://apt.armbian.com/ bullseye/bullseye-desktop all Packages
Get:42 http://apt.armbian.com/ bullseye/bullseye-desktop arm64 Packages [43.9 kB]
Fetched 25.3 MB in 22s (1131 kB/s)
Reading package lists...
E: Failed to fetch http://apt.armbian.com/dists/bullseye/main/binary-arm64/by-hash/SHA512/7215d7213322e7ea7c290ccda6f702ced24e9308edf6f72bef40d1b28324d4941c5b83e5c0827818765f358a2024ea94a4d758b7fe2debe32b4c54fc21a91a42  File has unexpected size (68237 != 638508). Mirror sync in progress? [IP: ::1 3142]
   Hashes of expected file:
    - Filesize:638508 [weak]
    - SHA512:7215d7213322e7ea7c290ccda6f702ced24e9308edf6f72bef40d1b28324d4941c5b83e5c0827818765f358a2024ea94a4d758b7fe2debe32b4c54fc21a91a42
    - SHA256:de044653d52804811f380c7f4e9a10b9cb89c1c46e103cc03429fba931d155c2
    - SHA1:11c5b6eb6cf78b7611c21f0a8b4ea2171fd11604 [weak]
    - MD5Sum:c0e8571d2c6895edab1e4c7352dabbcc [weak]
   Release file created at: Sun, 12 Feb 2023 09:21:30 +0000
E: Failed to fetch http://apt.armbian.com/dists/bullseye/bullseye-utils/binary-arm64/by-hash/SHA512/637c737886965ca0d122212f6d870b58fceea47bc267dad4f1dac22b6a99f8ee626a863e30834e54c19575a34c0f80ff03f026bc1abc6fc5d22a89d71b8a040c  
E: Failed to fetch http://apt.armbian.com/dists/bullseye/bullseye-utils/binary-all/by-hash/SHA512/6afbfd48c17c4ffa9d7e7a54f9aa292b9da58366a19a84214923058cddb3ce87ea153f15a69e61edcebe9834852d64f4513c4c56ed457efb625e44461c0f37e5  
E: Failed to fetch http://apt.armbian.com/dists/bullseye/bullseye-utils/binary-armhf/by-hash/SHA512/33ca0623f377bb3ad4ba606d61b72ed4545a7005251b81260dc113d08fc1ed400901fe63d613aaa33b5442907572bc989c1ef0305e1e7555a712a4329cb3fa29
```

igorpecovnik, 2023-02-21:
> how that could be possible?

https://www.google.com/search?q=apt+cacher+ng+problems

It helps if you add build parameter NO_APT_CACHER="yes"

But we are just about to [switch](https://www.armbian.com/newsflash/code-freeze-and-moving-to-new-framework/) to highly improved branch, where this problems is certainly gone.

mikedolx, 2023-02-22:
> https://www.google.com/search?q=apt+cacher+ng+problems

I don't google, so it would be https://duckduckgo.com/?t=ffab&q=apt+cacher+problems&ia=web 😁.

Thanks for the hint regarding the branch switch. I'll take a deep look into it. 

As the problem with the repos is gone, i'm closing this issue.

closed 2023-02-22 by mikedolx