armbian/build #4854: Kernel taken from cache even if `KERNEL_CONFIGURE=yes` is set

iav, 2023-02-23:
### What happened?

I want to build kernel with different options and use command 
```
./compile.sh BOARD=odroidn2 BRANCH=edge RELEASE=lunar BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes
```
but instead of invoking kernel configurator and compilation with new config I see
```
 Using kernel config file [ config/kernel/linux-meson64-edge.config ]
[🐳|🌱] Getting ORAS manifest [ ORAS manifest from ghcr.io/armbian/cache-kernel/kernel-meson64-edge:6.2-rc7-S4ec5-De511-P0af2-C6f95Hfe66-Bec1c ]
[🐳|🌱] Artifact is available in remote cache [ ghcr.io/armbian/cache-kernel/kernel-meson64-edge:6.2-rc7-S4ec5-De511-P0af2-C6f95Hfe66-Bec1c - 'kernel-meson64-edge - 6.2-rc7-S4ec5-De511-P0af2-C6f95Hfe66-Bec1c - version "6.2.0-rc7" git revision "4ec5183ec48656cec489c49f989c508b68b518e3" codename "Hurr durr Ima ninja sloth" drivers hash "e511b9159daace9f" patches hash "0af23bbf438c0401" .config hash "6f95495df112ecee" .config hook hash "fe661159ee64825e" framework bash hash "ec1c2353321a937a" - type: deb-tar' ]
[🐳|🌱] Obtaining artifact from remote cache [ ghcr.io/armbian/cache-kernel/kernel-meson64-edge:6.2-rc7-S4ec5-De511-P0af2-C6f95Hfe66-Bec1c into kernel-meson64-edge_6.2-rc7-S4ec5-De511-P0af2-C6f95Hfe66-Bec1c.tar ]
```

### How to reproduce?

./compile.sh BOARD=odroidn2 BRANCH=edge RELEASE=lunar BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-02-23:
Thanks @iav -- Confirmed. Will address... as workaround, add `ARTIFACT_IGNORE_CACHE=yes` 
Example:

```
./compile.sh BOARD=odroidn2 BRANCH=edge ARTIFACT_IGNORE_CACHE=yes kernel-config
```

`kernel-config`: CLI shortcut, defined in https://github.com/armbian/build/blob/main/lib/functions/cli/commands.sh

Unrelated: `KERNEL_ONLY` should be documented as deprecated...
Unrelated2: ignore the `.defconfig` file produced, it's... _experimental_ ;-)

The-going, 2023-02-24:
> ARTIFACT_IGNORE_CACHE=yes

Please tell me in a nutshell what this key does and why it is needed?

closed 2023-02-24 by igorpecovnik

rpardini, 2023-02-24:
> Please tell me in a nutshell what this key does and why it is needed?

@The-going  Nutshell is difficult, but I will try. We have implemented the concept of "artifacts", which are the individual pieces used to make image, and are individually cached. For now: u-boot/firmware (simple .deb), kernel (a tar of .debs), rootfs (a .tar.zstd). The artifacts work all the same: first, all inputs are computed (git source coordinates, branch/tag/commit, get the Makefile contents for version), plus a hash of all the patches, a hash of configuration, a hash of the bash code involved. This gives one "long" `Version: ` that represents (hopefully) all the inputs. If that hash matches either files in the local disk (cache/rootfs, or output/debs) or in remote cache (via ORAS), it is used. If not, then things are done like before (fetch git, configure, compile, package) and then cache is created.

In this case of `KERNEL_CONFIGURE=yes`, the cache might match, and thus the configuration is skipped... 🤦 
So `ARTIFACT_IGNORE_CACHE=yes` causes _all_ caches to be ignored and forces everything to be built.

Hopefully this is clear enough, if not, let me know. Also I hope someone takes this as base for some documentation...

commit d69f09d6 (2023-02-24): kernel-config: ignore artifact cache, makes no sense with make menuconfig (fixes #4854) (#4856)