armbian/build #4865: Color scheme less than ideal (black text on black background)

belegdol, 2023-02-24:
### What happened?

I tried building armbian kernel on ubuntu jammy and the console text was black on dark blue background. This is barely legible:
[image]

### How to reproduce?

```
./compile.sh BOARD=odroidxu4 BRANCH=current KERNEL_ONLY=yes KERNEL_CONFIGURE=no USE_CCACHE=no KEEP_KERNEL_CONFIG=no DEB_COMPRESS=xz
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-02-25:
Yeah, I've been chasing this for a while -- sorry for inconvenience...
The intention is for the text to be "gray", which in ANSI colors is "bright black". 
Some terminals/configs take that too literally and it is too dark, is the case with Alacritty for example:
https://github.com/armbian/build/blob/main/lib/functions/logging/logging.sh#L26-L29

Can you check and see if you can get what is your TERM env var, or some other env var (check `set` command output) that would indicate that condition? What terminal are you using?

rpardini, 2023-02-25:
@belegdol check #4866 -- you can set `TEXT_IS_TOO_DARK=yes` parameter, or even add `export TEXT_IS_TOO_DARK=yes` to your bashrc/profile file as a workaround.

belegdol, 2023-02-25:
> Yeah, I've been chasing this for a while -- sorry for inconvenience... The intention is for the text to be "gray", which in ANSI colors is "bright black". Some terminals/configs take that too [...]

This is with Ubuntu Jammy default gnome-terminal settings:
```
$ set | grep -i term
COLORTERM=truecolor
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/50012a9b_cc09_46fe_9907_0db0c53b5fe4
GNOME_TERMINAL_SERVICE=:1.101
TERM=xterm-256color
_terms () 
        command sed -ne 's/^\([^[:space:]#|]\{2,\}\)|.*/\1/p' /etc/termcap
        find /{etc,lib,usr/lib,usr/share}/terminfo/? -type f -maxdepth 1 |
        TERM)
            _terms
```

rpardini, 2023-02-25:
Thank you @belegdol  -- I think I will invert the logic, more sane: default to normal text, and if known-good terminal, use gray (bright black).

rpardini, 2023-03-17:
Logic inverted in https://github.com/armbian/build/pull/4955 -- now only iTerm2 goes bright black. Feel free to PR more adds if your terminal is gray-capable.

closed 2023-03-17 by rpardini