armbian/build #4867: 1) The mechanism for applying a series of patches is broken or missing.

The-going, 2023-02-25:
### What happened?

There is no possibility of further patch support for sunxi

### How to reproduce?

Command you gave to the framework!

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log output

```shell
There are no logs and there won't be.
I'm just reading the script text.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-02-25:
Phew. I've invested a month+ work to make series work, and it does. Why do you say it doesn't?
This doesn't help at all...

igorpecovnik, 2023-02-25:
Lets find out why this looks like it doesn't work. Compilation works https://paste.armbian.com/iwupilubez.sql but perhaps something is missing? I am sure @The-going didn't want to upset you for no reason.

I think we might need to defaults to compilation like it was before, not using caches ARTIFACT_IGNORE_CACHE=yes

rpardini, 2023-02-25:
Problem here is attitude. I've had enough. Good riddance...

The-going, 2023-02-25:
@rpardini You're rude. Do you know about this?

https://github.com/armbian/build/blob/b7921eb53e803213e48b3c06707eef53496b65e2/config/sources/families/include/sunxi64_common.inc#L12-L38

But the patch series already includes all the patches from this source.
https://github.com/armbian/build/blob/b7921eb53e803213e48b3c06707eef53496b65e2/patch/kernel/archive/sunxi-6.1/series.conf#L6-L394

#### How is this supposed to work?

igorpecovnik, 2023-02-25:
This looks like a debug / test switch, doing nothing, safe to be removed?

rpardini, 2023-02-25:
I kept the `KERNELBRANCH_MEGOUS` and `USE_MEGOUS` since that existed in some other form in the pre-next code.
I kept it as sign of respect, but as always, people don't actually test and open absurd issues like this one "series don't work", which is absolutely not true.

The-going, 2023-02-25:
> I kept the `KERNELBRANCH_MEGOUS` and `USE_MEGOUS` since that existed in some other form in the pre-next code.

This misled me. And I saw this code:
```bash
armbian-build> grep -nr series.conf lib/* extensions/* 
lib/tools/common/patching_utils.py:54:          # If the directory contains a series.conf file.
lib/tools/common/patching_utils.py:56:          series_conf_path = os.path.join(self.full_dir, "series.conf")
lib/tools/common/patching_utils.py:57:          if os.path.isfile(series_conf_path):
lib/tools/common/patching_utils.py:59:                  patches_in_series = self.parse_series_conf(series_conf_path)
lib/tools/common/patching_utils.py:67:                                  # Fix basename for patches in series.conf
lib/tools/common/patching_utils.py:73:                                          f"series.conf file {series_conf_path} contains a patch file {patch_file_name} that does not exist")
lib/tools/common/patching_utils.py:90:  def parse_series_conf(series_conf_path):
lib/tools/common/patching_utils.py:92:          with open(series_conf_path, "r") as series_conf_file:
lib/tools/common/patching_utils.py:93:                  for line in series_conf_file:
```
But you told me it was only for manual use. I was deluded.

The-going, 2023-02-25:
Okay guys, I just take your word for it.
I will be making pull requests, only for patches. But I will check the work of the patches in my sandbox.

The-going, 2023-02-25:
I'm closing this conversation.

closed 2023-02-25 by The-going

rpardini, 2023-02-25:
Ok. Sorry for sounding rude. I am under much pressure.