armbian/build #4868: Failed to enable unit, unit systemd-timesyncd.service does not exist...

AGM1968, 2023-02-25:
### What happened?

Evening Chaps, Very reluctantly 'asking' if perhaps someone else has experienced a similar build failure.

I cannot seem to move past this on VM Jammy 22.04. live server, there are some issues there, so reason I'm politely asking.

```
alistair@lalarm:~/build$ systemctl status systemd-networkd systemd-timedated systemd-timesyncd
? systemd-networkd.service - Network Configuration
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; enabled; ven>
     Active: active (running) since Sat 2023-02-25 19:57:58 SAST; 35min ago
TriggeredBy: ? systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)
   Main PID: 575 (systemd-network)
     Status: "Processing requests..."
      Tasks: 1 (limit: 1030)
     Memory: 1.5M
        CPU: 91ms
     CGroup: /system.slice/systemd-networkd.service
             +-575 /lib/systemd/systemd-networkd

Feb 25 19:57:58 lalarm systemd[1]: Starting Network Configuration...
Feb 25 19:57:58 lalarm systemd-networkd[575]: lo: Link UP
Feb 25 19:57:58 lalarm systemd-networkd[575]: lo: Gained carrier
Feb 25 19:57:58 lalarm systemd-networkd[575]: Enumeration completed
Feb 25 19:57:58 lalarm systemd[1]: Started Network Configuration.
Feb 25 19:57:58 lalarm systemd-networkd[575]: enp0s3: Link UP
Feb 25 19:57:58 lalarm systemd-networkd[575]: enp0s3: Gained carrier
Feb 25 19:57:58 lalarm systemd-networkd[575]: enp0s3: DHCPv4 address 10.0.2.15/>
Feb 25 19:58:00 lalarm systemd-networkd[575]: enp0s3: Gained IPv6LL

? systemd-timedated.service - Time & Date Service
     Loaded: loaded (/lib/systemd/system/systemd-timedated.service; static)
     Active: inactive (dead)
       Docs: man:systemd-timedated.service(8)
             man:localtime(5)
             man:org.freedesktop.timedate1(5)

Feb 25 20:02:31 lalarm systemd[1]: systemd-timedated.service: Deactivated succe>
Feb 25 20:03:23 lalarm systemd[1]: Starting Time & Date Service...
Feb 25 20:03:23 lalarm systemd[1]: Started Time & Date Service.
Feb 25 20:03:53 lalarm systemd[1]: systemd-timedated.service: Deactivated succe>
Feb 25 20:14:36 lalarm systemd[1]: Starting Time & Date Service...
Feb 25 20:14:36 lalarm systemd[1]: Started Time & Date Service.
Feb 25 20:15:06 lalarm systemd[1]: systemd-timedated.service: Deactivated succe>
Feb 25 20:28:05 lalarm systemd[1]: Starting Time & Date Service...
Feb 25 20:28:05 lalarm systemd[1]: Started Time & Date Service.
Feb 25 20:28:35 lalarm systemd[1]: systemd-timedated.service: Deactivated succe>

? systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; ve>
     Active: active (running) since Sat 2023-02-25 20:27:16 SAST; 5min ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 15316 (systemd-timesyn)
     Status: "Initial synchronization to time server 185.125.190.57:123 (ntp.ub>
      Tasks: 2 (limit: 1030)
     Memory: 1.3M
        CPU: 86ms
     CGroup: /system.slice/systemd-timesyncd.service
             +-15316 /lib/systemd/systemd-timesyncd

Feb 25 20:27:16 lalarm systemd[1]: Starting Network Time Synchronization...
Feb 25 20:27:16 lalarm systemd[1]: Started Network Time Synchronization.
Feb 25 20:27:16 lalarm systemd-timesyncd[15316]: Initial synchronization to tim>
alistair@lalarm:~/build$

alistair@lalarm:~/build$ systemctl status systemd-timesyncd.service
? systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; ve>
     Active: active (running) since Sat 2023-02-25 20:27:16 SAST; 7min ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 15316 (systemd-timesyn)
     Status: "Initial synchronization to time server 185.125.190.57:123 (ntp.ub>
      Tasks: 2 (limit: 1030)
     Memory: 1.3M
        CPU: 86ms
     CGroup: /system.slice/systemd-timesyncd.service
             +-15316 /lib/systemd/systemd-timesyncd

Feb 25 20:27:16 lalarm systemd[1]: Starting Network Time Synchronization...
Feb 25 20:27:16 lalarm systemd[1]: Started Network Time Synchronization.
Feb 25 20:27:16 lalarm systemd-timesyncd[15316]: Initial synchronization to tim>
alistair@lalarm:~/build$

```

However the build is ending here

```
[??]   Created symlink /etc/systemd/system/multi-user.target.wants/systemd-resolved.service ? /lib/systemd/system/systemd-resolved.service.
[??]   Created symlink /etc/systemd/system/NetworkManager.service ? /dev/null.
[??]   Failed to enable unit, unit systemd-timesyncd.service does not exist.
[??] Error 1 occurred in main shell [ at /home/alistair/build/lib/functions/logging/runners.sh:190
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:190
                  chroot_sdcard() --> lib/functions/logging/runners.sh:101
   install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:574
                do_with_logging() --> lib/functions/logging/section-logging.sh:72
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:21
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:10
          do_with_default_build() --> lib/functions/main/default-build.sh:17
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:17
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:126
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:164
                           main() --> compile.sh:52
 ]
[??] Cleaning up [ please wait for cleanups to finish ]
[??] Unmounting recursively [ SDCARD - be patient ]
[??] Unmounted OK after 2 attempt(s) [ SDCARD ]
[??] Unmounting recursively [ MOUNT - be patient ]
[??] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-361d5b8a-1a77-480a-b5c6-742aeab68163.log.ans ]
[??] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-361d5b8a-1a77-480a-b5c6-742aeab68163.log.ans https://paste.next.armbian.com/log ]

```

### How to reproduce?

Command you gave to the framework!

./compile.sh

menu select
```
[🌱] Configuration prepared for BOARD build [ orangepizero2.conf ]
[✨] Repeat Build Options (early) [ ./compile.sh BOARD=orangepizero2 BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,img ]
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
https://paste.next.armbian.com/mehahevahe
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-02-25:
Confirmed

https://paste.next.armbian.com/ayuzivedih

rpardini, 2023-02-25:
Nice catch. Can you spot the bug?
Position of bug was found via the stacktrace printed in the log (start from top, walk down until you find smth significant)

https://github.com/armbian/build/blob/main/lib/functions/rootfs/distro-agnostic.sh#L573-L575

rpardini, 2023-02-25:
> $SDCARD missing I think?

Yes, hopefully that is the only problem. #4870

closed 2023-02-25 by rpardini