armbian/build #4869: E: Package 'cryptsetup' has no installation candidate

derlocke-ng, 2023-02-25:
### What happened?

I'm trying to run following:
`./compile.sh CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="12345678" CRYPTROOT_SSH_UNLOCK_PORT=2223`
for the Orangepi 4 LTS, doesn't matter which kernel or os I select, same error all the time:
`E: Package 'cryptsetup' has no installation candidate`

### How to reproduce?

`git clone --depth=1 --branch=main https://github.com/armbian/build
./compile.sh CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="12345678" CRYPTROOT_SSH_UNLOCK_PORT=2223`
select opi4lts and any kernel/os combination.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-02-27:
> doesn't matter which kernel or os I select, same error all the time

I hope you read what's written below this functionality:
https://docs.armbian.com/Developer-Guide_Build-Options/

_Note: This function might not work well with all distributions. Debian Buster and Stretch were tested. For building under Docker you have to use privileged mode which can be enable in userpatches/config-docker.
Warning: This feature was added as community contribution and mostly functional. Under some circumstances though the prompt will not be shown. T[here](https://github.com/armbian/build/commit/681e58b6689acda6a957e325f12e7b748faa8330)fore it should be considered experimental. Check [here](https://github.com/armbian/build/issues/1183) and here
Hint: If you want to do the encryption part from scratch check out [this](https://forum.armbian.com/topic/15618-full-root-filesystem%C2%A0encryption%C2%A0on-an-armbian-system-new-replaces-2017-tutorial-on-this-topic/) forum post._

> got it working xD
> fix: add "openssh-client" after cryptsetup in /build/extensions/fs-cryptroot-support.sh

https://docs.armbian.com/Process_Contribute/

derlocke-ng, 2023-03-01:
Hello,

Yeah I eventually read that. Didn't know so much changed under the hood (docker part).
But for orangepi 4 lts (and some other sbcs), self-built images without encryption don't work either.

I don't know if contributing that fix would be enough. There's still the problem that you have to run it twice to enable cryptsetup at all... 2nd ssh-keygen should already be there, because it runs seconds before it says it's missing.

igorpecovnik, 2023-03-01:
> self-built images without encryption don't work either.

We made new release two weeks ago, then switch to new build system:
https://www.armbian.com/newsflash/code-freeze-and-moving-to-new-framework/

cryptsetup was never monitored so I don't know, but if you check out master branch, that you can build and they will work.

The rest is developed 24/7. We expect some general stability in 1-2 weeks.

derlocke-ng, 2023-03-02:
Thanks for your answer!

I wasn't lucky with booting images with the master branch too. But could be a problem on my side. I have to test further :)

Ok, for the main branch I would wait some weeks, then try again. If the problem persists with the missing dep I would contribute my fix. hope that's okay. I would close the issue, too.

closed 2023-11-14 by igorpecovnik