armbian/build #4871: main branch compile.sh docker image build fails on ubuntu-20.04

martinl, 2023-02-26:
### What happened?

I was planning to build armbian image on next (to test my bananapir2 board changes from master branch)

### How to reproduce?

git clone https://github.com/armbian/build.git
cd build
./compile.sh

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log output

```shell
cat output/logs/summary-docker-78ab7ee5-5f3c-4eb2-b949-80d2cf4274fc.md 
<details><summary>Build: </summary>
<p>

### Armbian logs for 78ab7ee5-5f3c-4eb2-b949-80d2cf4274fc
#### Armbian build at Sun Feb 26 13:09:34 EET 2023 on mhp
#### ARGs: ``
#### Current revision:

commit d453c873fd34bf1690d0bade1ef4c75987376073 (HEAD -> main, origin/main, origin/HEAD)
Author: Ricardo Pardini <ricardo@pardini.net>

    distro-agnostic.sh: fix: check if chroot (not host!) has `timesyncd.conf` as condition to enable `systemd-timesyncd.service` (#4870)
```
#### Git status:
```
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```
#### Git changes:
```

```

</p></details>
----

cat output/logs/log-docker-78ab7ee5-5f3c-4eb2-b949-80d2cf4274fc.log.ans 
# Armbian ANSI build logs for 78ab7ee5-5f3c-4eb2-b949-80d2cf4274fc - use "less -SR" to view
# Armbian build at Sun Feb 26 13:09:34 EET 2023 on mhp
------------------------------------------------------------------------------------------------------------
# ARGs: 
------------------------------------------------------------------------------------------------------------
# GIT revision
   commit d453c873fd34bf1690d0bade1ef4c75987376073 (HEAD -> main, origin/main, origin/HEAD)
   Author: Ricardo Pardini <ricardo@pardini.net>
   
       distro-agnostic.sh: fix: check if chroot (not host!) has `timesyncd.conf` as condition to enable `systemd-timesyncd.service` (#4870)

# GIT status
   On branch main
   Your branch is up to date with 'origin/main'.
   
   nothing to commit, working tree clean
### docker_cli_prepare.log 
--> (1) INFO: Using prebuilt Armbian image as base for 'ubuntu-jammy' [ DOCKER_ARMBIAN_BASE_IMAGE: ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
--> (1) SYSINFO: Docker info [ Docker 20.10.7 Kernel:5.4.0-139-generic RAM:11.56GiB CPUs:4 OS:'Ubuntu 20.04.5 LTS' hostname 'mhp' under 'Linux' - buildx:yes - loop-hacks:yes static-loops:no ]
[... 52 lines cut ...]
       run_host_command_logged_raw() --> lib/functions/logging/runners.sh:190
           run_host_command_logged() --> lib/functions/logging/runners.sh:172
       docker_cli_build_dockerfile() --> lib/functions/host/docker.sh:349
                   do_with_logging() --> lib/functions/logging/section-logging.sh:72
                    cli_docker_run() --> lib/functions/cli/cli-docker.sh:40
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:126
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:164
                              main() --> ./compile.sh:52
    ]
   -->--> command failed with error code 1 after 6 seconds
--> (8) WARNING: stacktrace for failed command [ exit code 1:/bin/bash -e -o pipefail -c docker buildx build --progress=plain --load -t armbian.local.only/armbian-build:initial -f /home/martinl/src/armbian-main/build/Dockerfile /home/martinl/src/armbian-main/build
           run_host_command_logged() --> lib/functions/logging/runners.sh:172
       docker_cli_build_dockerfile() --> lib/functions/host/docker.sh:349
                   do_with_logging() --> lib/functions/logging/section-logging.sh:72
                    cli_docker_run() --> lib/functions/cli/cli-docker.sh:40
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:126
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:164
                              main() --> ./compile.sh:52 ]
--> (8) ERROR: Cleaning up [ please wait for cleanups to finish ]
------------------------------------------------------------------------------------------------------------
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

martinl, 2023-02-26:
found a workaround from another issue
```
export PREFER_DOCKER=no
./compile.sh
```

rpardini, 2023-02-26:
Yes, under WSL2, Docker is neither supported nor necessary. Please use workaround until #4855 is resolved...

martinl, 2023-02-26:
Just to clarify, this issue is about failing docker image build on ubuntu 20.04 / focal. When I disable docker support using `export PREFER_DOCKER=no` then armbian image build works like it did on master branch.

rpardini, 2023-02-26:
Oh sorry -- I was too hasty in closing this, then?
I can't reproduce this though. There seems some error in the base image?
From what I could gather from the logs posted, error lies in something we don't call directly:
```
https://github.com/armbian/build/issues/5 1.563 Reading package lists...
https://github.com/armbian/build/issues/5 2.423 E: Problem executing scripts APT::Update::Post-Invoke 'rm -f /var/cache/apt/archives/.deb /var/cache/apt/archives/partial/.deb /var/cache/apt/*.bin || true'
https://github.com/armbian/build/issues/5 2.423 E: Sub-process returned an error code
```

igorpecovnik, 2023-02-26:
Corrupted deb cache / index on host machine?

rpardini, 2023-02-26:
Unlikely -- that's an error during the "build Dockerfile" phase. Might be caused by an outdated base image, but that sounds likely. I've been unable to reproduce.... could you?

martinl, 2023-02-27:
I did remove all the docker base images between each run using `docker image rm`

```
docker image ls
REPOSITORY   TAG       IMAGE ID   CREATED   SIZE
```

igorpecovnik, 2023-02-27:
> is working now with docker

Closing as we find the cause and a solution.

closed 2023-02-27 by igorpecovnik

rpardini, 2023-02-27:
@martinl Good to know, thanks!
@igorpecovnik  we could add a version check to at least warn the user....

rpardini, 2023-02-27:
@igorpecovnik realized you already updated documentation. Thanks!