armbian/build #4874: Linux kernel debs install differently than those built under master (at least for meson64-current)

SteeManMI, 2023-02-26:
### What happened?

When I install a kernel deb (meson64-current) built from master branch, I get the following kernel files installed to /boot (this is on a fat boot partition):
Image and uImage
When using a deb built from main, I only get the Image file (no uImage).
In the aml-s9xx-box builds the extlinux.conf file is trying to boot with the uImage file which doesn't exist and therefore the boot is failing.
I can fix my extlinux.conf to now expect Image instead of uImage, but before I make that change, I want to ask is this change in behavior intentional (I guess it probably isn't), or the real question I have is will this remain the behavior going forward?

### How to reproduce?

I did a build of aml-s9xx-box current jammy and the resulting img is missing /boot/uImage.  I also built it a second time (a day later so I had both 6.1.13 and a 6.1.14 kernel debs, so I installed the 6.1.14 deb to upgrade the kernel and the kernel upgrade also left /boot without a uImage file).

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-02-26:
Hey @SteeManMI -- I, sincerely, can't remember. I _think_ (can can very well be wrong) that `Image` and `uImage` were different in a very distant past -- one was directly produced by kernel build, the other a variation created by u-boot's tooling `mkimage`, back when u-boot couldn't directly load kernel images, and needed a pre-processed version. (that is still true for initrd's I think).
So I think what master had was a compatibility symlink, and I might have dropped that either accidentally or in a rage of "lets get rid of junk" (I had a few of those, lol). 
I wouldn't mind bringing it back, but lets search for references to `uImage` and if there are just a few, maybe get rid of them and point directly?

SteeManMI, 2023-02-26:
@rpardini You confirmed my thoughts as well.  I will take a look at the code and see what references to uImage remain and come back with my findings.  (I know you are pretty busy right now - keep up the good work)

SteeManMI, 2023-02-27:
@rpardini  I have generated a list of all references to 'uImage' in the code base.
[uImage-refs.txt](https://github.com/armbian/build/files/10840871/uImage-refs.txt)

While I haven't had much time to dig into them, it seems a bunch are for legacy stuff which would make sense as old u-boots likely need the uImage.
But something that caught my eye was the following patch:
patch/kernel/archive/meson64-6.1/general-kernel-odroidn2-current.patch:+mkimage -A arm64 -O linux -T kernel -C none -a 0x1080000 -e 0x1080000 -n "Linux" -d /$installed_image_path /boot/uImage  > /dev/null 2>&1
This patch is patching builddeb in the meson64-current kernel to create the uImage file.  But it doesn't seem like the build framework is using the patched builddeb to build the .deb files.  So what is this patch for?  (I need to dig into where the build framework is actually building the debs, but I would have assumed that it would be using the builddeb file and that if you wanted changes you would patch the builddeb file, like this patch is doing, so that if someone were to build a custom kernel using kernel tools, they would get a working .deb file equivalent to what armbian distributes).  
What am I missing in my understanding?

rpardini, 2023-02-27:
Yeah so your confusion is warranted, since I never explained much.
What I did was "externalize" (into the framework code, in `lib/functions/compilation/kernel*.sh`) the equivalent of `builddeb`.

Patching `builddeb` itself for different versions of kernel was a pain, and the fact that `mkdebian`/`builddeb` was run by `make`, not us, caused us to have to go around a bunch. 

So I've changed to use only the most  "basic"  operations of kernel `make`: things such as `make install`, `make install-dtb`, etc; those are packaging-agnostic and should "always work".

So now in the `kernel-debs.sh` we have a callback for each `image`/`dtb`/`headers` packaging, where we can "enjoy" the proper logging/utility functions/environment vars we know and "love" from the rest of the system.

It is _not perfect_ but should allow us to 1) have a single codebase of all kernels, instead of dozens of patches and 2) endure the next 10 years of hacks we need there.

hzyitc, 2023-03-02:
I'm also trying to solve this issue. #4891

rpardini, 2023-03-02:
Nice, @hzyitc -- I never knew kernel had `uinstall` and `zinstall` -- you also solved that horrible hack I had in there for zImage. @SteeManMI can you pick his #4891 and test?

hzyitc, 2023-03-02:
> @SteeManMI can you pick his https://github.com/armbian/build/pull/4891 and test?

Unfortunately, it seem like that only `arm` has `uinstall`. `arm64` only has `install` and `zinstall`.

https://github.com/torvalds/linux/blob/ee3f96b164688dae21e2466a57f2e806b64e8a37/arch/arm/Makefile#L322-L338

```
define archhelp
  echo  '* zImage        - Compressed kernel image (arch/$(ARCH)/boot/zImage)'
  echo  '  Image         - Uncompressed kernel image (arch/$(ARCH)/boot/Image)'
  echo  '* xipImage      - XIP kernel image, if configured (arch/$(ARCH)/boot/xipImage)'
  echo  '  uImage        - U-Boot wrapped zImage'
  echo  '  bootpImage    - Combined zImage and initial RAM disk'
  echo  '                  (supply initrd image via make variable INITRD=<path>)'
  echo  '  install       - Install uncompressed kernel'
  echo  '  zinstall      - Install compressed kernel'
  echo  '  uinstall      - Install U-Boot wrapped compressed kernel'
  echo  '                  Install using (your) ~/bin/$(INSTALLKERNEL) or'
  echo  '                  (distribution) /sbin/$(INSTALLKERNEL) or'
  echo  '                  install to $$(INSTALL_PATH) and run lilo'
  echo  '  vdso_install  - Install unstripped vdso.so to $$(INSTALL_MOD_PATH)/vdso'
  echo
  echo  '  multi_v7_lpae_defconfig     - multi_v7_defconfig with CONFIG_ARM_LPAE enabled'
endef
```

https://github.com/torvalds/linux/blob/ee3f96b164688dae21e2466a57f2e806b64e8a37/arch/arm64/Makefile#L214-L222

```
define archhelp
  echo  '* Image.gz      - Compressed kernel image (arch/$(ARCH)/boot/Image.gz)'
  echo  '  Image         - Uncompressed kernel image (arch/$(ARCH)/boot/Image)'
  echo  '  install       - Install uncompressed kernel'
  echo  '  zinstall      - Install compressed kernel'
  echo  '                  Install using (your) ~/bin/installkernel or'
  echo  '                  (distribution) /sbin/installkernel or'
  echo  '                  install to $$(INSTALL_PATH) and run lilo'
endef
```

hzyitc, 2023-03-02:
Maybe we can add a global generic patch to let `arm64` do the same thing as `armhf`?

SteeManMI, 2023-03-02:
But perhaps not exactly what armhf does..

`  echo  '  uinstall      - Install U-Boot wrapped compressed kernel'`

As I believe the expectation on arm64 is for a u-boot wrapped un-compressed kernel (not compressed)

rpardini, 2023-03-02:
Yeah, I think what we did before was to (using similar to hack) run `mkimage` directly. It's less clean, but since the hack for Z was already there, we could just add more for U... I'd rather use the kernel provided infra, but if it does not fit, we do it ourselves...

SteeManMI, 2023-03-12:
After spending some more time digging through the code here are my updated findings.
First here is an up to date list of references in the current code base for uImage:

`config/bootscripts/boot-odroid-c1.ini:ext4load mmc 0:1 0x20800000 /boot/uImage || fatload mmc 0:1 0x20800000 uImage || ext4load mmc 0:1 0x20800000 uImage
config/bootscripts/boot-odroid-c2.ini:if test -e mmc 0:1 boot/.next; then ext4load mmc 0:1 0x01080000 boot/uImage; fi
config/bootscripts/boot-odroid-n2.ini:	setenv kernelimage "uImage"
config/bootscripts/boot-onecloud.cmd:fatload ${bootdev} 0x20800000 /uImage || exit 1
config/bootscripts/boot-sun50iw9.cmd:load ${devtype} ${devnum} ${kernel_addr_r} ${prefix}uImage
patch/kernel/media-legacy/00980-builddeb.patch: 			rm -f /boot/System.map* /boot/config* /boot/vmlinuz* /boot/$image_name /boot/uImage
patch/kernel/rk3399-legacy/04-patch-4.4.180-181.patch:+all-$(CONFIG_MACH_PISTACHIO)		:= uImage.gz
patch/kernel/rockchip-legacy/04-patch-4.4.180-181.patch:+all-$(CONFIG_MACH_PISTACHIO)		:= uImage.gz
patch/kernel/rockchip-legacy/310_add_basic_dtbo_structure.patch: uImage
patch/kernel/sun50iw9-legacy/patch-4.9.180-181.patch:+all-$(CONFIG_MACH_PISTACHIO)		:= uImage.gz
patch/kernel/sun50iw9-legacy/patch-4.9.252-253.patch: boot_targets += uImage uImage.bin uImage.gz
patch/kernel/archive/meson-6.1/generate-uImage-instand-of-zImage.patch:Generate uImage instand of zImage
patch/kernel/archive/meson-6.1/generate-uImage-instand-of-zImage.patch:+mkimage -A arm -O linux -T kernel -C none -a 0x2080000 -e 0x2080000 -n "Linux" -d /boot/zImage /boot/uImage  > /dev/null 2>&1
patch/kernel/archive/meson64-5.10/general-kernel-odroidn2-current.patch:+mkimage -A arm64 -O linux -T kernel -C none -a 0x1080000 -e 0x1080000 -n "Linux" -d /$installed_image_path /boot/uImage  > /dev/null 2>&1
patch/kernel/archive/meson64-6.1/general-kernel-odroidn2-current.patch:+mkimage -A arm64 -O linux -T kernel -C none -a 0x1080000 -e 0x1080000 -n "Linux" -d /$installed_image_path /boot/uImage  > /dev/null 2>&1
patch/kernel/archive/meson64-6.2/general-kernel-odroidn2-current.patch:+mkimage -A arm64 -O linux -T kernel -C none -a 0x1080000 -e 0x1080000 -n "Linux" -d /$installed_image_path /boot/uImage  > /dev/null 2>&1
patch/kernel/archive/rk3399-4.4/04-patch-4.4.180-181.patch:+all-$(CONFIG_MACH_PISTACHIO)		:= uImage.gz
patch/kernel/archive/rockchip-4.4/04-patch-4.4.180-181.patch:+all-$(CONFIG_MACH_PISTACHIO)		:= uImage.gz
patch/kernel/archive/rockchip-4.4/310_add_basic_dtbo_structure.patch: uImage
patch/kernel/archive/sun50iw9-4.9/patch-4.9.180-181.patch:+all-$(CONFIG_MACH_PISTACHIO)		:= uImage.gz
patch/kernel/archive/sun50iw9-4.9/patch-4.9.252-253.patch: boot_targets += uImage uImage.bin uImage.gz
packages/bsp/aml-s9xx-box/boot/extlinux/extlinux.conf:LINUX /uImage
packages/bsp/common/usr/sbin/armbian-install:			/boot/zImage "${TempDir}"/bootfs/uImage >/dev/null 2>&1
packages/bsp/common/usr/sbin/armbian-install:			cp "${TempDir}"/bootfs/uImage "${TempDir}"/rootfs/boot/uImage
packages/bsp/common/usr/sbin/nand-sata-install:			/boot/zImage "${TempDir}"/bootfs/uImage >/dev/null 2>&1
packages/bsp/common/usr/sbin/nand-sata-install:			cp "${TempDir}"/bootfs/uImage "${TempDir}"/rootfs/boot/uImage
`

SteeManMI, 2023-03-12:
Based on the code references, it seems the only remaining uses of uImage in practice are:
 - odriod-c1 which is meson family and currently .eos
 - odroid-c2 which is meson64 family
 - onecloud which is meson family
 - orangepizero2(legacy) which is sun50iw9 board family
 - aml-s9xx-box which is meson64 family

If my findings are correct, my recommendation would be to completely move away from using uImage and fix these remaining boards.  The pending [PR4891](https://github.com/armbian/build/pull/4891) adds back uImage for the meson family (so would address onecloud and odriod-c1, but not by bringing them into line with other boards).
I will submit a PR to fix aml-s9xx-box by moving away from uImage.
That leaves orangepizero2(legacy) and odroid-c2 as broken.

rpardini, 2023-04-29:
Since then, `orangepizero2` has been fixed (with custom hooks to produce the needed `uImage`, with custom load address). `odroid-c2` could get same/similar treatment?

closed 2023-05-21 by rpardini