armbian/build #4884: Minimal build  armbian-firstlogin ... set_timezone_and_locales()

AGM1968, 2023-02-28:
### What happened?

Issue with setting TZ:

What have found, am assuming that the packages must be listed?

FLLOWING DO NOT HAVE LISTED:
/build/config/cli/bullseye/debootstrap/packages
/build/config/cli/bullseye/main/packages
/build/config/cli/bullseye/main/config_cli_standard/packages
/build/config/cli/bullseye/main/config_cli_standard/packages.additional

curl
nmcli
jq

FOLLOWING DOES NOT HAVE LISTED:
/build/config/cli/bullseye/main/packages.additional
nmcli

### How to reproduce?

[🌱] Configuration prepared for BOARD build [ orangepizero2.conf ]
[✨] Repeat Build Options (early) [ ./compile.sh BOARD=orangepizero2 BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,img ]

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Relevant log output

```shell
/usr/lib/armbian/armbian-firstlogin: line 111: curl: command not found
/usr/lib/armbian/armbian-firstlogin: line 114: nmcli: command not found
/usr/lib/armbian/armbian-firstlogin: line 147: curl: command not found
/usr/lib/armbian/armbian-firstlogin: line 147: jq: command not found

```
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-02-28:
> If on the right track will test and send pr

If possible, we would rather clean this. I made a plan https://armbian.atlassian.net/browse/AR-1557 If you can follow, assign a ticket. In case of troubles, we are here.

AGM1968, 2023-02-28:
Thank you, that's completely understandable. It was clear that the findings are not limited specifically there alone. Also the structure was not clear.

rpardini, 2023-02-28:
Maybe: we need to move packages that contain utilities used by firstrun from `packages.additional` (not included in minimal images), to the `packages` file (included in every build). my guess: currently this needs to be done for each distribution, so Igor is asking for help in rationalizing the whole system before attempting such a feat...?

igorpecovnik, 2023-02-28:
> we need to move packages that contain utilities used by firstrun from packages.additional (not included in minimal images), to the packages

Yes. 

> asking for help in rationalizing the whole system

Now its probably the best time for cleaning those lists.

AGM1968, 2023-03-05:
Thanks  @igorpecovnik  closing now, test  good with:  Armbian_23.05.0-trunk_Orangepizero2_bullseye_edge_6.2.2_minimal.img

closed 2023-03-05 by AGM1968