armbian/build #4885: KERNEL_CONFIGURE=yes saves .config into /armbian/config/kernel

iav, 2023-02-28:
### What happened?

When I add `KERNEL_CONFIGURE=yes` to build command after wotk with gui configurator finished config saves into 2 place:
```
[🐳|🌱] Exporting new kernel config [ /armbian/output/config/linux-rockchip64-edge.config ]
[🐳|🔨]   '.config' -> '/armbian/output/config/linux-rockchip64-edge.config'
[🐳|🌱] Exporting new kernel config - git commit pending [ /armbian/config/kernel/linux-rockchip64-edge.config ]
[🐳|🔨]   '.config' -> '/armbian/config/kernel/linux-rockchip64-edge.config'
[🐳|🔨]   'defconfig' -> '/armbian/output/config/linux-rockchip64-edge.defconfig'
[🐳|🔨]   'defconfig' -> '/armbian/config/kernel/linux-rockchip64-edge.config.defconfig'
```
`config/kernel/` and `output/config/`.

I think it's a bug, new config should be saved only into `output/config`, as early.

But I am not sure.
Am I right, or miss something?

### How to reproduce?

`./compile.sh BOARD=helios64 BRANCH=edge RELEASE=lunar BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes `

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
(https://paste.next.armbian.com/irugisalis)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-03-01:
Oh yes, you're right.  As you noticed I also added the `defconfig` there.

Those were experiments that I left in; might be a bug or a feature, depending on your point of view. In the end, it was meant as a "conversation starter" which just happened...

1) Writing to the place "where the .config came from originally" (could also be userpatches...) could slightly help with sending PR's (was intended for Igor's enjoyment, dunno if it happened, or if it just annoys). I am 100% ok with removing it or putting under a flag.

2) Switching to the `defconfig` might alleviate the absolute insanity that is managing all these configs long-term? Or not. 🤣  Again, also super OK with removing or going forward.

Thanks for catching this!

The-going, 2023-03-02:
> I think it's a bug, new config should be saved only into `output/config`, as early.

I totally agree.
The best option is the ability to save in this folder with a choice of file name.
And allow users to use this file in the subsequent assembly\reassembly by simply specifying this name in the key.

example.
old variant:
```bash
KERNEL_CONFIGURE="yes"
KERNEL_KEEP_CONFIG="yes"
```
new variant:
```bash
KERNEL_CONFIGURE="yes"
KERNEL_KEEP_CONFIG="bpim64-mini.v715-rt.config"
```

igorpecovnik, 2023-03-04:
> or putting under a flag.

This. I find it useful. For defconfig need some brainstorming ...

rpardini, 2023-03-15:
Hmm, `.defconfig` is not used at all (hopefully, unless I messed up badly), either in userpatches or config/kernel. You can copy/move the .defconfig as .config and that will work, though. 

I think @amazingfate had some benefits from using the defconfig for some config diff'ing, and I think even suggested to manually  switch from regular-huge-.config to .defconfig-slim-.config on rk3588 legacy... If experience is good, we can automate the switch and use for all families (just write `savedefconfig`'s output into .config, and drop the separate .defconfig).

For now, I think adding `*.defconfig` to .gitignore is probably a good idea?

enok71, 2024-04-02:
In yocto it's possible to create defconfig "fragments", that is a sort of overrides to the selected targets defconfig. These fragments tend to become really minimal and readable. Would be really convenient to also produce such fragment file in armbians kernel-config function (or some new function) and then just be able to drop such fragments under userpatches.
See e.g. [here](https://wiki.koansoftware.com/index.php/Modify_the_linux_kernel_with_configuration_fragments_in_Yocto)

igorpecovnik, 2025-01-19:
This probably becomes deprecated after https://github.com/armbian/build/pull/7651

closed 2025-01-19 by igorpecovnik