armbian/build #4892: OrangepiZero2 Builds Starting kernel ... panic

AGM1968, 2023-03-02:
### What happened?

Builds complete without reporting failures.....  Do not move beyond Starting kernel ...

STATUS: Investigating possible causes! 
`=> booti ${kernel_addr_r} - ${fdtcontroladdr}` etc

```
U-Boot SPL 2022.07-armbian (Mar 01 2023 - 13:16:25 +0200)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):armbian
NOTICE:  BL31: Built : 13:11:09, Mar  1 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0906e8, model: OrangePi Zero2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
INFO:    PMIC: aldo1 voltage: 3.300V
INFO:    PMIC: aldo2 voltage: 3.300V
INFO:    PMIC: aldo3 voltage: 3.300V
INFO:    PMIC: bldo1 voltage: 1.800V
INFO:    PMIC: bldo2 voltage: 1.800V
INFO:    PMIC: dcdcd voltage: 1.500V
INFO:    PMIC: dcdce voltage: 3.300V
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2022.07-armbian (Mar 01 2023 - 13:16:25 +0200) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero2
DRAM:  1 GiB
Core:  52 devices, 18 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   phy interface8
eth0: ethernet@5020000
Autoboot in 1 seconds, press <Space> to stop
=> bdinfo
boot_params = 0x0000000040000100
DRAM bank   = 0x0000000000000000
-> start    = 0x0000000040000000
-> size     = 0x0000000040000000
flashstart  = 0x0000000000000000
flashsize   = 0x0000000000000000
flashoffset = 0x0000000000000000
baudrate    = 115200 bps
relocaddr   = 0x000000007ff54000
reloc off   = 0x0000000035f54000
Build       = 64-bit
current eth = ethernet@5020000
ethaddr     = 02:00:b6:96:2c:af
IP addr     = <NULL>
fdt_blob    = 0x000000007bf2eb20
new_fdt     = 0x000000007bf2eb20
fdt_size    = 0x00000000000052a0
lmb_dump_all:
 memory.cnt  = 0x1
 memory[0]      [0x40000000-0x7fffffff], 0x40000000 bytes flags: 0
 reserved.cnt  = 0x2
 reserved[0]    [0x40000000-0x4003ffff], 0x00040000 bytes flags: 4
 reserved[1]    [0x7bf2a6b0-0x7fffffff], 0x040d5950 bytes flags: 0
devicetree  = separate
arch_number = 0x0000000000000000
TLB addr    = 0x000000007fff0000
irq_sp      = 0x000000007bf2eb10
sp start    = 0x000000007bf2eb10
Early malloc usage: 490 / 2000
=> mmcinfo
Device: mmc@4020000
Manufacturer ID: 96
OEM: 4445
Name: SD
Bus Speed: 50000000
Mode: SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 29.6 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
=> ls mmc 0 /boot
<DIR>       4096 .
<DIR>       4096 ..
<SYM>         22 Image
<SYM>         18 dtb
<DIR>       4096 dtb-6.1.11-sunxi64
          230454 boot.bmp
               0 .next
            1536 armbian_first_run.txt.template
         3495066 System.map-6.1.11-sunxi64
             131 armbianEnv.txt
<SYM>         22 uInitrd
            3187 boot.cmd
            3259 boot.scr
        11606729 uInitrd-6.1.11-sunxi64
          206049 config-6.1.11-sunxi64
        22423560 vmlinuz-6.1.11-sunxi64
        11606665 initrd.img-6.1.11-sunxi64
=>
```

```
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x4FC00000
```

### How to reproduce?

Both Standard and Minimal builds are failing

``` ./compile.sh BOARD=orangepizero2 BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,img```

```
./compile.sh BOARD=orangepizero2 BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,img```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
`https://paste.next.armbian.com/uvenaxekun`
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-03-02:
> Builds complete without reporting failures.
So not a building issue which this issue tracker is intended for. Please use forums for issues **running** Armbian

closed 2023-03-02 by EvilOlaf

AGM1968, 2023-03-02:
There no point in discussing this in the forums.
How sure are you that something in the build system is not causing a problem here?

EvilOlaf, 2023-03-02:
Since basically all bugs found in Armbian _when running_ images need code adjustment there would be no point to have forums at all following that logic.
Therefore we created a simple rule of deviation: Problems building -> Github. Problems running -> Forums

However we are on our way to move development related discussions entirely to Github but this will take a while to get there. Probably not before 23.05 release: https://armbian.atlassian.net/browse/AR-1529

Side questions: Did this work with old master?

AGM1968, 2023-03-02:
Thanks Werner, yes look Ive applied and Igor accepted for me to help maintain Zero2. 

I believe something is happening in the build system and am trying to figure it out.

Incidentally the released CI test images boot no problem I just tried that  right now again, so it not my system or running problem. https://forum.armbian.com/topic/26661-armbian-2302-quoll-testings/

Yes I will go back to master, its on list to rule out.

I must also test minimal that built a couple of days back again.

I don't think I've been on boarded quite yet and cant log into armbian Jira etc, so maybe I can message you on the forum about that.

Thanks for answering.

EvilOlaf, 2023-03-02:
Feel free to re-open if it works on master since then definitely something is fishy ;)

rpardini, 2023-03-02:
Yes, this might be related to #4874 and already being worked on https://github.com/armbian/build/pull/4891

rpardini, 2023-03-02:
@AGM1968 can you try stopping boot like you did, but then pasting the bootscript (.cmd / scr) line by line and seeing what the output is? My guess uImage will gag...

AGM1968, 2023-03-02:
Suspected it might well be, will do so and report..

AGM1968, 2023-03-04:
Output as requested @rpardini 

```

U-Boot SPL 2022.07-armbian (Mar 01 2023 - 13:16:25 +0200)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):armbian
NOTICE:  BL31: Built : 13:11:09, Mar  1 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0906e8, model: OrangePi Zero2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
INFO:    PMIC: aldo1 voltage: 3.300V
INFO:    PMIC: aldo2 voltage: 3.300V
INFO:    PMIC: aldo3 voltage: 3.300V
INFO:    PMIC: bldo1 voltage: 1.800V
INFO:    PMIC: bldo2 voltage: 1.800V
INFO:    PMIC: dcdcd voltage: 1.500V
INFO:    PMIC: dcdce voltage: 3.300V
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2022.07-armbian (Mar 01 2023 - 13:16:25 +0200) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero2
DRAM:  1 GiB
Core:  52 devices, 18 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   phy interface8
eth0: ethernet@5020000
Autoboot in 1 seconds, press <Space> to stop
=> setenv load_addr "0x45000000"
=> setenv overlay_error "false"
=> setenv rootdev "/dev/mmcblk0p1"
=> setenv verbosity "7"
=> setenv rootfstype "ext4"
=> setenv console "both"
=> setenv docker_optimizations "on"
=> setenv bootlogo "false"
=> itest.b *0x10028 == 0x00 && echo "U-boot loaded from SD"
U-boot loaded from SD
=> echo "Boot script loaded from ${devtype}"
Boot script loaded from
=> if test -e ${devtype} ${devnum} ${prefix}armbianEnv.txt; then
> load ${devtype} ${devnum} ${load_addr} ${prefix}armbianEnv.txt
> env import -t ${load_addr} ${filesize}
> fi
** Bad device specification 0x45000000 armbianEnv **
** Bad device specification 0x45000000 armbianEnv.txt **
Couldn't find partition 0x45000000 armbianEnv.txt
Can't set block device
## Warning: Input data exceeds 1048576 bytes - truncated
## Info: input data size = 1048578 = 0x100002
=> if test "${console}" = "display" || test "${console}" = "both"; then setenv consoleargs "console=ttyS0,115200 console=tty1"; fi
=> if test "${console}" = "serial"; then setenv consoleargs "console=ttyS0,115200"; fi
[... 45 lines cut ...]
** Bad device specification 0x45000000 dtb/allwinner/overlay/-fixup **
** Bad device specification 0x45000000 dtb/allwinner/overlay/-fixup.scr **
Couldn't find partition 0x45000000 dtb/allwinner/overlay/-fixup.scr
Can't set block device
** Bad device specification 0x45000000 fixup **
** Bad device specification 0x45000000 fixup.scr **
Couldn't find partition 0x45000000 fixup.scr
Can't set block device
Applying user provided fixup script (fixup.scr)
## Executing script at 45000000
Wrong image format for "source" command
=> load ${devtype} ${devnum} ${ramdisk_addr_r} ${prefix}uInitrd
** Bad device specification 0x4FF00000 uInitrd **
** Bad device specification 0x4FF00000 uInitrd **
Couldn't find partition 0x4FF00000 uInitrd
Can't set block device
=> load ${devtype} ${devnum} ${kernel_addr_r} ${prefix}Image
** Bad device specification 0x40080000 Image **
** Bad device specification 0x40080000 Image **
Couldn't find partition 0x40080000 Image
Can't set block device
=> booti ${kernel_addr_r} ${ramdisk_addr_r} ${fdt_addr_r}
Bad Linux ARM64 Image magic!
=>
```

AGM1968, 2023-03-04:
To start, from what I can see the script doesn't load

`=> echo "Boot script loaded from ${devtype}"
Boot script loaded from`

hzyitc, 2023-03-04:
It's seem like that `devtype` and `devnum` are not set.

AGM1968, 2023-03-04:
yes I agree, constant error there.. (am only investigating uboot specific atm, zero2 seems always in trouble)

AGM1968, 2023-03-04:
Further info - in process

```
U-Boot SPL 2022.07-armbian (Mar 01 2023 - 13:16:25 +0200)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):armbian
NOTICE:  BL31: Built : 13:11:09, Mar  1 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0906e8, model: OrangePi Zero2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
INFO:    PMIC: aldo1 voltage: 3.300V
INFO:    PMIC: aldo2 voltage: 3.300V
INFO:    PMIC: aldo3 voltage: 3.300V
INFO:    PMIC: bldo1 voltage: 1.800V
INFO:    PMIC: bldo2 voltage: 1.800V
INFO:    PMIC: dcdcd voltage: 1.500V
INFO:    PMIC: dcdce voltage: 3.300V
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2022.07-armbian (Mar 01 2023 - 13:16:25 +0200) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero2
DRAM:  1 GiB
Core:  52 devices, 18 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   phy interface8
eth0: ethernet@5020000
Autoboot in 1 seconds, press <Space> to stop
=> mmc list
mmc@4020000: 0 (SD)
=>
mmc@4020000: 0 (SD)
=> load mmc 0:1 4FC00000  /boot/boot.scr
3303 bytes read in 2 ms (1.6 MiB/s)
=> md.b 4FC00000 CE7
4fc00000: 27 05 19 56 a6 72 d2 35 64 00 81 09 00 00 0c a7  '..V.r.5d.......
4fc00010: 00 00 00 00 00 00 00 00 d9 3d ca fa 05 02 06 00  .........=......
4fc00020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
4fc00030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
4fc00040: 00 00 0c 9f 00 00 00 00 23 20 44 4f 20 4e 4f 54  ........# DO NOT
4fc00050: 20 45 44 49 54 20 54 48 49 53 20 46 49 4c 45 0a   EDIT THIS FILE.
4fc00060: 23 0a 23 20 50 6c 65 61 73 65 20 65 64 69 74 20  #.# Please edit
4fc00070: 2f 62 6f 6f 74 2f 61 72 6d 62 69 61 6e 45 6e 76  /boot/armbianEnv
4fc00080: 2e 74 78 74 20 74 6f 20 73 65 74 20 73 75 70 70  .txt to set supp
4fc00090: 6f 72 74 65 64 20 70 61 72 61 6d 65 74 65 72 73  orted parameters
4fc000a0: 0a 23 0a 0a 23 20 64 65 66 61 75 6c 74 20 76 61  .#..# default va
4fc000b0: 6c 75 65 73 0a 73 65 74 65 6e 76 20 6c 6f 61 64  lues.setenv load
4fc000c0: 5f 61 64 64 72 20 22 30 78 34 35 30 30 30 30 30  _addr "0x4500000
4fc000d0: 30 22 0a 73 65 74 65 6e 76 20 6f 76 65 72 6c 61  0".setenv overla
4fc000e0: 79 5f 65 72 72 6f 72 20 22 66 61 6c 73 65 22 0a  y_error "false".
4fc000f0: 73 65 74 65 6e 76 20 72 6f 6f 74 64 65 76 20 22  setenv rootdev "
4fc00100: 2f 64 65 76 2f 6d 6d 63 62 6c 6b 30 70 31 22 0a  /dev/mmcblk0p1".
4fc00110: 73 65 74 65 6e 76 20 76 65 72 62 6f 73 69 74 79  setenv verbosity
[... 167 lines cut ...]
4fc00b90: 2e 73 63 72 29 22 0a 09 09 73 6f 75 72 63 65 20  .scr)"...source
4fc00ba0: 24 7b 6c 6f 61 64 5f 61 64 64 72 7d 0a 09 66 69  ${load_addr}..fi
4fc00bb0: 0a 66 69 0a 0a 6c 6f 61 64 20 24 7b 64 65 76 74  .fi..load ${devt
4fc00bc0: 79 70 65 7d 20 24 7b 64 65 76 6e 75 6d 7d 20 24  ype} ${devnum} $
4fc00bd0: 7b 72 61 6d 64 69 73 6b 5f 61 64 64 72 5f 72 7d  {ramdisk_addr_r}
4fc00be0: 20 24 7b 70 72 65 66 69 78 7d 75 49 6e 69 74 72   ${prefix}uInitr
4fc00bf0: 64 0a 6c 6f 61 64 20 24 7b 64 65 76 74 79 70 65  d.load ${devtype
4fc00c00: 7d 20 24 7b 64 65 76 6e 75 6d 7d 20 24 7b 6b 65  } ${devnum} ${ke
4fc00c10: 72 6e 65 6c 5f 61 64 64 72 5f 72 7d 20 24 7b 70  rnel_addr_r} ${p
4fc00c20: 72 65 66 69 78 7d 49 6d 61 67 65 0a 0a 23 62 6f  refix}Image..#bo
4fc00c30: 6f 74 69 20 24 7b 6b 65 72 6e 65 6c 5f 61 64 64  oti ${kernel_add
4fc00c40: 72 5f 72 7d 20 24 7b 72 61 6d 64 69 73 6b 5f 61  r_r} ${ramdisk_a
4fc00c50: 64 64 72 5f 72 7d 20 24 7b 66 64 74 5f 61 64 64  ddr_r} ${fdt_add
4fc00c60: 72 5f 72 7d 0a 62 6f 6f 74 69 20 24 7b 6b 65 72  r_r}.booti ${ker
4fc00c70: 6e 65 6c 5f 61 64 64 72 5f 72 7d 20 2d 20 24 7b  nel_addr_r} - ${
4fc00c80: 66 64 74 63 6f 6e 74 72 6f 6c 61 64 64 72 7d 0a  fdtcontroladdr}.
4fc00c90: 0a 23 20 52 65 63 6f 6d 70 69 6c 65 20 77 69 74  .# Recompile wit
4fc00ca0: 68 3a 0a 23 20 6d 6b 69 6d 61 67 65 20 2d 43 20  h:.# mkimage -C
4fc00cb0: 6e 6f 6e 65 20 2d 41 20 61 72 6d 20 2d 54 20 73  none -A arm -T s
4fc00cc0: 63 72 69 70 74 20 2d 64 20 2f 62 6f 6f 74 2f 62  cript -d /boot/b
4fc00cd0: 6f 6f 74 2e 63 6d 64 20 2f 62 6f 6f 74 2f 62 6f  oot.cmd /boot/bo
4fc00ce0: 6f 74 2e 73 63 72 0a                             ot.scr.
=>

```

AGM1968, 2023-03-04:
```
=>
<DIR>       4096 .
<DIR>       4096 ..
<SYM>         22 Image
<SYM>         18 dtb
<DIR>       4096 dtb-6.1.11-sunxi64
          230454 boot.bmp
               0 .next
            1536 armbian_first_run.txt.template
         3495066 System.map-6.1.11-sunxi64
             131 armbianEnv.txt
<SYM>         22 uInitrd
            3231 boot.cmd
            3303 boot.scr
        11606729 uInitrd-6.1.11-sunxi64
          206049 config-6.1.11-sunxi64
        22423560 vmlinuz-6.1.11-sunxi64
        11606665 initrd.img-6.1.11-sunxi64
=> load mmc 0 $kernel_addr_r /boot/Image
22423560 bytes read in 930 ms (23 MiB/s)
=>
=> iminfo $kernel_addr_r

## Checking Image at 40080000 ...
Unknown image format!
=>
```

AGM1968, 2023-03-04:
`## Executing script at 45000000 Wrong image format for "source" command` Might also be related to boot.scr load config issue, requiring - CONFIG_LEGACY_IMAGE_FORMAT=y  .... nonetheless overlays are not present so that can be ignored (will check for completeness!)

rpardini, 2023-03-05:
Yeah, when stopping boot, distroboot is only partially ran, so you need to set the values of the devtype / devnum and prefix manually so it you can reproduce manually. This varies per uboot version too...

AGM1968, 2023-03-05:
Thanks - still uncertain of a few things.

```
=> mmc dev ${devnum}
switch to partitions #0, OK
mmc0 is current device
=> mmc dev ${devtype}
switch to partitions #0, OK
mmc0 is current device
=> ls mmc 0 /boot
<DIR>       4096 .
<DIR>       4096 ..
<SYM>         22 Image
<SYM>         18 dtb
<DIR>       4096 dtb-6.1.11-sunxi64
          230454 boot.bmp
               0 .next
            1536 armbian_first_run.txt.template
         3495066 System.map-6.1.11-sunxi64
             131 armbianEnv.txt
<SYM>         22 uInitrd
            3231 boot.cmd
            3303 boot.scr
        11606729 uInitrd-6.1.11-sunxi64
          206049 config-6.1.11-sunxi64
        22423560 vmlinuz-6.1.11-sunxi64
        11606665 initrd.img-6.1.11-sunxi64
              79 uboot.env
              79 uboot.txt
=> load mmc 0:1 $fdt_addr_r dtb-6.1.11-sunxi64
Failed to load 'dtb-6.1.11-sunxi64'
=> load mmc 0:1 $kernel_addr_r vmlinuz-6.1.11-sunxi64
Failed to load 'vmlinuz-6.1.11-sunxi64'
=> load mmc 0:1 $ramdisk_addr_r initrd.img-6.1.11-sunxi64
Failed to load 'initrd.img-6.1.11-sunxi64'
=> blkcache show
hits: 260
misses: 7
entries: 2
max blocks/entry: 8
max cache entries: 32
=>  load mmc 0:1 4FC00000 /boot/uInitrd-6.1.11-sunxi64
11606729 bytes read in 482 ms (23 MiB/s)
=>
=> load mmc 0:1 4FC00000 /boot/initrd.img-6.1.11-sunxi64
11606665 bytes read in 482 ms (23 MiB/s)
=>
=> load mmc 0:1 4FC00000 /boot/vmlinuz-6.1.11-sunxi64
22423560 bytes read in 929 ms (23 MiB/s)
=>
=> load mmc 0:1 4FC00000 /boot/dtb-6.1.11-sunxi64
Failed to load '/boot/dtb-6.1.11-sunxi64'
=>
```

AGM1968, 2023-03-05:
Armbian_23.05.0-trunk_Orangepizero2_bullseye_edge_6.2.2_minimal.img

NOW BOOTING - must  of been something to check with dtb?

Dunno,  -- thanks for the suggestions!

AGM1968, 2023-03-08:
Apologies for the continued long posts here!

Unfortunately adding patches in `main` to `Armbian_23.05.0-trunk_Orangepizero2_bullseye_edge_6.2.2_minimal.img` result is failed boot.

In order to prove/disprove theories, had to reset my master branch as follows:

```
$ git reset --hard babc8d52fe140d19d0a6ed5777956505cecccdd6

Revert extlinux for roc-rk3328-cc (#4840)

As extlinux is not matured within the armbian tools (armbian-config in particular), revert change until such a time as the proponents of extlinux mature the Armbian utilities. 

-Fixes use of overlays

$ git push -f
Total 0 (delta 0), reused 0 (delta 0)
To https://github.com/AGM1968/build.git
 + b2e9015c6...babc8d52f master -> master (forced update)
```

Then update remote/pull locally to include the depreciation note:

```
Add deprecation note (#4845)
```

Clone to VM and configure for building from master branch, all resulting in bootable image.

```
$ sudo git clone --branch master https://github.com/AGM1968/build.git
Cloning into 'build'...
remote: Enumerating objects: 102866, done.
remote: Counting objects: 100% (682/682), done.
remote: Compressing objects: 100% (549/549), done.
remote: Total 102866 (delta 134), reused 263 (delta 104), pack-reused 102184
Receiving objects: 100% (102866/102866), 503.76 MiB | 1.42 MiB/s, done.
Resolving deltas: 100% (68064/68064), done.
Updating files: 100% (11856/11856), done.

```
Walk through set-up completes Welcome to ARMBIAN!

```
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3259 bytes read in 2 ms (1.6 MiB/s)
## Executing script at 4fc00000
U-boot loaded from SD
Boot script loaded from mmc
131 bytes read in 1 ms (127.9 KiB/s)
23441 bytes read in 5 ms (4.5 MiB/s)
Failed to load '/boot/dtb/allwinner/overlay/-fixup.scr'
10692012 bytes read in 445 ms (22.9 MiB/s)
22192136 bytes read in 921 ms (23 MiB/s)
Moving Image from 0x40080000 to 0x40200000, end=417a0000
## Loading init Ramdisk from Legacy Image at 4ff00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    10691948 Bytes = 10.2 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 495cd000, end 49fff56c ... OK
   Loading Device Tree to 000000004955e000, end 00000000495ccfff ... OK
[... 219 lines cut ...]
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-sunxi
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default
*** ERROR: `serverip' not set
Config file not found
BOOTP broadcast 1
*** Unhandled DHCP Option in OFFER/ACK: 125
*** Unhandled DHCP Option in OFFER/ACK: 125
DHCP client bound to address 10.0.0.6 (104 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
BOOTP broadcast 1
*** Unhandled DHCP Option in OFFER/ACK: 125
*** Unhandled DHCP Option in OFFER/ACK: 125
DHCP client bound to address 10.0.0.6 (129 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
=>

```

AGM1968, 2023-03-08:
Package repositories

`Get:5 http://imola.armbian.com/apt bullseye/main arm64 armbian-bsp-cli-orangepizero2 arm64 23.02.2 [498 kB]`
`Get:8 http://mirrors.netix.net/armbian/apt bullseye/main arm64 linux-image-edge-sunxi64 arm64 23.02.2 [34.3 MB]`
`Get:7 http://imola.armbian.com/apt bullseye/main arm64 linux-dtb-edge-sunxi64 arm64 23.02.2 [101 kB]`
`Get:6 http://armbian.systemonachip.net/apt bullseye/main arm64 armbian-firmware all 23.02.2 [10.1 MB]`
`Get:9 http://armbian.systemonachip.net/apt bullseye/main arm64 linux-u-boot-orangepizero2-edge arm64 23.02.2 [273 kB]`

igorpecovnik, 2023-03-08:
> No FDT memory address configured. Please configure

IMO this is the core of the problem - why this is not defined in u-boot / lost in some way.

igorpecovnik, 2023-03-08:
Using this image:
https://redirect.armbian.com/region/EU/orangepizero2/Jammy_edge
Logs:
https://paste.armbian.com/odozoxeyux
HW revision 1.3

Can't reproduce :(

AGM1968, 2023-03-08:
To be honest I really don't know other than sometimes it works other times not so.

AGM1968, 2023-03-08:
Will investigate further this evening, thanks for having a look

igorpecovnik, 2023-03-08:
Still looking. I forgot to run upgrade ...

AGM1968, 2023-03-08:
Ja I saw that>>>> hahaha

AGM1968, 2023-03-08:
I'll also venture to be a bit bolder in agreement and say that the fdt address is the common problem ..
 Furthermore that the other common errors are red herrings

AGM1968, 2023-03-09:
For clarification purpose updating from master fails. From main there is less likely a problem, I cant test the update because of build failure.

on main  'wireless drivers' bluetooth specific are problematic.

```
		 [🔨]   In file included from ./include/net/bluetooth/hci_core.h:32,
[🔨]                    from drivers/bluetooth/hci_vhci.c:27:
[🔨]   ./include/net/bluetooth/hci.h:310:9: error: redeclaration of enumerator 'HCI_QUIRK_BROKEN_LOCAL_EXT_FTR_MAX_PAGE'
[🔨]     310 |         HCI_QUIRK_BROKEN_LOCAL_EXT_FTR_MAX_PAGE,
[🔨]         |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[🔨]   ./include/net/bluetooth/hci.h:283:9: note: previous definition of 'HCI_QUIRK_BROKEN_LOCAL_EXT_FTR_MAX_PAGE' with type 'int'
[🔨]     283 |         HCI_QUIRK_BROKEN_LOCAL_EXT_FTR_MAX_PAGE,
[🔨]         |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[🔨]   make[4]: *** [scripts/Makefile.build:252: drivers/bluetooth/hci_vhci.o] Error 1
[🔨]   make[3]: *** [scripts/Makefile.build:504: drivers/bluetooth] Error 2
[🔨]   make[2]: *** [scripts/Makefile.build:504: drivers] Error 2
[🔨]   make[1]: *** [Makefile:2023: .] Error 2
[🔨]   make: *** [Makefile:373: __build_one_by_one] Error 2
[💥] Error 2 occurred in main shell [ at /home/alistair/build/lib/functions/logging/runners.sh:190
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:190
        run_host_command_logged() --> lib/functions/logging/runners.sh:172
       run_kernel_make_internal() --> lib/functions/compilation/kernel-make.sh:52
   run_kernel_make_long_running() --> lib/functions/compilation/kernel-make.sh:65
      do_with_ccache_statistics() --> lib/functions/compilation/ccache.sh:30
                   kernel_build() --> lib/functions/compilation/kernel.sh:140
                  do_with_hooks() --> lib/functions/general/extensions.sh:445
                do_with_logging() --> lib/functions/logging/section-logging.sh:72
   kernel_prepare_build_and_package() --> lib/functions/compilation/kernel.sh:118
                 compile_kernel() --> lib/functions/compilation/kernel.sh:65
   artifact_kernel_build_from_sources() --> lib/functions/artifacts/artifact-kernel.sh:137
    artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:24
       obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:188
       build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:209
    main_default_build_packages() --> lib/functions/main/build-packages.sh:49
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:4
          do_with_default_build() --> lib/functions/main/default-build.sh:17
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:17
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:126
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:164
                           main() --> compile.sh:52
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-bcde583e-8484-483f-9108-49a5147d89b2.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-bcde583e-8484-483f-9108-49a5147d89b2.log.ans https://paste.next.armbian.com/log ]

```

The patching in my case, that working through is  **_failing on both main and master_** 

_This is a separate issue that I will have to solve_ and is unrelated to boot issues

igorpecovnik, 2023-03-09:
https://github.com/armbian/build/pull/4913#issuecomment-1461616773

AGM1968, 2023-03-10:
Thanks all concerned @ https://github.com/armbian/build/pull/4913#issuecomment-1461616773

main is building now - will do so more digging over the weekend

AGM1968, 2023-03-11:
I think original issues were possibly caused by directories not cleaning, despite not using `CREATE_PATCHES=yes ARTIFACT_IGNORE_CACHE=yes` when building both subsequent minimal & standard, hence my insistence that something went wrong with framework. I apologies if I was completely off in this regard.

So here is where we are today, I cannot upgrade on my user account.
Moreover the area that I'm working through has issues, 6.1 & 6.2.  in the regard 5.19.17 does not have these problems.

Noted on 6.1 & 6.2 kernel

```
Creating a new user account. Press <Ctrl-C> to abort

Please provide a username (eg. your first name): alistair
Create user (alistair) password: *******
Repeat user (alistair) password: *******
/usr/lib/armbian/armbian-firstlogin: line 271: cracklib-check: command not found

alistair@orangepizero2:~$ sudo shutdown
Command 'sudo' not found, but can be installed with:
sudo apt install sudo
sudo apt install sudo-ldap
alistair@orangepizero2:~$
```
Noted thermal failure on 6.1 & 6.2 kernels

```
[    1.224011] thermal_sys: Failed to find 'trips' node
[    1.224022] thermal_sys: Failed to find trip points for thermal-sensor id=0
[    1.224037] sun8i-thermal 5070400.thermal-sensor: Failed to register sensor 0 (-EINVAL)
[    1.224046] sun8i-thermal: probe of 5070400.thermal-sensor failed with error -22
```

Noted cpu backtrace on 6.2 kernels

```
[   19.647000] sysrq: Show backtrace of all active CPUs
[   19.651992] sysrq: CPU0:
[   19.652005] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C         6.2.2-sunxi64 #4
[   19.652016] Hardware name: OrangePi Zero2 (DT)
[   19.652022] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   19.652032] pc : arch_cpu_idle+0x18/0x2c
[   19.652052] lr : arch_cpu_idle+0x14/0x2c
[   19.652059] sp : ffff800009493d70
[   19.652064] x29: ffff800009493d70 x28: 00000000413e790c x27: ffff800009138258
[   19.652083] x26: ffff8000094a1940 x25: 0000000000000000 x24: 0000000000000000
[   19.652097] x23: ffff800009498aa0 x22: ffff8000092b1038 x21: ffff800009498a08
[   19.652112] x20: ffff800009498a98 x19: 0000000000000000 x18: 0000000000000000
[   19.652126] x17: 0000000000000000 x16: 0000000000000000 x15: 000000000000007a
[   19.652141] x14: 000000000000037b x13: 0000000000000000 x12: 0000000000000003
[   19.652155] x11: 0000000000000000 x10: 0000000000000950 x9 : ffff800009493cd0
[   19.652169] x8 : ffff8000094a22f0 x7 : 00000000000000c0 x6 : ffff8000092af798
[   19.652183] x5 : 4000000000000000 x4 : ffff800036ace000 x3 : 0000000000000000
[   19.652198] x2 : 0000000000014cc9 x1 : ffff00003fd7d798 x0 : 00000000000000e0
[   19.652213] Call trace:
[   19.652218]  arch_cpu_idle+0x18/0x2c
[   19.652228]  default_idle_call+0x30/0x6c
[   19.652241]  do_idle+0x23c/0x2b0
[   19.652253]  cpu_startup_entry+0x28/0x30
[   19.652261]  kernel_init+0x0/0x13c
[   19.652271]  arch_post_acpi_subsys_init+0x0/0x18
[   19.652281]  start_kernel+0x69c/0x6dc
[   19.652290]  __primary_switched+0xbc/0xc4
[   19.652419] sysrq: CPU1:
[   19.652432] Call trace:
[   19.652437]  dump_backtrace+0xd8/0x130
[   19.652452]  show_stack+0x18/0x30
[   19.652464]  showacpu+0x98/0x120
[   19.652476]  __flush_smp_call_function_queue+0xcc/0x254
[   19.652489]  generic_smp_call_function_single_interrupt+0x14/0x20
[   19.652499]  ipi_handler+0xa0/0xcc
[   19.652510]  handle_percpu_devid_irq+0x84/0x130
[   19.652523]  generic_handle_domain_irq+0x2c/0x4c
[   19.652531]  gic_handle_irq+0x5c/0x90
[   19.652544]  call_on_irq_stack+0x2c/0x5c
[   19.652554]  do_interrupt_handler+0x80/0x84
[   19.652564]  el1_interrupt+0x34/0x54
[   19.652572]  el1h_64_irq_handler+0x18/0x2c
[   19.652580]  el1h_64_irq+0x64/0x68
[   19.652588]  sprintf+0x10/0x90
[   19.652598]  info_print_prefix+0x88/0xdc
[   19.652610]  record_print_text+0x5c/0x160
[   19.652620]  syslog_print+0x1cc/0x30c
[   19.652631]  do_syslog.part.0+0x15c/0x3b0
[   19.652642]  do_syslog+0x44/0x60
[   19.652653]  kmsg_read+0x4c/0x70
[   19.652664]  proc_reg_read+0x30/0x100
[   19.652673]  vfs_read+0xc0/0x2a0
[   19.652684]  ksys_read+0x68/0xf4
[   19.652693]  __arm64_sys_read+0x1c/0x2c
[   19.652703]  invoke_syscall+0x48/0x114
[   19.652714]  el0_svc_common.constprop.0+0x44/0xfc
[   19.652724]  do_el0_svc+0x3c/0xc0
[   19.652734]  el0_svc+0x2c/0x84
[   19.652742]  el0t_64_sync_handler+0xbc/0x140
[   19.652750]  el0t_64_sync+0x190/0x194
[   19.652759] sysrq: CPU3:
[   19.652767] Call trace:
[   19.652771]  dump_backtrace+0xd8/0x130
[   19.652784]  show_stack+0x18/0x30
[   19.652796]  showacpu+0x98/0x120
[   19.652806]  __flush_smp_call_function_queue+0xcc/0x254
[   19.652816]  generic_smp_call_function_single_interrupt+0x14/0x20
[   19.652827]  ipi_handler+0xa0/0xcc
[   19.652836]  handle_percpu_devid_irq+0x84/0x130
[   19.652847]  generic_handle_domain_irq+0x2c/0x4c
[   19.652855]  gic_handle_irq+0x5c/0x90
[   19.652867]  call_on_irq_stack+0x2c/0x5c
[   19.652876]  do_interrupt_handler+0x80/0x84
[   19.652886]  el0_interrupt+0x4c/0xdc
[   19.652894]  __el0_irq_handler_common+0x18/0x24
[   19.652902]  el0t_64_irq_handler+0x10/0x20
[   19.652910]  el0t_64_irq+0x190/0x194
[   19.652919] sysrq: CPU2:
[   19.652928] Call trace:
[   19.652932]  dump_backtrace+0xd8/0x130
[   19.652946]  show_stack+0x18/0x30
[   19.652957]  showacpu+0x98/0x120
[   19.652967]  __flush_smp_call_function_queue+0xcc/0x254
[   19.652978]  generic_smp_call_function_single_interrupt+0x14/0x20
[   19.652988]  ipi_handler+0xa0/0xcc
[   19.652998]  handle_percpu_devid_irq+0x84/0x130
[   19.653008]  generic_handle_domain_irq+0x2c/0x4c
[   19.653016]  gic_handle_irq+0x5c/0x90
[   19.653028]  call_on_irq_stack+0x2c/0x5c
[   19.653037]  do_interrupt_handler+0x80/0x84
[   19.653047]  el0_interrupt+0x4c/0xdc
[   19.653055]  __el0_irq_handler_common+0x18/0x24
[   19.653063]  el0t_64_irq_handler+0x10/0x20
[   19.653071]  el0t_64_irq+0x190/0x194
``