armbian/build #4905: kernel configure screen skipped

SteeManMI, 2023-03-05:
### What happened?

First reported in the forums: https://forum.armbian.com/topic/27005-artifact-cache-supresses-kernel_configure
I can reproduce the issue.  If you already have built the kernel so it is cached, then you re-run configure.sh and this time say yes to modify the kernel configuration, that screen never shows, and the build exits not doing anything after finding existing artifacts in the cache.

### How to reproduce?

Build a kernel so the artifacts are chached, then rerun, this time selecting to change the kernel configuration, the build exits without showing the configuration screen.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
~/build$ ./compile.sh BOARD=odroidhc4 BRANCH=current BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,img
[🌿] Applying cmdline param [ 'BUILD_MINIMAL': '(empty)' --> 'no' early ]
[🌿] Applying cmdline param [ 'COMPRESS_OUTPUTIMAGE': '(empty)' --> 'sha,img' early ]
[🌿] Applying cmdline param [ 'BUILD_DESKTOP': '(empty)' --> 'no' early ]
[🌿] Applying cmdline param [ 'BRANCH': '(empty)' --> 'current' early ]
[🌿] Applying cmdline param [ 'BOARD': '(empty)' --> 'odroidhc4' early ]
[🌿] Applying cmdline param [ 'KERNEL_ONLY': '(empty)' --> 'yes' early ]
[🌿] Applying cmdline param [ 'KERNEL_CONFIGURE': '(empty)' --> 'yes' early ]
[🚸] This script requires root privileges; Docker is unavailable [ trying to use sudo ]
[🌿] Applying cmdline param [ 'PREFER_DOCKER': '(empty)' --> 'no' early ]
[🌿] Applying cmdline param [ 'BUILD_MINIMAL': '(empty)' --> 'no' early ]
[🌿] Applying cmdline param [ 'COMPRESS_OUTPUTIMAGE': '(empty)' --> 'sha,img' early ]
[🌿] Applying cmdline param [ 'BUILD_DESKTOP': '(empty)' --> 'no' early ]
[🌿] Applying cmdline param [ 'ARMBIAN_RELAUNCHED': '(empty)' --> 'yes' early ]
[🌿] Applying cmdline param [ 'BRANCH': '(empty)' --> 'current' early ]
[🌿] Applying cmdline param [ 'BOARD': '(empty)' --> 'odroidhc4' early ]
[🌿] Applying cmdline param [ 'SET_OWNER_TO_UID': '(empty)' --> '1000' early ]
[🌿] Applying cmdline param [ 'KERNEL_ONLY': '(empty)' --> 'yes' early ]
[🌿] Applying cmdline param [ 'KERNEL_CONFIGURE': '(empty)' --> 'yes' early ]
[🌱] Starting single build process [ odroidhc4 ]
[🌱] Checking [ basic host setup ]
[🌱] Build host OS release [ jammy ]
[🌱] Build host architecture [ arm64 ]
[🌱] Sourcing board configuration [ /home/steve/build/config/boards/odroidhc4.conf ]
[🌱] Starting main configuration
[🌱] Sourcing family configuration [ /home/steve/build/config/sources/families/meson-sm1.conf ]
[🌱] Sourcing arch configuration [ arm64.conf ]
[🌱] Extension manager [ processed 4 Extension Methods calls and 4 Extension Method implementations ]
[🌱] Configuration prepared for BOARD build [ odroidhc4.conf ]
[✨] Repeat Build Options (early) [ ./compile.sh BOARD=odroidhc4 BRANCH=current BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,img ]
[🌱] Checked directory OK for mount options [ /home/steve/build/.tmp ('main temporary dir') ]
[🌱] Preparing [ host ]
[🌱] Syncing clock [ host ]
[🔨]    4 Mar 18:52:16 ntpdate[41836]: adjust time server 38.229.56.9 offset -0.001440 sec
[🌱] artifact [ uboot :: uboot() ]
[🌱] Using cached [ GIT_INFO_UBOOT ]
[💖] artifact [ present in local cache: uboot-odroidhc4-current 2023.01-S62e2-P1cd3-B680b ]
[🌱] artifact [ kernel :: kernel() ]
[🌱] Using cached [ GIT_INFO_KERNEL ]
[🌱] Using kernel config file [ config/kernel/linux-meson64-current.config ]
[💖] artifact [ present in local cache: kernel-meson64-current 6.1.15-S4261-De511-Pb8ce-C072bHfe66-Bb436 ]
[🌱] artifact [ firmware :: firmware() ]
[🌱] Using cached [ GIT_INFO_ARMBIAN_FIRMWARE ]
[💖] artifact [ present in local cache: armbian-firmware 1-SA56ad-B6be6 ]
[🌱] Building deb [ armbian-plymouth-theme ]
[🌱] shellcheck found no problems in package scripts [ shellchecked 3 scripts in 'armbian-plymouth-theme' ]
[🔨]   dpkg-deb: building package 'armbian-plymouth-theme' in '/home/steve/build/.tmp/work-5237001b-8933-4274-b02b-98b6ec67e71c/tmp.exL1KIf5bj/armbian-plymouth-theme_23.05.0-trunk_all.deb'.
[🚸] using legacy option [ KERNEL_ONLY=yes; stopping build mid-packages ]
[🌱] Runtime [ 0 min ]
[✨] Repeat Build Options [ ./compile.sh BOARD=odroidhc4 BRANCH=current BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,img ]
[🌱] Cleaning up [ please wait for cleanups to finish ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-5237001b-8933-4274-b02b-98b6ec67e71c.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-5237001b-8933-4274-b02b-98b6ec67e71c.log.ans https://paste.next.armbian.com/log ]
steve@odroidhc4:~/build$
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-03-05:
Confirm, when doing kernel config _during image build_ -- it can be worked-around by running `./compile.sh` with command `kernel-config`, for now. I will try to come up with a way to make this work during image build, but inconsistencies arise, so fix is more complex than originally thought

rpardini, 2023-03-17:
Should be addressed in https://github.com/armbian/build/pull/4955
KERNEL_CONFIGURE=yes now causes `C999999` hash (always), some warnings (when used during image build, with instructions on how to do it "properly") but should work again. 
Could you confirm @SteeManMI  ?

EvilOlaf, 2023-03-17:
Works for me.

rpardini, 2023-03-17:
> Works for me.

Great, any suggestions on the wording of the warning produced? I tried to be terse. Could be we have a Wiki/docs somewhere and instead just like there?

EvilOlaf, 2023-03-17:
Documentation needs adjustments for next, yes but that is a whole different story.
I think in terms of wording it is fine. However maybe add a small delay to annoy so they use the new parameter instead ;)

iav, 2023-03-17:
Works for me.  Thank you.

AGM1968, 2023-03-17:
Good this side - Thanks

SteeManMI, 2023-03-17:
It works, but....
The first time I ran it, I thought it wasn't working because I didn't see any message, but the config edit dialog popped up and the image built successfully.
Then I ran it again and killed the run, and then looked at the log file and found the message:
KERNEL_CONFIGURE=yes during image build is deprecated. [ It still works, but please prefer the new way. First, run './compile.sh BOARD=aml-s9xx-box BRANCH=current kernel-config'; then commit your changes; then build the image as normal. This workflow ensures consistent hashing results. ]
So unless you force the user to acknowledge the message, it isn't providing any value, as no one will notice it.
So I would recommend either a hard wait for input to continue after the message, or a long pause before continuing.

As for the message itself, I think this one sentence either needs more explanation or should be eliminated: "This workflow ensures consistent hashing results"  This leads one to speculate all sorts of bad thoughts: does this mean if I continue I'll have inconsistent hashing results?  That sounds bad, will it corrupt my build, etc, etc, etc.
Either explain in more detail what that means, or drop it from the message.

rpardini, 2023-03-17:
> does this mean if I continue I'll have inconsistent hashing results

Yes. That's why I force `C999999` -- to signal that your hashing is inconsistent.

rpardini, 2023-03-17:
> So unless you force the user to acknowledge the message, it isn't providing any value, as no one will notice it.

So might be a countdown-to-continue after that message is in order. Noted. Thanks.

rpardini, 2023-03-25:
Yeah, let's close this. I've rethought this and I think the best is to re-show the warning (or error) messages in a single list at the end of build, so user can't miss it.

closed 2023-03-25 by rpardini