armbian/build #4907: COMPRESS_OUTPUTIMAGE=gz not work anymore

iav, 2023-03-05:
### What happened?

Set COMPRESS_OUTPUTIMAGE=gz not produces gzipped image anymore

### How to reproduce?

```
./compile.sh BOARD=helios64 BRANCH=edge RELEASE=bookworm BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=gz
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-03-07:
True. Also, 7-zip....

rpardini, 2023-05-21:
Under the impression we've discussed this under a PR, and gave up on it....

closed 2023-05-21 by rpardini

spitfire, 2024-07-02:
Why not remove it from the [docs](https://docs.armbian.com/Developer-Guide_Build-Options/) then? I've spent some time wondering why this doesn't work..

rpardini, 2024-07-02:
@spitfire Why not do it yourself? https://github.com/armbian/documentation/

spitfire, 2024-07-02:
I'm just asking the question. I'm not a maintainer, today was the first time I've touched it, so I have no idea whether that's a good idea or not.

EvilOlaf, 2024-07-03:
There isn't much that could go wrong when sending a PR to docs. It needs review anyways and if there is something wrong with the submitted code reviewer will let you know.