armbian/build #4909: Bug Hunting: Seems that `armbian-hardware-optimization` is causing kernel panic on sun50i* and sunxi64*

Kreyren, 2023-03-06:
### What happened?

```
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Successfully called chroot().
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Successfully dropped remaining capabilities.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: No service file found in /etc/avahi/services.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: New relevant interface lo.IPv6 for mDNS.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: New relevant interface lo.IPv4 for mDNS.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Network interface enumeration completed.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Registering new address record for ::1 on lo.*.
Mar 06 03:22:36 tsvetan avahi-daemon[724]: Registering new address record for 127.0.0.1 on lo.IPv4.
-> Mar 06 03:22:36 tsvetan CRON[847]: (root) CMD (/usr/lib/armbian/armbian-apt-updates)
-> Mar 06 03:22:37 tsvetan CRON[849]: (root) CMD (/usr/lib/armbian/armbian-truncate-logs)
-> Mar 06 03:22:37 tsvetan armbian-hardware-optimization[671]: /usr/lib/armbian/armbian-hardware-optimization: line 380: /proc/irq//smp_affinity: No such>
-> Mar 06 03:22:37 tsvetan armbian-hardware-optimization[671]: /usr/lib/armbian/armbian-hardware-optimization: line 381: /sys/class/net/eth0/queues/rx-0/>
Mar 06 03:22:37 tsvetan systemd[1]: Starting NetworkManager.service - Network Manager...
Mar 06 03:22:37 tsvetan systemd[1]: Starting wpa_supplicant.service - WPA supplicant...
Mar 06 03:22:37 tsvetan systemd[1]: Started avahi-daemon.service - Avahi mDNS/DNS-SD Stack.
Mar 06 03:22:37 tsvetan udisksd[768]: failed to load module crypto: libbd_crypto.so.2: cannot open shared object file: No such file or directory
Mar 06 03:22:37 tsvetan udisksd[768]: failed to load module mdraid: libbd_mdraid.so.2: cannot open shared object file: No such file or directory
```

### How to reproduce?

Command you gave to the framework!

### Branch

other

### On which host OS are you observing this problem?

Other

### Relevant log output

```shell
ss
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-03-06:
This is build framework issue tracker. I am sure with your technical background is able to differentiate this:
[image]
Stop hijacking, stop bending the rules and start learning. Project is about people first, solving bugs comes at the end. And we will not solve bugs you will report. That is your job.

closed 2023-03-06 by igorpecovnik

Kreyren, 2023-03-06:
@igorpecovnik i dunno what the issue is yet to know where to place it, the original hypothesis is that it was caused by the optimization scripts introduced by the framework which in my logic sounds that this is the appropriate tracker then?

igorpecovnik, 2023-03-06:
Use this project bug hunting schema as it was designed. Its not perfect, it will change some day, but until then ... read it and respect. Stop hacking relationships and communications. 

> my logic sounds that this is the appropriate tracker then?

Common (sense) logic only.

Limitations are here also to help preventing you / users from abusing maintainers and developers.