armbian/build #4916: Failed to fetch SHA1 of commit

jmrsnt, 2023-03-07:
### What happened?

`Failed to fetch SHA1 of 'https://github.com/ayufan-rock64/linux-kernel' 'tag' '4.4.202-1237-rockchip-ayufan^{}'`

It seems that the error occurs due to a change in the name of the TAG, in `git-ref2info.sh:12`

```
# Small detour here; if it's a tag, ask for the dereferenced commit, to avoid the annotated tag's sha1, instead get the commit tag points to.
if [[ "${ref_type}" == "tag" ]]; then
    ref_name="${ref_name}^{}" # dereference tag
fi
```

When running `git ls-remote --exit-code https://github.com/ayufan-rock64/linux-kernel 4.4.202-1237-rockchip-ayufan^{} | cut -f1` it returns empty, but if you run `git ls-remote --exit-code https://github.com/ayufan-rock64/linux-kernel 4.4.202-1237-rockchip-ayufan | cut -f1` without `^{}` it returns the SHA1

### How to reproduce?

./compile.sh BOARD=rockpi-4a BRANCH=legacy RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_ONLY=no KERNEL_CONFIGURE=yes DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base DESKTOP_APPGROUPS_SELECTED="internet multimedia" COMPRESS_OUTPUTIMAGE=sha,img

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Relevant log output

```shell
# Armbian ANSI build logs for 19521a25-e1e9-4eee-a826-c7861b828811 - use "less -SR" to view
# Armbian build at Tue Mar  7 23:40:06 UTC 2023 on 88e748d12b5a
------------------------------------------------------------------------------------------------------------
# ARGs: 'PREFER_DOCKER=no' 'BUILD_MINIMAL=no' 'COMPRESS_OUTPUTIMAGE=sha,img' 'DESKTOP_APPGROUPS_SELECTED=internet multimedia' 'BUILD_DESKTOP=yes' 'ARMBIAN_BUILD_UUID=19521a25-e1e9-4eee-a826-c7861b828811' 'ARMBIAN_RELAUNCHED=yes' 'BRANCH=legacy' 'DESKTOP_ENVIRONMENT=xfce' 'DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base' 'BOARD=rockpi-4a' 'SKIP_LOG_ARCHIVE=yes' 'SET_OWNER_TO_UID=1000' 'KERNEL_ONLY=no' 'RELEASE=bullseye' 'KERNEL_CONFIGURE=yes' 'build'
------------------------------------------------------------------------------------------------------------
# GIT revision
   commit e91e27cc83136e9d8f25994773809185d46892d7 (grafted, HEAD -> main, origin/main, origin/HEAD)
   Author: Paolo <paolo.sabatino@gmail.com>

       rockchip: remove mainlined patch (#4908)

# GIT status
   On branch main
   Your branch is up to date with 'origin/main'.

   nothing to commit, working tree clean
### config_early_init.log
--> (0) INFO: Starting single build process [ rockpi-4a ]
--> (0) INFO: Checking [ basic host setup ]
--> (0) INFO: Build host OS release [ jammy ]
--> (0) INFO: Build host architecture [ amd64 ]
--> (0) INFO: Detected WSL2 - experimental support [ Windows Subsystem for Linux 2 ]
--> (0) WARNING: Please use a terminal that supports UTF-8. For example: [ Windows Terminal ]
--> (0) WARNING: Get it at the Microsoft Store [ https://apps.microsoft.com/store/detail/windows-terminal/9N0DX20HK701 ]
--> (0) ERROR: Problem detected [ WSL2 Terminal does not support UTF-8 ]
--> (0) ERROR: Exiting in 10 seconds [ Press <Ctrl-C> to abort, <Enter> to ignore and continue ]
--> (2) WARNING: User pressed ENTER, continuing, albeit [ WSL2 Terminal does not support UTF-8 ]
------------------------------------------------------------------------------------------------------------
### config_source_board_file.log
--> (2) INFO: Sourcing board configuration [ /armbian/config/boards/rockpi-4a.conf ]
------------------------------------------------------------------------------------------------------------
### do_main_configuration.log
--> (2) INFO: Starting main configuration
--> (2) INFO: Sourcing family configuration [ /armbian/config/sources/families/rockchip64.conf ]
--> (2) INFO: Enabling extension [ rkbin-tools ]
--> (2) INFO: Determined BOOT_SOC from BOOTCONFIG [ BOOT_SOC: 'rk3399'; BOOTCONFIG: 'rock-pi-4-rk3399_defconfig' ]
--> (2) INFO: Sourcing arch configuration [ arm64.conf ]
--> (2) INFO: Extension manager [ processed 4 Extension Methods calls and 4 Extension Method implementations ]
------------------------------------------------------------------------------------------------------------
### config_post_main.log
--> (2) INFO: Configuration prepared for BOARD build [ rockpi-4a.conf ]
--> (2) EXT: Repeat Build Options (early) [ ./compile.sh BOARD=rockpi-4a BRANCH=legacy RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_ONLY=no KERNEL_CONFIGURE=yes DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base DESKTOP_APPGROUPS_SELECTED="internet multimedia" COMPRESS_OUTPUTIMAGE=sha,img ]
------------------------------------------------------------------------------------------------------------
### check_dir_for_mount_options.log
--> (2) INFO: Checked directory OK for mount options [ /armbian/.tmp ('main temporary dir') ]
------------------------------------------------------------------------------------------------------------
### prepare_host_noninteractive.log
--> (2) INFO: Preparing [ host ]
--> (2) INFO: Running in container [ Adding provisions for container building ]
--> (3) COMMAND: update-ccache-symlinks
--> (3) INFO: Ignoring toolchains [ SKIP_EXTERNAL_TOOLCHAINS: yes ]
------------------------------------------------------------------------------------------------------------
### aggregate_packages.log
--> (3) INFO: Aggregating packages [ rootfs ]
--> (3) INFO: Python3 version [ 3.10.6 - 'Python 3.10.6' ]
--> (3) INFO: pip3 version [ 22.0.2: 'pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)' ]
--> (3) INFO: Using cached pip packages for Python tools [ c40672ecb319ec9a04b3471e3c6da8d006211683c969a19c9412cdb42c737c51 ]
--> (3) COMMAND: [...shortened...] /usr/bin/python3 /armbian/lib/tools/aggregation.py
   Summary: Aggregation: debootstrap: 6; rootfs: 112; image: 1; desktop: 129; apt-sources: 0; hash: d5e1d639bd63ef3c
--> (3) COMMAND: rm -f /armbian/.tmp/work-19521a25-e1e9-4eee-a826-c7861b828811/tmp.azfvnZcIzo
------------------------------------------------------------------------------------------------------------
### clean_deprecated_mountpoints.log
--> (3) INFO: artifact [ uboot :: uboot() ]
------------------------------------------------------------------------------------------------------------
### artifact_prepare_version.log
--> (3) INFO: Using cached [ GIT_INFO_UBOOT ]
------------------------------------------------------------------------------------------------------------
### unpack_artifact_from_local_cache.log
--> (3) CACHEHIT: artifact [ present in local cache: uboot-rockpi-4a-legacy 2022.07-Se092-Pbb0c-B680b ]
--> (3) INFO: artifact [ kernel :: kernel() ]
------------------------------------------------------------------------------------------------------------
### artifact_prepare_version.log
--> (3) INFO: Producing new & caching [ GIT_INFO_KERNEL ]
--> (3) INFO: Fetching SHA1 of tag 4.4.202-1237-rockchip-ayufan^{} [ https://github.com/ayufan-rock64/linux-kernel ]
--> (4) INFO: SHA1 of tag 4.4.202-1237-rockchip-ayufan^{} [ '' ]
--> (4) ERROR: error! [ Failed to fetch SHA1 of 'https://github.com/ayufan-rock64/linux-kernel' 'tag' '4.4.202-1237-rockchip-ayufan^{}' - make sure it's correct  ]
--> (4) ERROR: Exiting with error 43 [ at /armbian/lib/functions/logging/traps.sh:1
                   exit_with_error() --> lib/functions/logging/traps.sh:1
          memoized_git_ref_to_info() --> lib/functions/general/git-ref2info.sh:30
                      run_memoized() --> lib/functions/general/memoize-cached.sh:52
      artifact_kernel_prepare_version() --> lib/functions/artifacts/artifact-kernel.sh:52
          artifact_prepare_version() --> lib/functions/artifacts/artifacts-obtain.sh:24
                   do_with_logging() --> lib/functions/logging/section-logging.sh:72
          obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:76
          build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:209
       main_default_build_packages() --> lib/functions/main/build-packages.sh:49
      full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:4
             do_with_default_build() --> lib/functions/main/default-build.sh:17
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:17
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:126
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:164
                              main() --> compile.sh:52
    ]
--> (4) ERROR: Cleaning up [ please wait for cleanups to finish ]
------------------------------------------------------------------------------------------------------------
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-03-08:
Hmm, very interesting. Supposedly git should not behave differently if dereference is undeeded.
Have you tried this on something else than WSL2/Jammy? I wonder if that might be a factor.

rpardini, 2023-03-08:
> Have you tried this on something else than WSL2/Jammy? I wonder if that might be a factor.

Scratch that. Easy to reproduce anywhere: `./compile.sh BOARD=rockpi-4a BRANCH=legacy kernel`

But, the problem is not the dereference, instead, the specified tag `` does not exist in that repo.

Eg: https://github.com/ayufan-rock64/linux-kernel/tree/release-4.4.202 exists OK.
But https://github.com/ayufan-rock64/linux-kernel/tree/4.4.202-1237-rockchip-ayufan does not. 

So problem is with board/family config, that needs love & maintaining, and not with tag -> SHA1 dereferencing.

I would "guess" that the corrected tag is `release-4.4.202` to be set in https://github.com/armbian/build/blob/main/config/sources/families/rockchip64.conf#L9 -- but I'm no expert.

Who knows ayufan stuff, @igorpecovnik  / @Tonymac32  ?

jmrsnt, 2023-03-09:
@rpardini Thanks for your support. Updating https://github.com/armbian/build/blob/main/config/sources/families/rockchip64.conf#L9 with `branch:release-4.4.202` worked.

@igorpecovnik We use Rock PI 4 devices with Rockchip RK3399 processor, We need to build a legacy buster image to use hardware acceleration (https://forum.armbian.com/topic/16516-rk3399-legacy-multimedia-framework/), for 4k media, unfortunately I haven't found any way to use hardware acceleration in newer kernel versions for this model. And the legacy buster image was the only one that worked without any problems, we need to customize the image, for that we need this Kernel version.

Guys, we managed to move to the next step, but unfortunately it was not possible to compile the Kernel, could you help us, please? I understand that now it is about another subject, if necessary I can open a new issue, to not mix the topics.

Any chance it could be something related to the snippet below?

```
net/bluetooth/hci_core.c: In function 'hci_setup_link_policy':
   net/bluetooth/hci_core.c:687:49: error: 'HCI_QUIRK_BROKEN_PARK_LINK_STATUS' undeclared (first use in this function); did you mean 'HCI_QUIRK_BROKEN_STORED_LINK_KEY'?
     687 |         if (lmp_park_capable(hdev) && !test_bit(HCI_QUIRK_BROKEN_PARK_LINK_STATUS, &hdev->quirks))
         |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         |                                                 HCI_QUIRK_BROKEN_STORED_LINK_KEY
```

```
drivers/bluetooth/hci_ldisc.c: In function 'hci_uart_register_dev':
   drivers/bluetooth/hci_ldisc.c:626:25: error: 'HCI_QUIRK_BROKEN_PARK_LINK_STATUS' undeclared (first use in this function); did you mean 'HCI_QUIRK_BROKEN_STORED_LINK_KEY'?
     626 |                 set_bit(HCI_QUIRK_BROKEN_PARK_LINK_STATUS, &hdev->quirks);
         |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         |                         HCI_QUIRK_BROKEN_STORED_LINK_KEY
```
Full Log:

```
# Armbian build at Wed Mar  8 21:55:57 UTC 2023 on b0dda7fe8688
------------------------------------------------------------------------------------------------------------
# ARGs: 'PREFER_DOCKER=no' 'BUILD_MINIMAL=no' 'COMPRESS_OUTPUTIMAGE=sha,img' 'DESKTOP_APPGROUPS_SELECTED=multimedia browsers' 'BUILD_DESKTOP=yes' 'ARMBIAN_BUILD_UUID=10a2c40f-6b8b-4207-8b0f-77d3c7af99dd' 'ARMBIAN_RELAUNCHED=yes' 'BRANCH=legacy' 'DESKTOP_ENVIRONMENT=xfce' 'DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base' 'BOARD=rockpi-4a' 'SKIP_LOG_ARCHIVE=yes' 'SET_OWNER_TO_UID=1000' 'KERNEL_ONLY=no' 'RELEASE=bullseye' 'KERNEL_CONFIGURE=no' 'build'
------------------------------------------------------------------------------------------------------------
# GIT revision
   commit e91e27cc83136e9d8f25994773809185d46892d7 (grafted, HEAD -> main, origin/main, origin/HEAD)
   Author: Paolo <paolo.sabatino@gmail.com>

       rockchip: remove mainlined patch (#4908)

# GIT status
   On branch main
   Your branch is up to date with 'origin/main'.

   Changes not staged for commit:
     (use "git add <file>..." to update what will be committed)
     (use "git restore <file>..." to discard changes in working directory)
        modified:   config/kernel/linux-rockchip64-legacy.config
        modified:   config/sources/families/rockchip64.conf

   Untracked files:
     (use "git add <file>..." to include in what will be committed)
        config/kernel/linux-rockchip64-legacy.config.defconfig

   no changes added to commit (use "git add" and/or "git commit -a")
### config_early_init.log
--> (0) INFO: Starting single build process [ rockpi-4a ]
--> (0) INFO: Checking [ basic host setup ]
--> (0) INFO: Build host OS release [ jammy ]
--> (0) INFO: Build host architecture [ amd64 ]
--> (0) INFO: Detected WSL2 - experimental support [ Windows Subsystem for Linux 2 ]
--> (0) WARNING: Please use a terminal that supports UTF-8. For example: [ Windows Terminal ]
--> (0) WARNING: Get it at the Microsoft Store [ https://apps.microsoft.com/store/detail/windows-terminal/9N0DX20HK701 ]
--> (0) ERROR: Problem detected [ WSL2 Terminal does not support UTF-8 ]
--> (0) ERROR: Exiting in 10 seconds [ Press <Ctrl-C> to abort, <Enter> to ignore and continue ]
--> (0) WARNING: User pressed ENTER, continuing, albeit [ WSL2 Terminal does not support UTF-8 ]
------------------------------------------------------------------------------------------------------------
### config_source_board_file.log
--> (1) INFO: Sourcing board configuration [ /armbian/config/boards/rockpi-4a.conf ]
------------------------------------------------------------------------------------------------------------
### do_main_configuration.log
--> (1) INFO: Starting main configuration
--> (1) INFO: Sourcing family configuration [ /armbian/config/sources/families/rockchip64.conf ]
--> (1) INFO: Enabling extension [ rkbin-tools ]
[... 2449 lines cut ...]
   -->--> command failed with error code 2 after 376 seconds
--> (617) WARNING: stacktrace for failed command [ exit code 2:/bin/bash -e -o pipefail -c env -i CCACHE_BASEDIR="/armbian/cache/sources/linux-kernel-worktree/4.4__rockchip64__arm64" CCACHE_TEMPDIR="/armbian/.tmp/work-10a2c40f-6b8b-4207-8b0f-77d3c7af99dd/ccache_tmp" PATH="/usr/bin:/usr/lib/ccache:/armbian/.tmp/work-10a2c40f-6b8b-4207-8b0f-77d3c7af99dd/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" DPKG_COLORS=always XZ_OPT='--threads=0' TERM='xterm-256color' CCACHE_DIR='/armbian/cache/ccache' make '-j12' 'ARCH=arm64' [... cut ...]
             do_with_default_build() --> lib/functions/main/default-build.sh:17
            cli_standard_build_run() --> lib/functions/cli/cli-build.sh:17
           armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:126
                    cli_entrypoint() --> lib/functions/cli/entrypoint.sh:164
                              main() --> compile.sh:52 ]
--> (617) ERROR: Cleaning up [ please wait for cleanups to finish ]
------------------------------------------------------------------------------------------------------------
```

rpardini, 2023-03-09:
Yep, that `HCI_QUIRK_BROKEN_PARK_LINK_STATUS` has come up elsewhere -- it's a bluetooth hack that's applied by the general drivers that conflicts with some patch. Since this is a very, very ancient kernel, no idea how to go...

(btw: the SHA1 deference problem is confirmed, I actually think its a Github-specific server bug, I will implement a workaround)

igorpecovnik, 2023-03-15:
>  I understand that now it is about another subject, if necessary 

Closing as resolved.

> I can open a new issue, to not mix the topics.

You can, but this will not be addressed as nobody will be dealing with this kernel in any way. You can try or sponsor moving to latest Rockchip kernel. We don't need this to function, you do. Our primary target of maintainace is build framework.

closed 2023-03-15 by igorpecovnik

rpardini, 2023-03-15:
The actual tag dereferencing problem needs a new issue, then. Otherwise I will forget...

igorpecovnik, 2023-03-15:
Yes @jmrsnt If you are in a hurry https://armbian.atlassian.net/browse/AR-1610