armbian/build #4933: Trying to build for aml tv box fails on WifI drivers (i think)

gigobait, 2023-03-10:
### What happened?

fails to compile 
log https://paste.next.armbian.com/apamorekib

### How to reproduce?

./compile.sh  EXPERT=yes BOARD=aml-s9xx-box  BRANCH=current RELEASE=lunar BUILD_MINIMAL=no BUILD_DESKTOP=no KERNER_CONFIG=yes KERNEL_ONLY=no

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-03-11:
`./compile.sh BOARD=aml-s9xx-box BRANCH=current RELEASE=lunar BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,img` works. Even hits cache. So you either have a dirty build directory or the patches you applied make it fail.

rpardini, 2023-03-11:
@gigobait log shows you've a _lot_ of changes to the patches. You'll have to handle those yourself, or, send a PR with the relevant changes so we can see what they are.

Nothing to do, closing

closed 2023-03-11 by rpardini