armbian/build #4935: Minimal builds are missing apt-utils package

AGM1968, 2023-03-11:
### What happened?

ISSUE:

Minimal builds are missing apt-utils package

$RELEASE/main/packages		 |minimal|

\build\config\cli\common\main\packages

apt-utils is not installed

ERROR: logs

--> (1425) COMMAND: DEBIAN_FRONTEND=noninteractive apt-get -y -qq -o Dpkg::Use-Pty=0 -o APT::Get::List-Cleanup=0 -o APT::Clean-Installed=0 --no-install-recommends install /root/linux-u-boot-orangepizero2-edge_2022.07-Se092-P12e0-B6324_arm64.deb
   debconf: delaying package configuration, since apt-utils is not installed
   Selecting previously unselected package linux-u-boot-orangepizero2-edge.
   (Reading database ... 19305 files and directories currently installed.)
   Preparing to unpack .../linux-u-boot-orangepizero2-edge_2022.07-Se092-P12e0-B6324_arm64.deb ...
   Unpacking linux-u-boot-orangepizero2-edge (2022.07-Se092-P12e0-B6324) ...
   Setting up linux-u-boot-orangepizero2-edge (2022.07-Se092-P12e0-B6324) ...
   
   
PR PROPOSAL:

add apt-utils to \build\config\cli\common\main\packages

REFERENCE:

Clenup minimal and server lists (#4893)

commit sha: e71d8dba103893e662e60fdf675b90ca78529888

https://github.com/AGM1968/build/blob/e71d8dba103893e662e60fdf675b90ca78529888/config/cli/common/main/packages

### How to reproduce?

### config_post_main.log 
--> (19) INFO: Configuration prepared for BOARD build [ orangepizero2.conf ]
--> (19) EXT: Repeat Build Options (early) [ ./compile.sh BOARD=orangepizero2 BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,img ]

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-03-11:
Go. And add this too: 
https://armbian.atlassian.net/browse/AR-1590

rpardini, 2023-03-11:
hmm. `sysstat` has some heavy dependencies. `apt-utils` is also very optional (generates warnings but should work without it).

AGM1968, 2023-03-11:
apt-utils  not detrimental to build

igorpecovnik, 2023-03-11:
> hmm. sysstat has some heavy dependencies.

Used by armbianmonitor ... then this has to be adjusted.

rpardini, 2023-03-11:
Oh, okie then, lets add

closed 2023-03-13 by AGM1968