armbian/build #4951: Problem found ... build and ./compile.sh: error in lib/functions/general/oci-oras.sh line 150

agl7, 2023-03-16:
### What happened?

If I use "git clone" to clone armbian/build and then start the build process with "./compile.sh SHOW_DEBUG=yes SHARE_LOG=yes" there is an error in line 150 in the script "lib/functions/general/oci-oras.sh".

Best Regards
Andreas (nick at Discord/IRC: agl)

### How to reproduce?

Command you gave to the framework!

./compile.sh SHOW_DEBUG=yes SHARE_LOG=yes

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
https://paste.armbian.com/bocaxaxomo
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AGM1968, 2023-03-16:
It's known, try build again. There's a reference here somewhere.

rpardini, 2023-03-16:
Duplicate of https://github.com/armbian/build/issues/4889

closed 2023-03-16 by rpardini