armbian/build #4958: Build doesn't stop/error on patch failures

SteeManMI, 2023-03-17:
### What happened?

I thought that a feature/benefit of the new build framework was that it would no longer quietly continue when errors occurred.
In testing other changes today, I noticed in my logs the following patches were failing to apply.  Shouldn't this have resulted in a failure of the build?  Instead the build completed successfully and I had no idea there were these issues.

| ❌`[failed_apply]`  | `board-bananapim5-002-switch-VDDIO_C-pin-to-OP` | `(+1/-1)[1M]` | `meson-sm1-bananapi-m5.dts` | `Christian Hewitt` _arm6
4: dts: meson: bananapi-m5: switch VDDIO_C pin to OPEN_DRAIN_ |
| ❌`[failed_apply]`  | `board-bananapim5-003-convert-dts-to-dtsi` | `(+436/-426)[1M, 1A]` | `meson-sm1-bananapi.dtsi`, `meson-sm1-bananapi-m5.d
ts` | `Christian Hewitt` _arm64: dts: meson: bananapi-m5: convert dts to dtsi_ |
| ❌`[failed_apply]`  | `board-radxa-zero-dts-otg-fix` | `(+0/-1)[1M]` | `meson-g12a-radxa-zero.dts` | `Stephen` _arm64: dts: radxa zero: set dr
_mode of usb node to otg_ |
| ❌`[failed_apply]`  | `general-arm64-dts-amlogic-Make-mmc-host-controller-interrupt` | `(+8/-8)[3M]` | `meson-g12-common.dtsi`, `meson-gx.dtsi
`, `meson-axg.dtsi` | `Heiner Kallweit` _arm64: dts: amlogic: Make mmc host controller interrupts level-sensitive_ |
| ❌`[failed_apply]`  | `general-meson-mmc-1-arm64-amlogic-mmc-meson-gx-Add-core-tx-rx-eMMC-SD-SD` | `(+48/-6)[1M, 1A]` | `meson-gx-mmc.h`, `mes
on-gx-mmc.c` | `Vyacheslav Bocharov` _arm64: amlogic: mmc: meson-gx: Add core, tx, rx eMMC/SD/SDIO phase clock settings from devicetree data_ |

### How to reproduce?

./compile.sh BOARD=aml-s9xx-box BRANCH=current RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,img

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-03-17:
The previous master actually went out of its way to still try to build when patching failed, so I tried to keep.
It doesnt make any sense, and patches might be half-applied, leading to chaos.
If all in favor I'd change to stop immediately on patching error.

SteeManMI, 2023-03-17:
I agree that continue on error doesn't make any sense, and would support this change.  I would propose we pick a date in the future about a month from now, to make the change.  And open issues for the patches that are currently failing to apply, so that they can be fixed between now and then.  Perhaps this is a topic for a future weekly developer meeting.

PR #5308 (2023-06-13): patching: Patching Summary and Patching Failures tables using Python's Rich + auto patchers: bare dts, bare overlays, DT Makefile
#### patching: Patching Summary and Patching Failures tables using Python's Rich + auto patchers: bare dts, bare overlays, DT Makefile

- patching: Patching Summary and Patching Failures tables using Python's Rich
  - even Rich'er patch output by colorizing certain strings green/yellow/red
  - BASE_GIT_TAG now very sneakily also accepts a branch name
  - IMPORTANT: this includes: BREAKING CHANGE: patches failing to apply now break the build. fixes #4958
    - also b [... cut]
commit 70a64a04 (2023-06-13): patching: Patching Summary and Patching Failures tables using Python's Rich

- even Rich'er patch output by colorizing certain strings green/yellow/red
- BASE_GIT_TAG now very sneakily also accepts a branch name
- IMPORTANT: this includes: BREAKING CHANGE: patches failing to apply now break the build. fixes #4958
  - also break on legacy `process_patch_file()` failure, remove `EXIT_PATCHING_ERROR`
closed 2023-06-13 by igorpecovnik