armbian/build #4960: ERROR in function compile_kernel

codesnas, 2023-03-18:
### What happened?

[ERROR] Failed to open ServerStat file /home/runner/work/armbian/armbian/build/cache/.aria2/server_stats for read.

[ warn ] * [l][c] board-bananapim5-002-switch-VDDIO_C-pin-to-OP.patch [ failed ]

[ error ] ERROR in function compile_kernel [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:280 -> functions/main/build-tasks.sh:155 -> functions/compilation/kernel.sh:143 -> functions/logging/traps.sh:0 ]
[1166](https://github.com/codesnas/amlogic-s9xxx-armbian/actions/runs/4452705531/jobs/7820593570#step:5:1167)
[ error ] Kernel was not built [ @host ]
[1167](https://github.com/codesnas/amlogic-s9xxx-armbian/actions/runs/4452705531/jobs/7820593570#step:5:1168)
[ o.k. ] Process terminated 
[1168](https://github.com/codesnas/amlogic-s9xxx-armbian/actions/runs/4452705531/jobs/7820593570#step:5:1169)
Error: Process completed with exit code 255.

### How to reproduce?

sudo ./compile.sh RELEASE=jammy BOARD=odroidn2 BRANCH=current BUILD_ONLY=default HOST=armbian EXPERT=yes  BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no CLEAN_LEVEL="make,debs" COMPRESS_OUTPUTIMAGE="sha"

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
[ o.k. ] Syncing clock [ host ]
[ o.k. ] Checking for external GCC compilers 
[ .... ] downloading using http(s) network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]
03/17 20:33:33 [ERROR] Failed to open ServerStat file /home/runner/work/armbian/armbian/build/cache/.aria2/server_stats for read.
[ o.k. ] Verified [ PGP ]
[ .... ] decompressing 

[ o.k. ] Checking git sources [ rtl8811cu 7b8c45a270454f05e2dbf3beeb4afcf817db65da ]
[ .... ] Creating local copy 
[ .... ] Fetching updates 
fatal: could not read Username for 'https://github.com': No such device or address
fatal: could not read Username for 'https://github.com': No such device or address
[ .... ] Checking out 
error: pathspec 'FETCH_HEAD' did not match any file(s) known to git
cp: cannot stat '/home/runner/work//armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/core': No such file or directory
cp: cannot stat '/home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/hal': No such file or directory
cp: cannot stat '/home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/include': No such file or directory
cp: cannot stat '/home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/os_dep': No such file or directory
cp: cannot stat '/home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/platform': No such file or directory
cp: cannot stat '/home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/Makefile': No such file or directory
sed: can't read /home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/Kconfig: No such file or directory
cp: cannot stat '/home/runner/work/armbian/armbian/build/cache/sources/rtl8811cu/7b8c45a270454f05e2dbf3beeb4afcf817db65da/Kconfig': No such file or directory
sed: can't read /home/runner/work/armbian/armbian/build/cache/sources/linux-mainline/linux-6.1.y/drivers/net/wireless/rtl8811cu/Makefile: No such file or directory
sed: can't read /home/runner/work/armbian/armbian/build/cache/sources/linux-mainline/linux-6.1.y/drivers/net/wireless/rtl8811cu/Makefile: No such file or directory
[ warn ] * applying wireless-rtl8811cu-Fix-p2p-go-advertising.patch [ failed ]
[ warn ] * applying wireless-rtl8811cu-Fix-VFS-import.patch [ failed ]
[ o.k. ] Adding [ Wireless drivers for Realtek 88x2bu chipsets commit:2590672d717e2516dd2e96ed66f1037a6815bced ]
[ o.k. ] Checking git sources [ rtl88x2bu 2590672d717e2516dd2e96ed66f1037a6815bced ]
[ .... ] Creating local copy 

[ warn ] * [l][c] board-bananapim5-002-switch-VDDIO_C-pin-to-OP.patch [ failed ]
[ warn ] * [l][c] board-bananapim5-003-convert-dts-to-dtsi.patch [ failed ]

[ warn ] * [l][c] board-radxa-zero-dts-otg-fix.patch [ failed ]
[ warn ] * [l][c] general-arm64-dts-amlogic-Make-mmc-host-controller-interrupt.patch [ failed ]

[ warn ] * [l][c] general-meson-mmc-1-arm64-amlogic-mmc-meson-gx-Add-core-tx-rx-eMMC-SD-SD.patch [ failed ]

[ o.k. ] Using kernel config file [ config/kernel/linux-meson64-current.config ]
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
drivers/net/wireless/Kconfig:42: can't open file "drivers/net/wireless/rtl8811cu/Kconfig"
make[1]: *** [scripts/kconfig/Makefile:77: olddefconfig] Error 1
make: *** [Makefile:704: olddefconfig] Error 2
[ .... ] Compressing sources for the linux-source package 
dpkg-deb: building package 'linux-source-6.1.20-current-meson64' in '/tmp/tmp.E76LBhoY9L/linux-source-current-meson64_23.02.0-trunk_all.deb'.
[ o.k. ] Make the linux-source package [ 163 sec. ]
  SYNC    include/config/auto.conf.cmd
make[2]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[1]: *** [Makefile:704: syncconfig] Error 2
make: *** [Makefile:805: include/config/auto.conf.cmd] Error 2
[ error ] ERROR in function compile_kernel [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:280 -> functions/main/build-tasks.sh:155 -> functions/compilation/kernel.sh:143 -> functions/logging/traps.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
Error: Process completed with exit code 255.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-03-18:
Try compiling with EXTRAWIFI="no"

codesnas, 2023-03-19:
Thanks, after disabling the wifi module, the error of rtl8811cu was skipped.

This rtl8811cu cache missing problem, will it be fixed in the future?

igorpecovnik, 2023-03-19:
> This rtl8811cu cache missing problem, will it be fixed in the future?

You are using old build framework from master branch which was completely abandoned by the team 3 weeks ago. 
https://www.armbian.com/newsflash/code-freeze-and-moving-to-new-framework/
I would advise moving to main. But. There are some changes, documentation is not yet done, and we have some stability issues. "Now" is the worse time ... few weeks from today should be production ready, most obvious bugs squashed.

We froze sources in attempt to keep master in at least build-able state, but we didn't cover all corner cases - when 3rd party repository is removed - that didn't happen in past few years. Since our resources are slim and R&D almost entirely financed from our private pockets, there is little we can do. It is already crazy to maintain one project branch with so little people, while two is impossible.

closed 2023-03-19 by igorpecovnik

pierre-pretorius, 2023-07-09:
This issue can be closed.  The problem was that on the deprecated master branch, this repo doesn't exist anymore:

https://github.com/morrownr/8821cu-20210118

It's referenced here:
https://github.com/armbian/build/blob/master/lib/functions/compilation/patch/drivers_network.sh

On the "main" branch the problem is resolved.

lambda-zhang, 2023-09-01:
https://github.com/Pix13/8821cu-20210118.git