armbian/build #4984: RTL8189FTV Driver

SajjadSabzkar, 2023-03-21:
### Which feature would you like to have?

Hi , why this driver not in your moudles?

### Funding

- [X] [I am willing to cover expenses](https://www.armbian.com/contact/)

igorpecovnik, 2023-03-21:
Please be more specific. Where? https://github.com/armbian/build/tree/main/config/kernel

SajjadSabzkar, 2023-03-21:
> Please be more specific. Where? https://github.com/armbian/build/tree/main/config/kernel

I use Linux Community aml-s9xxx version and i would like to use rtl8189ftv in my device

igorpecovnik, 2023-03-21:
> I use armbian Linux Community aml-s9xxx

This you need to start resolving here https://forum.armbian.com/forum/24-tv-boxes/ We don't support TV boxes and this is build framework issues tracker which does not cover hardware related issues.

All what maintainers can do is to accept tested changes provided by someone https://github.com/armbian/build/pulls