armbian/build #4985: Add option to copy whole resolv.conf file for chroot instead of nameserver only

rick-heig, 2023-03-21:
### Which feature would you like to have?

When building with certain network configurations and the current build scripts the build would fail because it could not resolve some network names (e.g., could not update apt package list nor install packages when in chroot).

See https://forum.armbian.com/topic/27270-build-fails-on-could-not-resolve-portsubuntucom-on-machine-that-has-built-successfully-in-the-past/ for example and build log.

Adding a specific nameserver to `${SDCARD}/etc/resolv.conf` would not suffice.

I would suggest an option to copy the whole `/etc/resolv.conf`file to the rootfs prior to chroot.

Something like :
```shell
if [[ $COPY_RESOLVCONF == yes ]]; then
    cp /etc/resolv.conf ${SDCARD}/etc/resolv.conf
fi
```

in https://github.com/armbian/build/blob/main/lib/functions/rootfs/distro-agnostic.sh#L270

Thank you for your consideration.
Sincerely,
Rick

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

rpardini, 2023-03-22:
Yeah, pull requests are welcome. I'd not do it by default, though.
Could you detail what's the contents of said resolv.conf?

rick-heig, 2023-03-22:
I have the extra `edns0` and `trust-ad` options
and a custom search list for host-name lookup (company specific).
https://manpages.ubuntu.com/manpages/jammy/man5/resolv.conf.5.html

It looks like this :
```
nameserver 127.0.0.53
options edns0 trust-ad
search example.search.domain1.com example.search.domain2.com ...
```

rpardini, 2023-03-25:
Ok thanks

closed 2023-03-25 by rpardini