armbian/build #4986: Append local-version to kernel breaks symlink in /boot/

rick-heig, 2023-03-21:
### What happened?

Adding a local-version string to the Linux kernel now breaks the symlink in /boot

Here I added "armbian" as local-version in kernel config menu : 
```
General setup --->
	Local version : armbian
```

If I do a `ls -l` on the SD card in /boot we can see that the symlink `Image -> vmlinuz-6.1.20-media` is broken because the kernel file is `vmlinuz-6.1.20armbian-media`

```shell
(base) rick@SomePC:/mnt/space/armbian/build_test_image/output/images/sd_mount/boot$ ls -l
total 96596
-rw-rw-r-- 1 root root     1536 Mar 21 08:42 armbian_first_run.txt.template
-rw-rw-r-- 1 root root    38518 Mar 21 08:42 boot.bmp
-rw-r--r-- 1 root root   260382 Mar 21 08:17 config-6.1.20armbian-media
lrwxrwxrwx 1 root root       16 Mar 21 08:40 dtb -> dtb-6.1.20-media
drwxr-xr-x 4 root root     4096 Mar 21 08:40 dtb-6.1.20-media
drwxrwxr-x 2 root root     4096 Mar 21 08:36 extlinux
lrwxrwxrwx 1 root root       20 Mar 21 08:40 Image -> vmlinuz-6.1.20-media
-rw-r--r-- 1 root root 33535694 Mar 21 08:46 initrd.img-6.1.20armbian-media
-rw-r--r-- 1 root root  4261171 Mar 21 08:17 System.map-6.1.20armbian-media
lrwxrwxrwx 1 root root       27 Mar 21 08:46 uInitrd -> uInitrd-6.1.20armbian-media
-rw-r--r-- 1 root root 33535758 Mar 21 08:46 uInitrd-6.1.20armbian-media
-rw-r--r-- 1 root root 27257344 Mar 21 08:17 vmlinuz-6.1.20armbian-media
```

This results in uBoot complaining that it cannot find the kernel image, and no boot ensues.

Appending local version string to the kernel worked in the past, so it seems it broke fairly recently.
Best regards,
Rick

### How to reproduce?

Last commit when I cloned : 7be216f4b5807e61ef46b7e5b9bfb9fe62da3a8e

```shell
git clone https://github.com/armbian/build.git
cd build
./compile.sh BOARD=nanopct4 BRANCH=current RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_CONFIGURE=yes DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base DESKTOP_APPGROUPS_SELECTED="3dsupport browsers desktop_tools editors internet programming remote_desktop" COMPRESS_OUTPUTIMAGE=sha,img
```

* Change the local version in kernel config
  General setup --->
    Local version : Set to something
* Save and exit
* Let the build finish ...
* Mount the SD card image and see that the symlink is broken

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rick-heig, 2023-03-21:
Also when installing the `.deb` in armbian when rebuilding the kernel only results in symlinks being broken (and uIinitrd missing).

Install the `.deb` log :
```shell
rick@nanopct4:~$ sudo dpkg -i linux-image-current-media_23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d_arm64.deb
(Reading database ... 145552 files and directories currently installed.)
Preparing to unpack linux-image-current-media_23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d_arm64.deb ...
Armbian 'linux-image-current-media' for '6.1.20-media': 'prerm' starting.
+ export 'DEB_MAINT_PARAMS=upgrade 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ DEB_MAINT_PARAMS='upgrade 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ export INITRD=Yes
+ INITRD=Yes
+ test -d /etc/kernel/prerm.d
+ run-parts --arg=6.1.20-media --arg=/boot/vmlinuz-6.1.20-media /etc/kernel/prerm.d
+ set +x
Armbian 'linux-image-current-media' for '6.1.20-media': 'prerm' finishing.
Armbian 'linux-image-current-media' for '6.1.20-media': 'preinst' starting.
+ export 'DEB_MAINT_PARAMS=upgrade 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ DEB_MAINT_PARAMS='upgrade 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ export INITRD=Yes
+ INITRD=Yes
+ test -d /etc/kernel/preinst.d
+ run-parts --arg=6.1.20-media --arg=/boot/vmlinuz-6.1.20-media /etc/kernel/preinst.d
+ mountpoint -q /boot
+ set +x
Armbian 'linux-image-current-media' for '6.1.20-media': 'preinst' finishing.
Unpacking linux-image-current-media (23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d) over (23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d) ...
Armbian 'linux-image-current-media' for '6.1.20-media': 'postrm' starting.
+ export 'DEB_MAINT_PARAMS=upgrade 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ DEB_MAINT_PARAMS='upgrade 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ export INITRD=Yes
+ INITRD=Yes
+ test -d /etc/kernel/postrm.d
+ run-parts --arg=6.1.20-media --arg=/boot/vmlinuz-6.1.20-media /etc/kernel/postrm.d
+ set +x
Armbian 'linux-image-current-media' for '6.1.20-media': 'postrm' finishing.
Setting up linux-image-current-media (23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d) ...
Armbian 'linux-image-current-media' for '6.1.20-media': 'postinst' starting.
+ export 'DEB_MAINT_PARAMS=configure 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ DEB_MAINT_PARAMS='configure 23.05.0-trunk--6.1.20-S7eae-D2a58-Pe89a-C999999Hfe66-B262d'
+ export INITRD=Yes
+ INITRD=Yes
+ test -d /etc/kernel/postinst.d
+ run-parts --arg=6.1.20-media --arg=/boot/vmlinuz-6.1.20-media /etc/kernel/postinst.d
 * dkms: running auto installation service for kernel 6.1.20-media       [ OK ] 
update-initramfs: Generating /boot/initrd.img-6.1.20-media
W: missing /lib/modules/6.1.20-media
W: Ensure all necessary drivers are built into the linux image!
depmod: ERROR: could not open directory /lib/modules/6.1.20-media: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
cat: /var/tmp/mkinitramfs_pBRL9y/lib/modules/6.1.20-media/modules.builtin: No such file or directory
W: Can't find modules.builtin.modinfo (for locating built-in drivers' firmware, supported in Linux >=5.2)
depmod: WARNING: could not open modules.order at /var/tmp/mkinitramfs_pBRL9y/lib/modules/6.1.20-media: No such file or directory
depmod: WARNING: could not open modules.builtin at /var/tmp/mkinitramfs_pBRL9y/lib/modules/6.1.20-media: No such file or directory
update-initramfs: Converting to u-boot format
Remove unused generated file: /boot/uInitrd-6.1.20-verb-v3-210323-media
Remove unused generated file: /boot/uInitrd-6.1.20-media
Remove unused generated file: /boot/initrd.img-6.1.20-media
Remove unused generated file: /boot/initrd.img-6.1.20-verb-v3-210323-media
+ echo 'Armbian: update last-installed kernel symlink to '\''Image'\''...'
Armbian: update last-installed kernel symlink to 'Image'...
+ ln -sfv vmlinuz-6.1.20-media /boot/Image
'/boot/Image' -> 'vmlinuz-6.1.20-media'
+ touch /boot/.next
+ set +x
Armbian 'linux-image-current-media' for '6.1.20-media': 'postinst' finishing.
```

The error : `depmod: ERROR: could not open directory /lib/modules/6.1.20-media: No such file or directory` is because the actual directory has the "local version" string appended...

Checking /boot : 
```shell
rick@nanopct4:~$ ls -l /boot/
total 31092
-rw-rw-r-- 1 root root     1536 Mar 21 07:43 armbian_first_run.txt.template
-rw-rw-r-- 1 root root    38518 Mar 21 07:43 boot.bmp
-rw-r--r-- 1 root root   260315 Mar 21 08:46 config-6.1.20-verbi-v3-210323-media
lrwxrwxrwx 1 root root       16 Mar 21 07:40 dtb -> dtb-6.1.20-media
drwxr-xr-x 4 root root     4096 Mar 21 07:40 dtb-6.1.20-media
drwxrwxr-x 2 root root     4096 Mar 21 07:36 extlinux
lrwxrwxrwx 1 root root       20 Mar 21 10:36 Image -> vmlinuz-6.1.20-media
-rw-r--r-- 1 root root  4261174 Mar 21 08:46 System.map-6.1.20-verbi-v3-210323-media
lrwxrwxrwx 1 root root       20 Mar 21 10:36 uInitrd -> uInitrd-6.1.20-media
-rw-r--r-- 1 root root 27257344 Mar 21 08:46 vmlinuz-6.1.20-verbi-v3-210323-media
```

Both targets for `Image` and `uInitrd` are missing missing

rpardini, 2023-03-22:
Yeah, that's simply not supported. We actually disable localversion forcefully in any given `.config`, but that is done before manual configuration. Probably should do it after, as well. Is there any reason why you need such a thing?

closed 2023-03-22 by rpardini