armbian/build #4991: Can't built new linux kernels 6.3‑rc2 6.3‑rc3

xlazom00, 2023-03-21:
### What happened?

Can't built new linux kernels 6.3‑rc2 6.3‑rc3
on ubuntu 22.04

commenting this line fixed problem
https://github.com/armbian/build/blob/main/lib/functions/compilation/kernel-debs.sh#L374

### How to reproduce?

Command you gave to the framework!

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log output

```shell
[🔨]   dpkg-deb: building package 'linux-dtb-edge-sc8280xp' in '/home/misko/armbian12/.tmp/work-fde4e23a-8f57-46b3-9258-7b35bd5bd3f0/tmp.0aiROwfVad//linux-dtb-edge-sc8280xp_23.05.0-trunk--6.3-rc3-Se8d0-D2a58-P1bd3-C5e95Hfe66-B262d_arm64.deb'.
[🌱] Packaging linux-headers [ sc8280xp linux-sc8280xp-edge ]
[🐛] linux-headers packaging [ /home/misko/armbian12/.tmp/work-fde4e23a-8f57-46b3-9258-7b35bd5bd3f0/tmp.NY2MGFsMRH ]
[🔨]   '/home/misko/armbian12/.tmp/work-fde4e23a-8f57-46b3-9258-7b35bd5bd3f0/tmp.NY2MGFsMRH/lib/modules/6.3.0-rc3-sc8280xp/build' -> '/usr/src/linux-headers-6.3.0-rc3-sc8280xp'
[🔨]   '/home/misko/armbian12/cache/sources/linux-kernel-worktree/6.3__sc8280xp__arm64/.config' -> '/home/misko/armbian12/.tmp/work-fde4e23a-8f57-46b3-9258-7b35bd5bd3f0/tmp.NY2MGFsMRH/usr/src/linux-headers-6.3.0-rc3-sc8280xp/.config'
[🔨]     RMOBJ    tools/acpidbg/
[🔨]     RM       acpidbg
[🔨]     RMINC    include
[🔨]     RMOBJ    tools/acpidump/
[🔨]     RM       acpidump
[🔨]     RMINC    include
[🔨]     RMOBJ    tools/ec/
[🔨]     RM       ec
[🔨]     RMINC    include
[🔨]     RMOBJ    tools/pfrut/
[🔨]     RM       pfrut
[🔨]     RMINC    include
[🔨]     CLEAN   arm64
[🔨]     CLEAN   libapi
[🔨]     CLEAN   libbpf
[🔨]     CLEAN   libsubcmd
[🔨]     CLEAN   libsymbol
[🔨]     CLEAN   libperf
[🔨]     CLEAN   fixdep
[🔨]     CLEAN   feature-detect
[🔨]     CLEAN   python
[🔨]     CLEAN   bpf-skel
[🔨]     CLEAN   coresight
[🔨]     CLEAN   core-objs
[🔨]     CLEAN   core-progs
[🔨]     CLEAN   core-gen
[🔨]     CLEAN   Documentation
[🔨]   Makefile:32: compiler "gcc" does not support the ARMv8.5 MTE extension.
[🔨]   Makefile:33: test program "mte" will not be created.
[🔨]     CLEAN    
[🔨]   make[2]: *** No rule to make target 'clean'.  Stop.
[🔨]   make[1]: *** No rule to make target 'clean'.  Stop.
[🔨]   make: *** [Makefile:173: vm_clean] Error 2
[💥] Error 2 occurred in main shell
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xlazom00, 2023-03-21:
All fine with 6.2.x

rpardini, 2023-03-22:
Thanks, this has been reported elsewhere as well - @lanefu ?
I will check this in a few days.

lanefu, 2023-03-22:
> Thanks, this has been reported elsewhere as well - @lanefu ?
> I will check this in a few days.

Yep looks like what I ran into on a 6.3 tree i was using from collabora's repo.    I applied a[ similar workaround](https://github.com/lanefu/build/commit/adbad36d3ce6a65666224dbc1c2f5ae86358a593)

rpardini, 2023-03-25:
Question is, does linux-headers actually _work_ when installed? My feeling is that without `make clean` in that strategic spot, when building on x86 the headers for arm64 won't work.

The-going, 2023-04-04:
> Question is, does linux-headers actually _work_ when installed? My feeling is that without `make clean` in that strategic spot, when building on x86 the headers for arm64 won't work.

When cross-compiling, auxiliary binary files are created that should work on the architecture of the assembly machine (x86), but should do something for the target architecture (arm, arm64, ...).
These files should be recreated directly on the target architecture, i.e. first clean and then create again on the target architecture.

rpardini, 2023-04-04:
> When cross-compiling, auxiliary binary files are created that should work on the architecture of the assembly machine (x86), but should do something for the target architecture (arm, arm64, ...). [...]

If you _read the code_ you'll understand we now have a separation/copy between the actual tree used for building the image and the the (partial) tree used for packaging the headers, which is where that specific `make clean` is ran (and now fails with 6.3-rcX+). 

The headers package actually contain a mix of source code and artifacts produced during the image build, and has a `postinst` script that builds them _when installed_ (not when package built).

Maintaining this _mess_ has allowed us to cross-compile a kernel and produce somewhat-working headers for the target architecture, but is proving each day more and more difficult to maintain. The breakage on 6.3 clearly indicates we're swimming against the current.

The-going, 2023-04-04:
> If you _read the code_ you'll understand we

I have a problem with that. I understand the Makefile syntax well.

> (and now fails with 6.3-rcX+).

Oh-ho! I confirm that the changes exist and they are significant for 6.3.
This requires urgent attention for me.
By the way, this is an expected process for the development of scripts for building packages from the kernel source code.

Hint for changes:
```diff
....
-echo "1.0" > debian/source/format
+echo "3.0 (quilt)" > debian/source/format
.....
-Build-Depends: bc, rsync, kmod, cpio, bison, flex $extra_build_depends
+Build-Depends: bc, debhelper, rsync, kmod, cpio, bison, flex $extra_build_depends
.....
 clean:
        rm -rf debian/files debian/linux-*
-       \$(MAKE) clean
+       \$(MAKE) -f \$(srctree)/Makefile ARCH=${ARCH} clean
 
 binary: binary-arch
 EOF
```

rpardini, 2023-04-04:
That's not as bad as I thought. Thanks. We could make work with some 3 lines of code? ("if version > 6.3, pass the root Makefile to make")

(also, Interesting they moved to debhelper after 20 years).

The-going, 2023-04-04:
> We could make work with some 3 lines of code?

There are much more changes.
Yes, it used to be a simple script to show and test that the kernel can be assembled into a "1.0" format package. Now it is a simple script to build a "3.0 (quilt) format package.
And this involves using a series of patches. And does it.

My investigation is ongoing.

rpardini, 2023-04-15:
Closing this.
The real problem behind 6.3 and our make clean was a bug upstream.
I've contacted the developer and he sent a fix. https://lore.kernel.org/linux-mm/20230415203110.13858-1-sj@kernel.org/T/#u (hopefully lands before 6.3 final)
Armbian-side, there's a workaround in #5059 merged, which I'll keep since we'll be building -rcX kernels anyway.

PR #5059 (2023-04-15): kernel: fix for 6.3+ linux-headers pkg due to upstream dir change
#### kernel: fix for 6.3+ linux-headers pkg due to upstream dir change
- extra: run kernel make through `pipetty` for _moar colors_

>	 Small detour: in v6.3-rc1, in commit https://github.com/torvalds/linux/commit/799fb82aa132fa3a3886b7872997a5a84e820062,
>	               the `tools/vm` dir was renamed to `tools/mm`. Unfortunately `tools/Makefile` still expects it to exist,
>	               and `make clean` in the `tools` dir fails. Drop in a fake Makefile there to work around this.

Should fix #4991
closed 2023-04-15 by rpardini