armbian/build #4993: Extra Packages won't be built anymore

maximilianvoss, 2023-03-22:
### What happened?

On migration from the old build infrastructure to the new, all packages in packages/extra-buildpkgs are no longer built and included in the image.

### How to reproduce?

./compile.sh \
BOARD=retrostone2 \
BRANCH=current \
RELEASE=jammy \
BUILD_DESKTOP=yes \
DESKTOP_ENVIRONMENT=retrostone2 \
KERNEL_CONFIGURE=no \
DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base \
DESKTOP_APPGROUPS_SELECTED=y \
EXTERNAL_NEW=compile \
KERNEL_GIT=shallow

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hzyitc, 2023-03-23:
May I ask which package do you need?

It seem like that all packages in `packages/extra-buildpkgs` are uncessary now.

maximilianvoss, 2023-03-23:
To build custom images with custom or patched packages the way forward was to include them via extra packages.

iav, 2023-03-23:
kernel-sources?
`perf`?

maximilianvoss, 2023-03-23:
Something like custom libsdl and depending applications which like retropie

rpardini, 2023-03-24:
Yeah, `extras/buildpkg` has been removed since Developer Meeting 3.
The use case can be achieved by building pkgs separately and installing during customization phase.
I do have a far-future plan to provide lower-level tooling for a userspace build chroot based on our cached rootfs, but the priority is very low.
Developers (which would contribute userspace pkgs to Armbian, not only for their own images) are claiming for a "Launchpad, but with Debian support" which is a bit higher up the list.

closed 2023-03-24 by rpardini