armbian/build #4995: Unisoc uwe5622 patches fail on Ayufan legacy 4.4 kernel

amcduffee, 2023-03-24:
### What happened?

When trying to build the kernel for `BOARD=rockpi-4b` on the `legacy` kernel branch the Unisoc uwe5622 patches fail to apply as shown here:
```
[ o.k. ] Adding [ Drivers for Unisoc uwe5622 found on some Allwinner and Rockchip boards ]
[ warn ] * applying wireless-driver-for-uwe5622-allwinner.patch [ failed ]
[ o.k. ] * applying wireless-driver-for-uwe5622-allwinner-bugfix.patch 
[ o.k. ] * applying wireless-driver-for-uwe5622-warnings.patch 
[ warn ] * applying wireless-driver-for-uwe5622-park-link-pre-v6.1.patch [ failed ]
```

I am able to get past it using the following change:
```
diff --git a/lib/functions/compilation/patch/drivers_network.sh b/lib/functions/compilation/patch/drivers_network.sh
index 6a62055b0..37fb58962 100644
--- a/lib/functions/compilation/patch/drivers_network.sh
+++ b/lib/functions/compilation/patch/drivers_network.sh
@@ -548,7 +548,7 @@ driver_rtl8822BS()
 driver_uwe5622_allwinner()
 {
        # Unisoc uwe5622 wireless Support
-       if linux-version compare "${version}" ge 4.4 && linux-version compare "${version}" le 6.2 && [[ "$LINUXFAMILY" == sunxi* || "$LINUXFAMILY" == rockchip64 ]]; then
+       if linux-version compare "${version}" ge 4.4 && linux-version compare "${version}" le 6.2 && [[ "$LINUXFAMILY" == sunxi* || "$LINUXFAMILY" == rockchip64 ]] && [[ "$BRANCH" != legacy ]]; then
                display_alert "Adding" "Drivers for Unisoc uwe5622 found on some Allwinner and Rockchip boards" "info"
 
                process_patch_file "${SRC}/patch/misc/wireless-driver-for-uwe5622-allwinner.patch" "applying"
```

I know it is undesirable to touch core files but I can't find another way to prevent these patches from applying on the `legacy` kernel branch.  I have so far been unsuccessful in changing `driver_uwe5622_allwinner` using a `userpatches/sources/families/rockchip64.conf` override and a quick read of `lib/functions/compilation/patch/drivers_network.sh` suggests I can't use an extension to change the behavior.

I am not even sure it makes sense to apply this patch series on the Ayufan 4.4.y kernels?

### How to reproduce?

On branch `v23.02` use:
```
./compile.sh  BOARD=rockpi-4b BRANCH=legacy BUILD_MINIMAL=no BUILD_DESKTOP=no BUILD_ONLY=u-boot,kernel KERNEL_CONFIGURE=no
```

My build host is running Ubuntu Hirsute.

### Branch

other

### On which host OS are you observing this problem?

Other

### Relevant log output

```shell
== u-boot make rock-pi-4-rk3399_defconfig ==

        == u-boot make BL31=/home/anderson/git/armbian/cache/sources/rkbin-tools/rk33/rk3399_bl31_v1.35.elf tpl/u-boot-tpl.bin spl/u-boot-spl.bin u-boot.itb u-boot-dtb.bin ==

.config:2232:warning: override: reassigning to symbol BOOTDELAY

        == kernel ==

drivers/bluetooth/hci_ldisc.c: In function \u2018hci_uart_register_dev\u2019:
drivers/bluetooth/hci_ldisc.c:626:11: error: \u2018HCI_QUIRK_BROKEN_PARK_LINK_STATUS\u2019 undeclared (first use in this function); did you mean \u2018HCI_QUIRK_BROKEN_STORED_LINK_KEY\u2019?
   set_bit(HCI_QUIRK_BROKEN_PARK_LINK_STATUS, &hdev->quirks);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           HCI_QUIRK_BROKEN_STORED_LINK_KEY
drivers/bluetooth/hci_ldisc.c:626:11: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:277: drivers/bluetooth/hci_ldisc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:484: drivers/bluetooth] Error 2
make[1]: *** Waiting for unfinished jobs....
net/bluetooth/hci_core.c: In function \u2018hci_setup_link_policy\u2019:
net/bluetooth/hci_core.c:687:42: error: \u2018HCI_QUIRK_BROKEN_PARK_LINK_STATUS\u2019 undeclared (first use in this function); did you mean \u2018HCI_QUIRK_BROKEN_STORED_LINK_KEY\u2019?
  if (lmp_park_capable(hdev) && !test_bit(HCI_QUIRK_BROKEN_PARK_LINK_STATUS, &hdev->quirks))
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                          HCI_QUIRK_BROKEN_STORED_LINK_KEY
net/bluetooth/hci_core.c:687:42: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:277: net/bluetooth/hci_core.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:484: net/bluetooth] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1048: drivers] Error 2
make: *** Waiting for unfinished jobs....
make: *** [Makefile:1048: net] Error 2
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-03-24:
This very old code has to be removed / upgraded to latest Rockchip vendor kernel. Kernel is here https://github.com/armbian/linux-rockchip but we don't have any RK3399 maintainers for.

Workaround is building with EXTRAWIFI=no

amcduffee, 2023-03-24:
Yea, I already tried `EXTRAWIFI=no` and it doesn't help.  Likely because the `driver_uwe5622_allwinner()` function is missing the `[ "$EXTRAWIFI" == yes ]` check that many other functions have.  So, maybe that is the real bug here?

I agree on the point of the 4.4.y kernels being very old code.  However, as is often the case with theses SBCs, only the vendor kernel with their BSP properly supports the OTG capability.  Having functioning OTG is a requirement of my project so I can't just switch to the mainline kernels that are built by Armbian (as much as I would like to).

Fixing broken kernel drivers is above my skill level currently, so I have to continue using the 4.4.y unless there is a newer, ideally mainline, kernel that has fully functioning OTG.  I have yet to find one.  I will take a look at the one you suggest.

If you are aware of the proper upstream channels to submit bug requests to in order to get OTG working on mainline kernels I can try to pursue that in hopes that it will flow back down to Armbian via mainline.

amcduffee, 2023-03-24:
On a similar note it looks like the `driver_rtl8822cs_bt()` function is also missing the `[ "$EXTRAWIFI" == yes ]` check.

rpardini, 2023-04-29:
See https://github.com/armbian/build/issues/5120

closed 2023-04-29 by rpardini