armbian/build #5008: Website with packages search

PanderMusubi, 2023-03-28:
### Which feature would you like to have?

Please provide a website with packages search such as https://packages.ubuntu.com/ and https://packages.debian.org/ for https://apt.armbian.com/ and link to it in Armbian's documentation and pages such as http://xogium.performanceservers.nl/

Implementing this offers greater insight to Armbian's packages and allows users to more easily find packages and see which version is available for which version without having to install a specific version.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

closed 2023-04-04 by rpardini

PanderMusubi, 2023-04-04:
The websites from Ubuntu and Debian provide searching in different ways. Software for that is FOSS, just needs to get installed, configured and maintained.

igorpecovnik, 2023-04-04:
I am using those search engines often, when packages levels needs adjustments, depending on which base Armbian is on, Debian / Ubuntu. Some of the details sometimes matters.

> and maintained.

Sadly most of volunteers runs away when boring maintenance is needed and professionals needs to pay their living expenses while customers thinks we are idling waiting for wishes in exchange for less then 0.5% of the time which is covered by donations. We are totally overloaded and when people wants to help its hard as dealing with new people, if they needs guidance, is again "just" more work ...

Already covering and maintaining what we have now, this entire project, is mission impossible, difficult to stay happy.

With lets say 100 people, full time engineers and support staff, we are unable to clear features requests that are getting assembled here and https://forum.armbian.com/forum/38-feature-requests/ on top of endless stream of bugs and support requests.

There are just a few of us and we can be more if you step up ... and do what is already on our 2do list :) Then, after a while, this wish might be addressed.

Best effort.