armbian/build #5012: armbian-audio-config fails in minimal images; alsa-utils missing

matthijskooijman, 2023-03-29:
### What happened?

I booted a minimal cli image and found audio output was muted. armbian-audio-config is supposed to set sane volumes for audio devices, but it relies on `amixer` from the `alsa-utils` package, which is not installed in a minimal image.

I originally saw this on a customized 23.02 image on an orangepipc, but I confirmed this by:
 - building today's main branch
 - booting it
 - Installing mpg321
 - Trying to play some audio file on the minijack output: `mpg321 -a hw:0 sample-15s.mp3`, which produces no audible sound
 - Installing `alsa-utils`
 - Using `alsamixer` to see that the "Line out" control of card 0 is indeed muted
 - Rebooting
 - Using `alsamixer` to see that the "Line out" control of card 0 is now successfully unmuted and set to 45% (0dB), presumably by `armbian-audio-config`
 - Using the same `mpg321` command to play audio, which is now audible

The example is maybe a bit contrived, but it serves to confirm the problem.

The main question is, is this situation:
 1. Expected (i.e. not-a-bug), because a minimal image is not intended for audio, so if you need audio you should install `alsa-utils` in addition to whatever audio player you also install?
 2. Unexpected (i.e. a bug), because `armbian-audio-config` should work in a minimal image too and depends on `alsa-utils`, so that should be installed in minimal packages too?

If 1., feel free to close this issue. If 2., I can prepare a PR to fix this.

Installing alsa-utils on a minimal cli orangepipc image takes up about 6MB of extra disk space (according to apt when installing it).

### How to reproduce?

 ./compile.sh BOARD=orangepipc BRANCH=current RELEASE=jammy BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_CONFIGURE=prebuilt

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-03-29:
2. I would say we put alsa-utils back into minimal image.

matthijskooijman, 2023-03-29:
Ok, submitted PR #5013.

PR #5013 (2023-03-29): Include alsa-utils in minimal cli builds
This ensures that the armbian-audio-config script actually works in minimal builds too.

This fixes #5012 (AR-1644)

# How Has This Been Tested?

 - `./compile.sh BOARD=orangepipc BRANCH=current RELEASE=jammy BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_CONFIGURE=prebuilt`
 - booting it
 - Installing mpg321
 - Trying to play some audio file on the minijack output: `mpg321 -a hw:0 sample-15s.mp3`, which produces audible sound
 - Using alsamixer to see that the "Line out" control of card 0 is indeed successfully unmuted and set to 45% (0dB), presumably by ar [... cut]
commit 54636219 (2023-03-29): Include alsa-utils in minimal cli builds

This ensures that the armbian-audio-config script actually works in
minimal builds too.

This fixes #5012 (AR-1644)
closed 2023-03-29 by igorpecovnik