armbian/build #5015: Installer should handle input in the same way everywhere

PanderMusubi, 2023-03-30:
### What happened?

On first run, all answers to the console for setting up the system require an Enter at the end of each answer, except for one question. That is on

    Set user language based on your location? [Y/n]

When answering with `n` and Enter, the Enter is interpreted as an answer to the next question, the locale. This prompts again all locales as it is interpreted as an empty answer.

Hence, after y or n of the question above, an enter should be required. This also gives the user the opportunity to change the answer and not directly continue.

### How to reproduce?

Boot from image and follow the steps of supplying root password and then setting locale etc. 

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PanderMusubi, 2023-03-30:
The same goes for

    Choose default system command shell:
    
    1) bash
    2) zsh

This should also wait for Enter before continuing.

EvilOlaf, 2023-03-30:
Can probably be worked around by using `-n2` instead of `1` on multiple-choice questions and/or Y/n questions.

PanderMusubi, 2023-04-09:
Would you welcome a PR?

SteeManMI, 2023-04-09:
Yes a PR would be welcomed to clean up and standardize these prompts.  It has been an issue on my wish list, but the priority is too low to get around to fixing it myself.

PanderMusubi, 2023-04-09:
Can you point me where to find the script?

closed 2023-04-22 by igorpecovnik