armbian/build #5017: Missing modules in sunxi64

PanderMusubi, 2023-03-30:
### What happened?

Module s5k4ecgx which is available in current https://github.com/armbian/build/blob/main/config/kernel/linux-sunxi64-current.config#L4503 should also be available in edge above the line https://github.com/armbian/build/blob/main/config/kernel/linux-sunxi64-edge.config#L4526

Also the CONFIG_VIDEO_SUN4I_CSI and CONFIG_VIDEO_SUN6I_CSI modules from https://github.com/armbian/build/blob/main/config/kernel/linux-sunxi-edge.config#L4408 should be made available in the current and edge config files for sunxi64 which are mentioned above.

### How to reproduce?

Try to load the missing modules.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-03-31:
Feel free to build kernel with those, test them and then provide PR.

PanderMusubi, 2023-04-01:
Working on a PR

PanderMusubi, 2023-04-06:
Please see https://github.com/armbian/build/pull/5041

rpardini, 2023-04-08:
[merged](https://github.com/armbian/build/pull/5041), thanks, can you check and close this?

EricYin, 2023-04-10:
does anyone know how to use general-packaging-5.10.y.patch when building armbian23.02 with kernal 5.10.110?
the reason I need this patch is because it seems it fixed issue about missing module.lds file

EvilOlaf, 2023-04-10:
Please don't hijack issue reports for unrelated things.

closed 2023-04-21 by PanderMusubi