armbian/build #5024: I need to setup devicetree for GRUB

xlazom00, 2023-03-31:
### Which feature would you like to have?

I need to setup devicethree for GRUB
armbian don't suppor this feature so some new variable will be nice
and
armbian delete all dt files https://github.com/armbian/build/blob/main/extensions/grub.sh#L110

for now grub.cfg point to empty folder

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

xlazom00, 2023-03-31:
I need this for https://github.com/armbian/build/pull/4904

rpardini, 2023-04-01:
Yeah, to get to the bottom of this, you need to look into `update-grub`.  Find its source, see what is does, what/where it expects the DTB to be, and how to override it. Hint: it's actually [`grub-mkconfig`](https://www.gnu.org/software/grub/manual/grub/html_node/Simple-configuration.html#Simple-configuration), and the stuff that matters is in `/etc/grub.d ` specially `10_linux` or such.  You can either add a new custom with the `devicetree` config option, or somehow understand the filename rules for the dtb file and put it there. You have to write a generator, not a config template, like we do for extlinux.

The reason the dir is removed is cos the Armbian u-boot-oriented "linux-dtb" package conflicts with what grub expects and things so terribly bad on UEFI (full ACPI) systems, where the DTB is not needed, and grub tries to use them if present. There's some `/boot/dtb-<kernel_version>` GRUB convention that expects a file, finds Armbian's directory, and flops.

xlazom00, 2023-04-02:
Mainline grub don't have any property to setup devicetree from some property 
Fedora has this patch to support that so if anybody can ask debian maintainers to add that :)
https://src.fedoraproject.org/rpms/grub2/blob/rawhide/f/0022-Add-devicetree-loading.patch

I made bug report on ubuntu
https://answers.launchpad.net/ubuntu/+source/grub2/+question/706051

What if I rewrite 10_linux with sed
as it is here
https://github.com/armbian/build/blob/main/extensions/grub-sbc-media.sh#L60
Are you ok with that?

rpardini, 2023-04-02:
mainline grub _does_ have [`devicetree`](https://www.gnu.org/software/grub/manual/grub/html_node/devicetree.html) command. They _don't_ have any overlay support. That's a different matter. 

Stick to what's common between Debian & Ubuntu as well. We use the standard versions from each release, so patching grub for now is out of the question.

`grub-media-sbc` shouldn't be used as example or reference -- it something exclusive for Jetson Nano/`media` family/Oleg's efforts.

rpardini, 2023-04-02:
btw most kernels, when presented with both ACPI and a DTB, will ignore the DT. check `acpi=` kernel parameter reference for how to overcome that, but again, if you're doing UEFI -> GRUB -> DTB instead of ACPI, why is it even booting UEFI to begin with?

xlazom00, 2023-04-06:
From ubuntu bug
https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/2015058
```
We have our own patch already, all you need to do is place the dtb in the right place and it will be picked up by grub,

/boot/dtb-$(uname -r)
/boot/dtb
```

rpardini, 2023-04-06:
Yeah I told you that in https://github.com/armbian/build/issues/5024#issuecomment-1493126043
Let's close this, you've all that is needed to implement?

rpardini, 2023-08-27:
We now have `grub-with-dtb` [extension](https://github.com/armbian/build/blob/main/extensions/grub-with-dtb.sh) for this purpose.
See example usage in [thinkpad-x13s](https://github.com/armbian/build/blob/main/config/boards/thinkpad-x13s.wip#L9-L12) board file.