armbian/build #5046: missing header scripts/moudle.lds for building kernal 5.10.100

EricYin, 2023-04-10:
### What happened?

I tried to build orangepi5 armbian23.02 with kernal 5.10.110 using ./compile.sh EXPERT="yes". build finished but there is no scripts/module.lds file in header .deb file if i extract this .deb file 

### How to reproduce?

it seems general-packaging-5.10.y.patch fixed this issue? does anyone know how to use general-packaging-5.10.y.patch when building image?

### Branch

other

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-04-10:
> other

Try building from `main`. Nobody will waste time investigating issues on any other branch.

rpardini, 2023-04-11:
Exactly, try on `main`. For rk3588-legacy, we also need this merged in https://github.com/armbian/linux-rockchip/pull/13
Even so the headers work partially.
Can't expect much from such a vendor kernel.

closed 2023-04-11 by rpardini