armbian/build #5047: OrangePi Zero (LTS) TV output not working [PATCHing needed]

radoslavv, 2023-04-10:
### What happened?

In the default build there are missing few fixes to allow TV output to work on this board.
Unfortunately only way to connect some screen to this board and see where booting got frozen (time to time).

There is a good description how to fix issue available here:
[https://github.com/robertojguerra/orangepi-zero-full-setup/blob/main/README2.md](https://github.com/robertojguerra/orangepi-zero-full-setup/blob/main/README2.md)
7.3 Add the TV encoder driver to the Linux kernel source

Needs to be applied:
1. [sunxi-6.1/0036-wip-h3-h5-cvbs-armbian.patch](https://github.com/robertojguerra/orangepi-zero-full-setup/blob/main/sunxi-6.1/0036-wip-h3-h5-cvbs-armbian.patch) : makes additions to the "dts", which tells the kernel where are the new devices. Adds kernel code to interact with the tv encoder. With my modifications, now it is applicable to Armbian (this patch came from the LibreElec github).
2. [sunxi-6.1/zzzz2-tv.patch](https://github.com/robertojguerra/orangepi-zero-full-setup/blob/main/sunxi-6.1/zzzz2-tv.patch) : by Armbian user "gleam2003", adds directives to make sure that the dtbo (device tree binary overlay) is compiled
3. [sunxi-6.1/zzzz3-tv.patch](https://github.com/robertojguerra/orangepi-zero-full-setup/blob/main/sunxi-6.1/zzzz3-tv.patch) : more additions to the "dts" and "dtsi" (like C include files), which I noticed were included in "yam" patch, but missing from the LibreElec patch

Please include in the master branch so apt update and upgrades could be safely used.
I can confirm this works correctly (compiled Armbian 23.05.0-trunk Bullseye with Linux 6.1.23-sunxi).

======================================================
Additionally should be adjusted also **_/build/config/boards/orangepizero.conf_**

```
# Allwinner H2+ quad core 256/512MB RAM SoC WiFi SPI
BOARD_NAME="Orange Pi Zero"
BOARDFAMILY="sun8i"
BOOTCONFIG="orangepi_zero_defconfig"
MODULES_CURRENT="g_serial"
#MODULES_BLACKLIST="lima sunxi_cedrus"
MODULES_BLACKLIST="sunxi_cedrus"
#DEFAULT_OVERLAYS="usbhost2 usbhost3"
DEFAULT_OVERLAYS="usbhost2 usbhost3 tve"
#DEFAULT_CONSOLE="serial"
DEFAULT_CONSOLE="both"
#HAS_VIDEO_OUTPUT="no"
HAS_VIDEO_OUTPUT="yes"
SERIALCON="ttyS0,ttyGS0"
KERNEL_TARGET="legacy,current,edge"

function post_family_tweaks_bsp__orangepizero_BSP() {
    display_alert "Installing BSP firmware and fixups"

	if [[ $BRANCH == legacy ]]; then

		# Bluetooth for most of others (custom patchram is needed only in legacy)
		install -m 755 $SRC/packages/bsp/rk3399/brcm_patchram_plus_rk3399 $destination/usr/bin
		cp $SRC/packages/bsp/rk3399/rk3399-bluetooth.service $destination/lib/systemd/system/

	fi

	return 0
}

```
However not critical, as this could be done via **_armbian-config_** tool.

Additionally there were needed some TV output adjustmens in:
1. to /etc/default/console-setup added:
SCREEN_WIDTH=79
SCREEN_HEIGHT=34
2. to /etc/rc.local added:
/home/pi/allwinner_tvout_manipulator/src/bin/Release/tvout -m -r
/home/pi/allwinner_tvout_manipulator/src/bin/Release/tvout -m -x 93
/home/pi/allwinner_tvout_manipulator/src/bin/Release/tvout -m -y 45
(binary built from source [https://code.nwrk.biz/allwinner_tvout_manipulator.git
](https://code.nwrk.biz/allwinner_tvout_manipulator.git)

But this bothers me the least as does not need kernel rebuild.

### How to reproduce?

Not possible using code.
Human needs to check TV output.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Relevant log output

```shell
Not relevant.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

SteeManMI, 2023-04-10:
This issue tracker is for issues with the build framework, not for board specific issues.  Please use the forums to report board specific issues.  Or since you have already done most of the work, submit a PR with the changes for review.

closed 2023-04-10 by SteeManMI

radoslavv, 2023-04-10:
I tried to made changes & prepared pull request.
[5048 OrangePi Zero TV Output kernel patch](https://github.com/armbian/build/pull/5048)