armbian/build #5049: Add JetHub D1+

PanderMusubi, 2023-04-11:
### Which feature would you like to have?

The board JetHub D1+ is supported, but is not in the documentation https://github.com/armbian/build/blob/main/config/boards/jethubj100.conf

Difference with D1 is 
`# JetHub D1+ (j100) Amlogic A113X quad core 2Gb RAM SoC, eMMC 32Gb` (or perhaps `(j100p)`)

Not sure how to add it, either in that file or get it listed separately like https://www.armbian.com/jethub-d1/ as https://www.armbian.com/jethub-d1p/

Sources:
- Hardware details https://docs.jethome.ru/en/controllers/linux/din_rail/d1p.html
- Buy  board https://jethome.ru/d1p/
- image https://jethome.ru/images/detailed/9/D1__persp_bottom_o35s-0v.png

Additionally, on https://www.armbian.com/jethub-d1/ the link to Hardware details should be changed from https://wiki.jethome.ru/en/jethub_d1 to https://docs.jethome.ru/en/controllers/linux/din_rail/d1.html

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

PanderMusubi, 2023-04-11:
FYI @adeepn

adeepn, 2023-04-16:
There is no need to add a separate device, since the firmware is the same for all j1xx family: D1 and D1+.
armbian.com info is updated.

closed 2023-04-17 by adeepn