armbian/build #5089: boot logo cannot be customized

croscar, 2023-04-22:
### What happened?

A few months ago, I replaced build/packages/blobs/splash/logo.png and it worked well; After upgrading the build framework recently, the default boot logo is always displayed during startup. 

Then I put the new bootsplash.armbian file into the old system,and run "upgrade-initramfs -u"，
the customized logo appeared。

It seems that the new version of uboot reads the logo from elsewhere.

### How to reproduce?

modify build/packages/blobs/splash/logo.png and compile

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hzyitc, 2023-04-22:
What we use now is `plymouth`.

croscar, 2023-04-22:
I update the images in /usr/share/plymouth/themes/armbian
It works on the shutdown screen, but the startup screen has not changed

> What we use now is `plymouth`.

croscar, 2023-04-22:
It works now.

https://github.com/adi1090x/plymouth-themes

closed 2023-04-22 by croscar

Jibun-no-Kage, 2024-02-27:
@croscar, Update images how?  Mind providing some details on what you did?

Saketeas, 2024-05-17:
@Jibun-no-Kage If you haven't changed the theme, it will default to armbian.
In  /usr/share/plymouth/themes/armbian, change bgrt-fallback.png to the logo you want to replace and run upgrading-initramfs -u
At least I succeeded