armbian/build #5099: uboot v2018.05-sun50iw9 compile error

AGM1968, 2023-04-24:
### What happened?

```
[ðŸŒ±] Compiling u-boot [ 2018.05 /home/alistair/build/cache/sources/u-boot-worktree/u-boot/v2018.05-sun50iw9 ]

env: 'arm-linux-gnueabi-gcc': No such file or directory

[ðŸ’¥] Error  127 occurred in SUBSHELL [ SUBSHELL at /home/alistair/build/lib/functions/compilation/uboot.sh:320 ]
[ðŸ’¥] Error  127 occurred in SUBSHELL [ SUBSHELL at /home/alistair/build/lib/functions/compilation/uboot.sh:320 ]
[ðŸ’¥] Error 127 occurred in main shell [ at /home/alistair/build/lib/functions/compilation/uboot.sh:320
                  compile_uboot() --> lib/functions/compilation/uboot.sh:320
```

### How to reproduce?

```
./compile.sh build ARTIFACT_IGNORE_CACHE=yes BOARD=orangepizero2 BRANCH=legacy BUILD_MINIMAL=yes CREATE_PATCHES=yes KERNEL_CONFIGURE=no RELEASE=bullseye 
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
https://paste.next.armbian.com/bogatoxomi
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-04-25:
Builds fine for me with the same build command.

EvilOlaf, 2023-04-25:
Hm probably, yes.

AGM1968, 2023-04-25:
Thanks for checking.

Adding the 32bit compiler somewhere here is what's suggested?

https://github.com/armbian/build/blob/main/config/sources/families/sun50iw9.conf#L40

igorpecovnik, 2023-04-25:
There is already a function:
https://github.com/armbian/build/blob/main/config/sources/families/sun50iw9.conf#L73
so just this package has to be added there.

AGM1968, 2023-04-25:
Ok great thanks for advising, I should have read completely, I will propose here before pr

PR #5111 (2023-04-28): rpardini's dirty laundry, late April/'23
#### rpardini's dirty laundry, late April/'23

- `uefi`/`edge`: bump to `6.3`
- `bcm2711`/`grub`: add `busybox` (for initrd grep/sed) so `initramfs-growroot` works again on Debian
  - bcm2711: bomb early when `RELEASE` is set and not supported
- `bcm2711`/`edge`: bump to `6.3` (`rpi4b`, RPi Foundation Kernel)
- `rockchip-rk3588`/`edge`: bump to `6.3`, drop 6.2 patches
- `rk3568-odroid`/`edge`: bump to `6.3` (`odroidm1`)
- `virtual-qemu`: move to `eos` (replaced by `qemu-uboot-x86` and `qemu-uboot-arm64`)
- `qemu-uboot-arm64`/`qemu-uboot-x86`: split more [... cut]
commit 6417a221 (2023-04-28): sun50iw9: don't inadvertently override pre-existing hook function name - fixes #5099
closed 2023-04-28 by igorpecovnik