armbian/build #5114: Can't build `linux-u-boot-radxa-e25-legacy` / `linux-u-boot-radxa-e25-edge`

rpardini, 2023-04-29:
### What happened?

- Can't build `linux-u-boot-radxa-e25-legacy` / `linux-u-boot-radxa-e25-edge` 

### How to reproduce?

  - `FATAL ERROR: Couldn't open "./u-boot.dtb": No such file or directory` 

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
- https://paste.next.armbian.com/esedudikub
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2023-04-29:
I think it is spi related: https://github.com/armbian/build/blob/main/config/boards/radxa-e25.wip#L11
If this board doesn't have spi flash we can remove it.

rpardini, 2023-04-29:
Thanks @amazingfate indeed it has to do with the SPI target.

But there's something more crazy. I _can_ build this no problem on my workstation, SPI included, but CI (Github actions, hosted runner) is consistently failing. Both are Docker and x86. https://paste.next.armbian.com/ihabepukay

After comparing log output, turns out the only difference (thankfully) was that my workstation has 16 cores, while the GHA hosted instance has 3. And, in fact, if I manually force `-j3` in my build, it also breaks. It does not break if I use `-j1`.

So it seems the makefile target `u-boot.itb` does not correctly depend on the `u-boot.dtb` target, and works "by accident" on multicore machines, which are faster and build it fast enough in parallel. 

My suggestion is to add `u-boot.dtb` to the target map, so make's hand is forced to do the right thing.

```patch
--- a/config/sources/families/include/rockchip64_common.inc	(revision ffc5d55fb1a7336f5da1dc10543ed10f77911ec2)
+++ b/config/sources/families/include/rockchip64_common.inc	(date 1682794883037)
@@ -189,7 +189,7 @@
 	if [[ $BOOT_SUPPORT_SPI == yes ]]; then
 
 		if [[ "${BOOT_SPI_RKSPI_LOADER:-"no"}" != "yes" ]]; then
-			UBOOT_TARGET_MAP="BL31=$RKBIN_DIR/$BL31_BLOB tpl/u-boot-tpl.bin spl/u-boot-spl.bin u-boot.itb ${UBOOT_TARGET_MAP} rkspi_loader.img"
+			UBOOT_TARGET_MAP="BL31=$RKBIN_DIR/$BL31_BLOB tpl/u-boot-tpl.bin spl/u-boot-spl.bin u-boot.dtb u-boot.itb ${UBOOT_TARGET_MAP} rkspi_loader.img"
 		else
 			UBOOT_TARGET_MAP="${UBOOT_TARGET_MAP} rkspi_loader.img"
 		fi
```

amazingfate, 2023-04-30:
`u-boot.dtb` is already in ${UBOOT_TARGET_MAP}. `BOOT_SPI_RKSPI_LOADER` should be declared if rk35xx has to build spi image. I don't know much about other family like 3399.

rpardini, 2023-04-30:
> `u-boot.dtb` is already in ${UBOOT_TARGET_MAP}.

It is, but in the "regular" (non-SPI) target map, and thus is included "later" in the command than the `u-boot.itb`.
With my patch, it's included twice, indeed, but that fixes it. I Just built some 400+ u-boot's to prove it ;-)

amazingfate, 2023-04-30:
Options included twice is not normal. The line you modified should not get executed for the uboot of rk35xx. Maybe it only works for rk3399.

rpardini, 2023-04-30:
Let me prove it works:
- E25: (3568) https://paste.next.armbian.com/udekegipac
- Rock-5b: (3588) https://paste.next.armbian.com/zekanupage
- RockPro64: (3399): https://paste.next.armbian.com/dexaxisoha
- Rock3a (3568): https://paste.next.armbian.com/egomajoruc
and many, many more.

Either way, if not by forcing make's hand with this, what would you suggest?

amazingfate, 2023-04-30:
I think we should fix the twice set make options.
When only `BOOT_SUPPORT_SPI` is set for legacy rockchip spi image, I think just
 ```
UBOOT_TARGET_MAP="${UBOOT_TARGET_MAP} tpl/u-boot-tpl.bin spl/u-boot-spl.bin u-boot.itb rkspi_loader.img
``` 
should be enough because BL31 is already set before. I have a feeling that the twice set BL31 will causes the second set targets not built.
For rk35xx `BOOT_SPI_RKSPI_LOADER` must get declared for spi image. Declaring it for radxa e25 or undeclaring `BOOT_SUPPORT_SPI` should fix this issue.

rpardini, 2023-04-30:
> should be enough because BL31 is already set before

True, but unrelated. See the evidence I posted above; the problem arises from a concurrency issue, not from the declaration order.

closed 2023-05-01 by igorpecovnik

amazingfate, 2023-05-04:
I added this patch to rockchip64_common.inc:
```
@@ -189,7 +189,7 @@ prepare_boot_configuration() {
        if [[ $BOOT_SUPPORT_SPI == yes ]]; then

                if [[ "${BOOT_SPI_RKSPI_LOADER:-"no"}" != "yes" ]]; then
-                       UBOOT_TARGET_MAP="BL31=$RKBIN_DIR/$BL31_BLOB tpl/u-boot-tpl.bin spl/u-boot-spl.bin u-boot.dtb u-boot.itb ${UBOOT_TARGET_MAP} rkspi_loader.img"
+                       UBOOT_TARGET_MAP="${UBOOT_TARGET_MAP} tpl/u-boot-tpl.bin spl/u-boot-spl.bin u-boot.itb rkspi_loader.img"
                else
                        UBOOT_TARGET_MAP="${UBOOT_TARGET_MAP} rkspi_loader.img"
                fi
```
Rockpro64 is built fine: https://paste.next.armbian.com/okogutizaw
No need to build `u-boot.dtb`

rpardini, 2023-05-04:
Thanks, this is cleaner than the b.s. I sent up in https://github.com/armbian/build/commit/84164c401210d4e7a3dfe2fb4671180a1fb2622e

commit 84164c40 (2023-05-01): rockchip64_common: add explicit make target `u-boot.dtb` before `u-boot.itb` in the target map for SPI; fixes #5114

- when `BOOT_SUPPORT_SPI=yes` (and not using the rkspi_loader)
- it seems the makefile target u-boot.itb does not correctly depend on the u-boot.dtb target, and works "by accident" on multicore machines, which are faster and build it fast enough in parallel. this adds u-boot.dtb to the target map, so make's hand is forced to do the right thing.