armbian/build #5115: Can't build `kernel-rk322x-legacy`

rpardini, 2023-04-29:
### What happened?

  - `patch/kernel/archive/rk322x-4.4/01-linux-0005-dts.patch(:13)` broken
  - `patch/kernel/archive/rk322x-4.4/01-linux-1000-pl330.patch(:31)` severely broken
  - probably more errors later

### How to reproduce?

.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
- https://paste.next.armbian.com/ucuyalerey
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-04-29:
hey @paolosabatino  what do you think, worth investing time in cleaning this up, or just removing it`*`? I fear after the basic patching problems are gone, there are gonna be more problems.
- `*` = only `BOARD=rk322x-box` uses this kernel, removing it from it's KERNEL_TARGET will make it effectively disappear, we could remove the family code/patches too.

paolosabatino, 2023-04-29:
Mmmh, perhaps it could be an idea to fix the legacy kernel to a specific commit/tag?

My intention is not to maintain it anymore in the future and switch totally to mainline. Actually it is long time I don't even build a legacy kernel for that and don't touch the patches, but some people may still need the legacy kernel because it is the only way to let onboard NAND flash memories work (some boards still have raw NANDs... ) 

I will take a look tomorrow and see if I can fix the patches, then think about a possible strategy; in the meantime if the legacy kernel is somehow making automated build difficult, it can be safely removed from the KERNEL_TARGET variable for the time being.

rpardini, 2023-04-29:
> Mmmh, perhaps it could be an idea to fix the legacy kernel to a specific commit/tag?

Yes, this is a valid strategy, in general, for unmaintained-but-still-working things, but I don't see how it helps in this case: the breakage came from "armbian-next"'s new patch parsing, not from any "upstream" change....

> if the legacy kernel is somehow making automated build difficult

It's not, we could easily remove it indeed, it's more my OCD that's triggered here. Waste no more than a few minutes on this...

paolosabatino, 2023-04-30:
> Yes, this is a valid strategy, in general, for unmaintained-but-still-working things, but I don't see how it helps in this case: the breakage came from "armbian-next"'s new patch parsing, not from any "upstream" change....

Okay, got it. I thought it was due to upstream changes in first place, but later got aware Igor already forked the 4.4-stable branch from rockchip, so it is already "frozen".

Here it is a pull request to fix the whole bunch of patch problems and also fix compilation issues with gcc-11:

https://github.com/armbian/build/pull/5121

rpardini, 2023-04-30:
Excellent, thanks @paolosabatino, confirmed fixed: https://paste.next.armbian.com/lanavokuxa

closed 2023-04-30 by rpardini