armbian/build #5117: Can't build `kernel-jetson-nano-legacy`

rpardini, 2023-04-29:
### What happened?

- a lot of `-Werror=` stuff from gcc

### How to reproduce?

.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
- https://paste.next.armbian.com/yofajukaka
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-04-30:
As you can see, I tried, and failed. In the end I could make it build, but not install, since this kernel's Makefile is so crippled it does not have a dtb install target. It heavily depends on a custom builddeb to work, and custom builddebs are no more.
It's being removed. Git history is your friend if anyone needs this for some reason.

igorpecovnik, 2023-05-01:
> It heavily depends on a custom builddeb to work

Have old way of per kernel packaging patch for those exotics targets?

But fundamental question - what kind of value this kernel represent to keep having in the build system? @150balbes

I assume current and edge (media) are in working state? Or that also needs adjustment in the packaging area?

rpardini, 2023-05-01:
I've not tested, but I think this doesn't work either in old master, unless you revert to old-gcc with toolchains etc. 

> Have old way of per kernel packaging patch for those exotics targets?

This is the _only_ exotic, and it has been _made exotic_ for no reason. We've 63 others working. It makes no sense.

closed 2023-05-01 by igorpecovnik

150balbes, 2023-05-05:
> But fundamental question - what kind of value this kernel represent to keep having in the build system? @150balbes

As I have already written, there are a number of important elements missing in the main core, so this core was used.
I see all the decisions have already been taken and used, I see no point in further discussions.

> I assume current and edge (media) are in working state?

Completely destroyed .

p.s. It's funny to read the reasoning of a character who could not understand a few lines of source code and the logic of how it works, but at the same time talks about the lack of need for this option.

igorpecovnik, 2023-05-05:
> the decisions have already been taken and used

Decisions can be reverted, especially if there is strong and real opposition. IMO this can be deal with after release. Important yes, critical no.

> Completely destroyed

Important yes, critical yes. This has to be fixed.

150balbes, 2023-05-05:
> Important yes, critical no.

not important and not critical

> Important yes, critical yes.

not important and not critical

> This has to be fixed.

Just delete the media group and all devices from it. Let the dreams of the characters who really hate and do not understand everything related to the media group come true. There are no complaints on my part.

PR #5122 (2023-05-01): more of rpardini's dirty laundry, very late April/'23
#### more of rpardini's dirty laundry, very late April/'23

> Remove legacy branches after trying, and failing, to ressurect them

- `rockchip`: completely remove `legacy` branch; fixes #5118
  - there's absolutely no point in trying to keep this alive, current/edge are good enough, legacy's abandoned for years and broken
- `jetson-nano` `legacy`: give up on this legacy and remove it; fixes #5117
- `jetson-nano` `legacy`: try, and fail, to fix this legacy; (does NOT) fixes #5117
  - `jetson-nano` `legacy`: kernel config: `CONFIG_SECTION_MIS [... cut]
commit 145ae847 (2023-05-01): `jetson-nano` `legacy`: try, and fail, to fix this legacy; (does NOT) fixes #5117

- `jetson-nano` `legacy`: kernel config: `CONFIG_SECTION_MISMATCH_WARN_ONLY=y`
- `jetson-nano` `legacy`: add `-mno-outline-atomics` to fix linking
- `jetson-nano` `legacy`: add `xxd` host-dependency
- `jetson-nano` `legacy`: downgrade _a lot_ of gcc errors to allow this legacy to build
commit f860106a (2023-05-01): `jetson-nano` `legacy`: give up on this legacy and remove it; fixes #5117