armbian/build #5118: Can't build `kernel-rockchip-legacy`

rpardini, 2023-04-29:
### What happened?

- lots of _assembly code_ errors, like `selected processor does not support `isb ' in ARM mode`

### How to reproduce?

.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
- https://paste.next.armbian.com/isesarelel
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-04-30:
After talking with a few key people, we've decided to just remove this branch. There's no sense in spending any time with it.

rpardini, 2023-04-30:
Sorry I forgot to mention @paolosabatino -- @Tonymac32 guessed you might be interested in this?
Could we simply remove, as suggested in the commit above?

paolosabatino, 2023-04-30:
I see no practical reasons to keep legacy rockchip branch around, except for some people that may be using rkmpp for encoding/decoding purposes. I don't know if hantro mainline driver + ffmpeg is already able to do that.

rpardini, 2023-04-30:
I think mainline hantro can decode, but not decode.
Either way, `git` history will have this, but I've removed it, and will PR soon.
Thanks!

PR #5122 (2023-05-01): more of rpardini's dirty laundry, very late April/'23
#### more of rpardini's dirty laundry, very late April/'23

> Remove legacy branches after trying, and failing, to ressurect them

- `rockchip`: completely remove `legacy` branch; fixes #5118
  - there's absolutely no point in trying to keep this alive, current/edge are good enough, legacy's abandoned for years and broken
- `jetson-nano` `legacy`: give up on this legacy and remove it; fixes #5117
- `jetson-nano` `legacy`: try, and fail, to fix this legacy; (does NOT) fixes #5117
  - `jetson-nano` `legacy`: kernel config: `CONFIG_SECTION_MIS [... cut]
commit 30737fa6 (2023-05-01): `rockchip`: completely remove `legacy` branch; fixes #5118

- there's absolutely no point in trying to keep this alive, current/edge are good enough, legacy's abandoned for years and broken
closed 2023-05-01 by igorpecovnik