armbian/build #5146: strange gzip: stdout: No space left on device

banditopazzo, 2023-05-05:
### What happened?

Building an image for `rock-5b` the process exited  complaining about `No space left on device`.

Looking at the file system stats of the builder machine during the process seems there is a lot of free space, so it shouldn't exit with that error.

### How to reproduce?

```bash
./compile.sh BOARD=rock-5b BRANCH=edge kernel-config
```

changed some options and then:

```bash
./compile.sh BOARD=rock-5b BRANCH=edge
```

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
[🐳|🔨]   update-initramfs: Generating /boot/initrd.img-6.3.1-rockchip-rk3588
[🐳|🔨]   
[🐳|🔨]   gzip: stdout: No space left on device
[🐳|🔨]   E: mkinitramfs failure gzip 1
[🐳|🔨]   update-initramfs: failed for /boot/initrd.img-6.3.1-rockchip-rk3588 with 1.
[🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
     chroot_custom_long_running() --> lib/functions/logging/runners.sh:140
               update_initramfs() --> lib/functions/image/initrd.sh:96
   create_image_from_sdcard_rootfs() --> lib/functions/image/rootfs-to-image.sh:74
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:68
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:37
          do_with_default_build() --> lib/functions/main/default-build.sh:43
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[🐳|💥] Cleaning up [ please wait for cleanups to finish ]
[🐳|🌿] Unmounting recursively [ SDCARD - be patient ]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hzyitc, 2023-05-06:
`update-initramfs` will output some temp files in `/tmp` which is mount as `tmpfs`.

Please make sure that you have enough RAM.

banditopazzo, 2023-05-06:
I think it's not my case, this is what I get from `df` command. Apparently `/tmp` is not a `tmpfs`. This setup comes from a standard Ubuntu installation

```console
builder@builder:~$ df -h
Filesystem                         Size  Used Avail Use% Mounted on
tmpfs                              6.3G  1.3M  6.3G   1% /run
/dev/mapper/ubuntu--vg-ubuntu--lv  413G   98G  298G  25% /
tmpfs                               32G     0   32G   0% /dev/shm
tmpfs                              5.0M     0  5.0M   0% /run/lock
/dev/sda2                          2.0G  254M  1.6G  14% /boot
/dev/sda1                          1.1G  6.1M  1.1G   1% /boot/efi
tmpfs                              6.3G  4.0K  6.3G   1% /run/user/1000
builder@builder:~$ df -h /tmp/
Filesystem                         Size  Used Avail Use% Mounted on
/dev/mapper/ubuntu--vg-ubuntu--lv  413G   98G  298G  25% /

```

hzyitc, 2023-05-06:
Not the system's `/tmp`. The build framework will mount `/tmp` for `chroot`

https://github.com/armbian/build/blob/495d35be49630c39653fb04c5650d3a629e2ac80/lib/functions/general/chroot-helpers.sh#L22

rpardini, 2023-05-06:
Yeah @banditopazzo (lol!) how much RAM is in the host/builder machine?
Indeed it uses tmpfs.
I'd like to add a check to at least warn you if you don't have enough RAM, you're not the first one hitting this.

banditopazzo, 2023-05-06:
Hi, the host machine is quite big, it has more than 60Gb of RAM:

```console
builder@builder:~$ free -h
               total        used        free      shared  buff/cache   available
Mem:            62Gi       413Mi        61Gi       1.0Mi       645Mi        61Gi
Swap:          8.0Gi          0B       8.0Gi
```
This is why I think it's a strange error

rpardini, 2023-05-07:
Yeah that's _very strange_
Please build with `DEBUG=yes SHARE_LOGS=yes` and please paste the log URL produced at the end here?

banditopazzo, 2023-05-07:
Here is my log https://paste.next.armbian.com/cehupudizu . thanks for the help

(I had to manually upload it with the provided `curl` command string , to do it automatically probably the option is `SHARE_LOG=yes`, without the `S`, as it was shown at the end of the process)

hzyitc, 2023-05-07:
Sorry, I'm wrong.

It's not enough space in `/boot` instand of `/tmp`.

The gzip outputs directly to the final file.

Here is the code of `mkinitramfs`:
```
{
if [ -s "${__TMPEARLYCPIO}" ]; then
	cat "${__TMPEARLYCPIO}" || exit 1
fi

$compress -c "${__TMPMAINCPIO}" ||
	{ echo "E: mkinitramfs failure $compress $?" >&2; exit 1; }

if [ -s "${__TMPCPIOGZ}" ]; then
	cat "${__TMPCPIOGZ}" || exit 1
fi
} >"${outfile}" || exit 1
```

hzyitc, 2023-05-07:
You can try to increase the size of `boot` by argument `BOOTSIZE`

banditopazzo, 2023-05-07:
**SOLVED:** @hzyitc increasing `BOOTSIZE` I was able to build my image.    

First I tried with `BOOTSIZE=512` and I got a different error, then increasing to `1024` it completes without errors.    

Thank you both for helping me. 

It would be good to improve the error message. I don't know if you want to keep this issue opened to track this task, otherwise you can mark it as solved. Thanks again.

hzyitc, 2023-05-07:
Fine.

Could you post the output of `du -h --max-depth=1 /boot/`. It's really strange that why it require such large space.

banditopazzo, 2023-05-08:
@hzyitc I suppose you are referring to `/boot` inside the `.img`:

```console
❯ du -h --max-depth=1 boot/
4.8M    boot/dtb-6.3.1-rockchip-rk3588
16K     boot/lost+found
764M    boot/
```

It doesn't explain well check this tree with size view:

```console
❯ erd boot --level 1 -H
         - ┌─ vmlinuz.old → vmlinuz-6.3.1-rockchip-rk3588
         - ├─ vmlinuz → vmlinuz-6.3.1-rockchip-rk3588
         - ├─ uInitrd → uInitrd-6.3.1-rockchip-rk3588
         - ├─ initrd.img → initrd.img-6.3.1-rockchip-rk3588
         - ├─ initrd.img.old → initrd.img-6.3.1-rockchip-rk3588
         - ├─ dtb → dtb-6.3.1-rockchip-rk3588
         - ├─ Image → vmlinuz-6.3.1-rockchip-rk3588
  4.00 KiB ├─ boot.cmd
  4.00 KiB ├─ armbian_first_run.txt.template
  4.00 KiB ├─ boot.scr
  4.00 KiB ├─ armbianEnv.txt
 40.00 KiB ├─ boot.bmp
252.00 KiB ├─ config-6.3.1-rockchip-rk3588
  4.75 MiB ├─ dtb-6.3.1-rockchip-rk3588
  5.08 MiB ├─ System.map-6.3.1-rockchip-rk3588
 39.80 MiB ├─ vmlinuz-6.3.1-rockchip-rk3588
356.71 MiB ├─ uInitrd-6.3.1-rockchip-rk3588
356.71 MiB ├─ initrd.img-6.3.1-rockchip-rk3588
763.35 MiB boot
``` 
 
However it's more than `700Mb`, it explains both errors with the default `BOOTSIZE` and `BOOTSIZE=512`

rpardini, 2023-05-08:
> probably the option is `SHARE_LOG=yes`, without the `S`, as it was shown at the end of the process

Fact, I'm terribly sorry for the curve ball. (too many PARAMs=, even though I've "invented" `SHARE_LOG`, heh

)

> ```
> 356.71 MiB ├─ initrd.img-6.3.1-rockchip-rk3588
> ```

Oh wow! I don't think I ever saw an initrd this big...

hzyitc, 2023-05-08:
I think you may enable the `CONFIG_DEBUG` ?

banditopazzo, 2023-05-08:
I have enabled options for debug (and also btf info), bpf, and tracing features

rpardini, 2023-05-08:
> I have enabled options for debug (and also btf info), bpf, and tracing features

Well in this case bump the `BOOTSIZE` -- nothing much else we can do, I think

closed 2023-05-19 by igorpecovnik