armbian/build #5155: Problem building Kinetic

Ekmansoft, 2023-05-07:
### What happened?

Building Jammy works fine, but if I select Kineic for example using this

`./compile.sh BOARD=rock-5b BRANCH=legacy RELEASE=kinetic COMPRESS_OUTPUTIMAGE=img KERNEL_CONFIGURE=no BUILD_MINIMAL=no BUILD_DESKTOP=no
`
it fails in this way 

```
[🌱] Installing base system with 5 packages [ Stage 1/2 ]
[🔨]   E: No such script: /usr/share/debootstrap/scripts/kinetic
[💥] error! [ Debootstrap first stage failed kinetic   no ]

```
Now I re-fetched a new Armbian git but the problem remains.

Why? I see other people building Kinetic successfully. 

I'm building it from Debian 11.

`Linux debian11 5.10.0-22-amd64 #1 SMP Debian 5.10.178-3 (2023-04-22) x86_64 GNU/Linux`

### How to reproduce?

./compile.sh BOARD=rock-5b BRANCH=legacy RELEASE=kinetic COMPRESS_OUTPUTIMAGE=img KERNEL_CONFIGURE=no BUILD_MINIMAL=no BUILD_DESKTOP=no

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Other

### Relevant log output

```shell
[🌱] Going to build rootfs [ packages_hash: '1e7f7599eb48B22f026' cache_type: 'cli' rootfs_cache_id: '1e7f7599eb48B22f026' ]
[🌱] Getting ORAS manifest [ ORAS manifest from ghcr.io/armbian/os/rootfs-arm64-kinetic-cli:202305-1e7f7599eb48B22f026 ]
Error: GET "https://ghcr.io/v2/armbian/os/rootfs-arm64-kinetic-cli/manifests/202305-1e7f7599eb48B22f026": GET "https://ghcr.io/token?scope=repository%3Aarmbian%2Fos%2Frootfs-arm64-kinetic-cli%3Apull&service=ghcr.io": response status code 403: denied: requested access to the resource is denied
[🌱] Artifact is not available in remote cache [ ghcr.io/armbian/os/rootfs-arm64-kinetic-cli:202305-1e7f7599eb48B22f026 ]
[🔨]   mkdir: created directory '/home/kenneth/armbian/build/.tmp/rootfs-fa01d212-1b34-4092-8e50-756456faba50'
[🔨]   mkdir: created directory '/home/kenneth/armbian/build/.tmp/mount-fa01d212-1b34-4092-8e50-756456faba50'
[🔨]   mkdir: created directory '/home/kenneth/armbian/build/output/images'
[🌱] Using tmpfs for rootfs build [ RAM available: 8329MiB > 2000MiB estimated ]
[🌱] Creating new rootfs cache [ 'rootfs-arm64-kinetic-cli_202305-1e7f7599eb48B22f026.tar.zst' ]
[🌱] Installing base system with 5 packages [ Stage 1/2 ]
[🔨]   E: No such script: /usr/share/debootstrap/scripts/kinetic
[💥] error! [ Debootstrap first stage failed kinetic   no ]
[💥] Exiting with error 43 [ at /home/kenneth/armbian/build/lib/functions/logging/traps.sh:1
                exit_with_error() --> lib/functions/logging/traps.sh:1
   create_new_rootfs_cache_via_debootstrap() --> lib/functions/rootfs/rootfs-create.sh:100
        create_new_rootfs_cache() --> lib/functions/rootfs/create-cache.sh:61
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
   artifact_rootfs_build_from_sources() --> lib/functions/artifacts/artifact-rootfs.sh:82
    artifact_build_from_sources() --> lib/functions/artifacts/artifacts-obtain.sh:34
       obtain_complete_artifact() --> lib/functions/artifacts/artifacts-obtain.sh:231
       build_artifact_for_image() --> lib/functions/artifacts/artifacts-obtain.sh:283
   get_or_create_rootfs_cache_chroot_sdcard() --> lib/functions/rootfs/create-cache.sh:75
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:14
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:37
          do_with_default_build() --> lib/functions/main/default-build.sh:43
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
[🌿] Unmounting recursively [ SDCARD - be patient ]
[🌿] Unmounting recursively [ MOUNT - be patient ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-fa01d212-1b34-4092-8e50-756456faba50.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-build-fa01d212-1b34-4092-8e50-756456faba50.log.ans https://paste.next.armbian.com/log ]
kenneth@debian11:~/armbian/build$ uname -a
Linux debian11 5.10.0-22-amd64 #1 SMP Debian 5.10.178-3 (2023-04-22) x86_64 GNU/Linux
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hzyitc, 2023-05-07:
Please use `Ubuntu Jammy 22.04` or `Docker` as our requirements. https://github.com/armbian/build#basic-requirements

The version of `debootstrap` in your OS is too low.

closed 2023-05-07 by hzyitc

rpardini, 2023-05-08:
Yeah, we're using the `debootstrap` from the host.
If you're building on Debian 11 (bullseye), you'll have trouble building more-recent things, since bullseye's debootstrap does not even know they exist.
As suggested by @hzyitc, try with Docker, this will use a Jammy base image, which (mostly) knows how to build everything.
For the future, we might consider a different debootstrap implementation, but for now there's not much we can do.

Ekmansoft, 2023-05-08:
Ok, thanks. I found an unused hard drive on my PC and installed Jazzy there .. ;-) Works better now.

EvilOlaf, 2023-05-08:
Thanks for verifying that Jammy works.