armbian/build #5160: Radxa Zero "minimal" build has no networking

marcone, 2023-05-08:
### What happened?

(filing under "build framework" because creating an image without any networking support seems like a build problem)

The Radxa Zero Jammy Minimal image doesn't have any network configuration tools installed, and in fact doesn't even include `curl`. This leads to errors during initial setup:

[image]

and without any tools to configure the network interfaces, it's really hard to get to a usable state.

By contrast the Rock Pi 4C+ Jammy Minimal image does include nmtui and other networking tools, and it would make sense for "minimal" to mean the same across different boards.

### How to reproduce?

Flash Armbian  Jammy Minimal CLI image from https://www.armbian.com/radxa-zero/
Go through initial setup

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-05-08:
We fixed this problem but some images at the download pages could still be broken. Images will be updated to new release soon.

I just made image for Odroid N2 (i don't have this particular board) and minimal images works as expected. Build system wise we don't have this bug anymore, so I propose we close this.

rpardini, 2023-05-08:
Yeah I've seen multiple commits adding stuff to minimal so this kind of stuff is probably fixed in recent builds yeah.

igorpecovnik, 2023-05-19:
I think this was fixed, closing.

closed 2023-05-19 by igorpecovnik