armbian/build #5170: Reporting wrong error

igorpecovnik, 2023-05-13:
### What happened?

Cosmetic issues. When trying to flash to SD card directly it tells that Docker is the reason, while the reason is device was not there.

[🐳|🚸] Can't write to /dev/sda [ Under Docker ]

### How to reproduce?

# generating any image
./compile.sh CARD_DEVICE=/dev/sda ARTIFACT_IGNORE_CACHE='no' 

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-05-21:
What is the suggested wording here?
The meaning is just to help user understand that there _might_ be difficulties writing due to Docker.

igorpecovnik, 2023-05-21:
Huh, let me think.

The question is - does this difficulties can be fixed? For example - when I am doing images for myself on my x64 linux desktop, i want to burn images directly. Its pretty handy feature. Now, when docker fires up ... feature is gone. I assume we need a privilege mode to be able to do this? If feature for recording to SD card is uses, I would error it out at start in case docker is used ... or warn that privilege mode is going to be enabled.

rpardini, 2023-05-21:
No. We're already using privileged mode, and I think it mostly works, at least on the cases where Docker client and Docker daemon are running on the same host.
For machines (eg MacOS, Windows) where there is a VM, and thus Docker client and Docker daemon are running in separate hosts, passing through devices simply is not possible, and never will be.
In my view there's nothing to fix or fixable here.

igorpecovnik, 2023-05-21:
> For machines (eg MacOS, Windows) where there is a VM, and thus Docker client and Docker daemon are running in separate hosts

Where I am trying to do this is Armbian x86 on bare metal.

rpardini, 2023-05-21:
That should just work, as long as your `/dev/sda` actually exists and is ready to go.

igorpecovnik, 2023-05-21:
Is it possible to differentiate Can't write to /dev/sda [ Device is missing ]  and Can't write to /dev/sda [ Under Docker ] ? If not just mentioning both?

rpardini, 2023-05-21:
It could be. Docker and devices (incl the `/dev/loopX` stuff) is tricky.
We could test the device in the Docker launcher (docker.sh) before passing it down to Docker.
It would be interesting to see how to have a good experience using `dockershell` too, maybe passing down all devices like the loop hack already does.
Also the mess with `/dev/disk/by-id/` already showing up with grub and others is similarly related, but involves udev somehow.
I think we should handle all these things together in a future epic/saga for coordination

igorpecovnik, 2023-05-21:
OK, lets close this one and proceed with https://armbian.atlassian.net/browse/AR-1740 Can be expanded later if needed. Just to have a record for now.

closed 2023-05-21 by igorpecovnik