armbian/build #5171: Raspberry Pi does not boot

igorpecovnik, 2023-05-13:
### What happened?

Can't find rootfs and hangs.

```
[🐳|🔨]   update-initramfs: Generating /boot/initrd.img-6.1.28-bcm2711
[🐳|🔨]   UUID ddaa9e6d-c9d4-4f43-99c3-e1d231c48010 doesn't exist in /dev/disk/by-uuid
[🐳|🔨]   Warning: root device /dev/disk/by-uuid/ddaa9e6d-c9d4-4f43-99c3-e1d231c48010 does not exist
[🐳|🔨]   W: Couldn't identify type of root file system for fsck hook
```

### How to reproduce?

Build CLI Jammy image and try to boot on Rpi4.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/tivovejono

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-05-14:
Problem related to kernel. (`BRANCH=ddk` or `BRANCH=legacy` (5.15) boots and works no problem.)

I tried booting `edge` and not even the serial console works.

Bumped verbosity and got at least those (sorry for char errors, had to take photo of HDMI).

```
Beg In: Mounting root file system... Begin: Running /scripts/local-top... done. Breg In: Running /scripts/local-premount... [ 13.7983141 platform fe101000.cprman: di
13.7984401 amba fe201000.serial: deferred probe pending
13.7985121 platform cam1 regulator: deferred probe pending 
13.798586] platform regulator-sd-vcc: deferred probe pending
13.798660] platform regulator-sd-10-1v8: deferred probe pending 
13.798733) platform fe215000.aux: deferred probe pending
13.798801] platform fe100000.watchdog: deferred probe pending
13.798875] platform leds: deferred probe pending
[...]LABEL=armbi_root does not exist. Dropping to a shell!
```

So regulators and much more missing, wth? no wonder it does not boot.

We've (yes me included) been a bit blindly bumping `current` and `edge`, but I think changes in the Kconfig structure in RpiFoundation's kernels require more work, possibly a `.config` rework.

> [🐳|🔨]   UUID ddaa9e6d-c9d4-4f43-99c3-e1d231c48010 doesn't exist in /dev/disk/by-uuid

My guess this is unrelated. We've the same with GRUB (but that caused an error, so I had to hack /dev/disk/by-uuid). It's related to Docker & `udev` stuff in the chroot. It's highly dependent on docker version, -ce vs distro's, host kernel, etc. Needs further investigation but should be safe here.  better to  invest time towards rpi4b u-boot.

rpardini, 2023-05-14:
> possibly a `.config` rework.

It seems you done some work on https://github.com/armbian/build/pull/4969 already?

igorpecovnik, 2023-05-14:
> It seems you done some work on https://github.com/armbian/build/pull/4969 already?

OK. Working on config ...