armbian/build #5174: Audio not working anymore for PINE64

PanderMusubi, 2023-05-14:
### What happened?

Audio out and audio in not working anymore for PINE64

### How to reproduce?

Connect audio device in the audio jack and run command line `speaker-test`. When running in desktop environment, the audio levels move but no sound is going out of the audio port.

Detailed report and solution is found in https://forum.armbian.com/topic/15848-pine64-a64-kernel-59x-analog-sound-problem/ See also https://github.com/MichaIng/DietPi/issues/4805

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.armbian.com/iwolololec

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PanderMusubi, 2023-05-14:
Related might be that GPIO isn't working. Had a problem with that too like one of the other reporters at DietPi, but not sure how to investigate/analyse it, other than that I could no longer make readings of temperature and humidty sensor over GPIO, which also used to work in the past. Also don't know if this warrents another issue.

EvilOlaf, 2023-05-15:
This issue tracker is for issues **building** Armbian. For **running** there are community forums.

closed 2023-05-15 by EvilOlaf