armbian/build #5185: Docker can't create folder that existed before

igorpecovnik, 2023-05-16:
### What happened?

```
Section 'docker_cli_prepare' took 0s to execute.
[🥑] Group docker_cli_prepare_dockerfile
[🥑] Group docker_cli_build_dockerfile
Section 'docker_cli_build_dockerfile' took 1s to execute.
[🥑] Group docker_cli_prepare_launch
  [🌱] Passing down to Docker [ GITHUB_OUTPUT: '/home/actions-runner-04/_work/_temp/_runner_file_commands/set_output_3c6453e1-18dc-4339-8870-5e973a3f3f18' ]
  [🌱] Passing down to Docker [ GITHUB_STEP_SUMMARY: '/home/actions-runner-04/_work/_temp/_runner_file_commands/step_summary_3c6453e1-18dc-4339-88[70](https://github.com/armbian/os/actions/runs/4995863064/jobs/8948465633#step:8:75)-5e973a3f3f18' ]
  mkdir: cannot create directory ‘/home/actions-runner-04/_work/os/os/output/debs’: File exists
  [💥] Error 1 occurred in main shell [ at /home/actions-runner-04/_work/os/os/lib/functions/host/docker.sh:489
```

### How to reproduce?

On some runners with preexisting build environment, pattern undetermined.

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log URL

https://paste.next.armbian.com/xokazomafu

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-05-21:
Seems like `/home/actions-runner-04/_work/os/os/output/debs` exists and **is not a directory**, instead, a file.
How did that happen?
We're already using `mkdir -p` there.

igorpecovnik, 2023-05-21:
> How did that happen?

No idea, I hope you had one :) Lets keep this open for awhile to see if this resurfaced again.

closed 2023-11-15 by igorpecovnik