armbian/build #5202: Cannot build with ./compile.sh docker with main git

prahal, 2023-05-21:
### What happened?

While building with the  latest main 8eeb9379d I cannot build with docker without it aborting this way (even from a fresh checkout):
```
./compile.sh docker BOARD=helios64 BRANCH=edge BUILD_ONLY="u-boot" RELEASE=bullseye KERNEL_CONFIGURE=no
[🌿] Applying cmdline param [ 'BRANCH': '(unset)' --> 'edge' early ]
[🌿] Applying cmdline param [ 'BOARD': '(unset)' --> 'helios64' early ]
[🌿] Applying cmdline param [ 'BUILD_ONLY': '(unset)' --> 'u-boot' early ]
[🌿] Applying cmdline param [ 'RELEASE': '(unset)' --> 'bullseye' early ]
[🌿] Applying cmdline param [ 'KERNEL_CONFIGURE': '(unset)' --> 'no' early ]
[🌱] Using prebuilt Armbian image as base for 'ubuntu-jammy' [ DOCKER_ARMBIAN_BASE_IMAGE: ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
[🌿] Docker info [ Docker 20.10.24+dfsg1 Kernel:6.1.0-7-amd64 RAM:7.527GiB CPUs:4 OS:'Debian GNU/Linux 11 (bullseye)' hostname 'cyclope' under 'Linux' - buildx:no - loop-hacks:yes static-loops:no ]
[🌱] Creating [ .dockerignore ]
[🌱] Docker launcher [ enabling all extensions looking for Docker dependencies ]
[🌱] Extension manager [ processed 20 Extension Methods calls and 63 Extension Method implementations ]
[🌱] Preparing rkdevflash host-side dependencies [ rkdevflash ]
[🌱] Creating [ Dockerfile; FROM ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
[🌱] Armbian docker image [ already exists: ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
[🌱] Building [ Dockerfile via 'build' ]
[🔨]   Sending build context to Docker daemon  1.567MB
[🔨]   Step 1/7 : FROM ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest
[🔨]    ---> d5724dc15e8c
[🔨]   Step 2/7 : RUN echo "--> CACHE MISS IN DOCKERFILE: apt packages." &&  DEBIAN_FRONTEND=noninteractive apt-get -y update &&  DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends bash git psmisc uuid-runtime bc binfmt-support bison libc6-dev make dpkg-dev gcc ca-certificates ccache cpio debootstrap device-tree-compiler dialog dirmngr dosfstools dwarves fakeroot flex gawk gnupg gpg imagemagick jq kmod libbison-dev libelf-dev libfdt-dev libfile-fcntllock-perl libmpc-dev libfl-dev liblz4-tool libncurses-dev libssl-dev libusb-1.0-0-dev linux-base locales lsof ncurses-base ncurses-term ntpdate patchutils pkg-config pv qemu-user-static rsync swig u-boot-tools udev uuid-dev zlib1g-dev file tree expect colorized-logs unzip zip pigz xz-utils pbzip2 lzop zstd parted gdisk fdisk aria2 curl wget axel parallel python3-dev python3-distutils python3-setuptools python3-pip python2 python2-dev gcc-x86-64-linux-gnu gcc-aarch64-linux-gnu gcc-arm-linux-gnueabihf gcc-arm-linux-gnueabi gcc-riscv64-linux-gnu debian-ports-archive-keyring libc6-amd64-cross g++-aarch64-linux-gnu g++ btrfs-progs cryptsetup f2fs-tools nilfs-tools xfsprogs zerofree qemu-utils qemu-utils libudev-dev libusb-1.0-0-dev dh-autoreconf build-essential gcc-arm-linux-gnueabi
[🔨]    ---> Using cache
[🔨]    ---> d4c9dc0fdd83
[🔨]   Step 3/7 : RUN sed -i 's/# en_US.UTF-8/en_US.UTF-8/' /etc/locale.gen
[🔨]    ---> Using cache
[🔨]    ---> 54775726e809
[🔨]   Step 4/7 : RUN locale-gen
[🔨]    ---> Using cache
[🔨]    ---> d84510c98382
[🔨]   Step 5/7 : WORKDIR /armbian
[🔨]    ---> Using cache
[🔨]    ---> 460ce8478907
[🔨]   Step 6/7 : ENV ARMBIAN_RUNNING_IN_CONTAINER=yes
[🔨]    ---> Using cache
[🔨]    ---> c500888b90fd
[🔨]   Step 7/7 : ADD . /armbian/
[🔨]    ---> 1a35fc0f8a17
[🔨]   Successfully built 1a35fc0f8a17
[🔨]   Successfully tagged armbian.local.only/armbian-build:initial
[🌱] -----------------Relaunching in Docker after 4s------------------ [ here comes the 🐳 ]
[🐳|🌿] Applying cmdline param [ 'ARMBIAN_BUILD_UUID': '(unset)' --> 'eabc7dfd-715d-4f68-9514-c407a4f0b95c' early ]
[🐳|🌱] Skip     cmdline param [ 'ARMBIAN_RELAUNCHED': already set to 'yes' early ]
[🐳|🌿] Applying cmdline param [ 'BRANCH': '(unset)' --> 'edge' early ]
[🐳|🌿] Applying cmdline param [ 'BOARD': '(unset)' --> 'helios64' early ]
[🐳|🌿] Applying cmdline param [ 'SKIP_LOG_ARCHIVE': '(unset)' --> 'yes' early ]
[🐳|🌿] Applying cmdline param [ 'SET_OWNER_TO_UID': '(unset)' --> '1000' early ]
[🐳|🌿] Applying cmdline param [ 'BUILD_ONLY': '(unset)' --> 'u-boot' early ]
[🐳|🌿] Applying cmdline param [ 'RELEASE': '(unset)' --> 'bullseye' early ]
[🐳|🌿] Applying cmdline param [ 'KERNEL_CONFIGURE': '(unset)' --> 'no' early ]
[🐳|💥] error! [ asking for docker... inside docker. how did this happen? Tip: you don't need 'docker' to run armbian-next inside Docker; it's automatically detected and used when appropriate.  ]
[🐳|💥] Exiting with error 43 [ at /armbian/lib/functions/logging/traps.sh:1
                exit_with_error() --> lib/functions/logging/traps.sh:1
             cli_docker_pre_run() --> lib/functions/cli/cli-docker.sh:28
    armbian_cli_pre_run_command() --> lib/functions/cli/utils-cli.sh:122
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:79
                           main() --> compile.sh:50
 ]
[💥] -------------Docker run failed after 6s-------------------------- [ 🐳 failed ]
[💥] Docker Log file for this run [ not found ]
[🌱] Cleaning up [ please wait for cleanups to finish ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-docker-eabc7dfd-715d-4f68-9514-c407a4f0b95c.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-docker-eabc7dfd-715d-4f68-9514-c407a4f0b95c.log.ans https://paste.next.armbian.com/log ]
```

### How to reproduce?

`./compile.sh docker BOARD=helios64 BRANCH=edge BUILD_ONLY="u-boot" RELEASE=bullseye KERNEL_CONFIGURE=no`

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Relevant log URL

https://paste.next.armbian.com/qobadodovu

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prahal, 2023-05-21:
Note that this issue is minimal as per the log provided tip removing docker from the command still runs docker behind the scene and works.
That is `./compile.sh BOARD=helios64 BRANCH=edge BUILD_ONLY="u-boot" RELEASE=bullseye KERNEL_CONFIGURE=no BUILD_MINIMAL=yes` is fine. As is `./compile.sh BOARD=helios64 BRANCH=edge  RELEASE=bullseye uboot`.

EvilOlaf, 2023-05-21:
So basically not an issue. 
FWIW perhaps this error can be downgraded to a warning and simply ignored.

igorpecovnik, 2023-05-21:
Documentation regarding docker is wrong. Probably whole section can be removed.

rpardini, 2023-05-21:
The error tries hard to explain what is going on... is it being ignored by user, or needs rewording?
```
[🐳|💥] error! [ asking for docker... inside docker. how did this happen? Tip: you don't need 'docker' to run armbian-next inside Docker; it's automatically detected and used when appropriate.  ]
```

closed 2023-05-21 by rpardini

prahal, 2023-05-22:
Maybe updating the wiki? But maybe it waits for armbian-next to be published?

As I was told that I do not need docker, when I saw the message I thought I should not use docker anymore not that I should remove docker from the compile.sh command line.

With, the wiki https://docs.armbian.com/Developer-Guide_Building-with-Docker/ telling to have docker in the compile.sh it was even more difficult to understand what was to be done, ie remove docker from compile.sh command line.

The message could be reworded to explicitly state to not pass 'docker' to the compile.sh cli.
I admit that on a later read of the message, focusing on `it's automatically detected and used when appropriate.  ]
` lead me to try to not pass docker in the command.

The issue is with `you don't need 'docker' to run armbian-next inside Docker;`.
Maybe `do not pass 'docker' to compile.sh to run armbian-next inside Docker;`.

Also because 'you don't need' means that we should be able to do it even if optional. While in fact, it does not work at all if we pass 'docker' to compile.sh.

rpardini, 2023-05-23:
Thanks @prahal -- I think here we boil down to "docs need updating".

praul, 2023-08-08:
... when omitting "docker" from build command, build won't start on my machine because of unsupported build environment.
It works with "docker-shell" and then running build command inside the container.

[https://docs.armbian.com/Developer-Guide_Building-with-Docker/  -> ](https://docs.armbian.com/Developer-Guide_Building-with-Docker/#docker-shell-interactive-mode)

prahal, 2023-10-13:
@praul are you on armbian-next (the current main branch or such)?
Because with armbian-next, docker-shell and building inside are broken https://github.com/armbian/build/issues/5203, or at least it was a few months ago (I did not check recently). This is because the compile.sh script refuses to run as root.

With armbian-next you have to pass an argument to build only u-boot, the kernel, patch the kernel ... ie ./compiles.sh uboot <..>; ./compile.sh kernel-patch <...>; ./compile.sh kernel <...>;

The documentation on the site is for the obsolete armbian master branch (not updated since February 2023).

0wulf, 2024-04-08:
As today, this error is still present. I am trying to build an armbian image in a non debian distribution and failing due to the same error.
I am pretty sure its a bug or mismaintain.
As one tries to run the dockerized version, it fails with the same goofy "runing docker inside docker? how did this happen?" error.

igorpecovnik, 2024-04-08:
> As today, this error is still present.

Command parameters has change since. Try this - it works as expected:

`./compile.sh uboot BOARD=helios64 BRANCH=edge KERNEL_CONFIGURE=no`

Logs: https://paste.armbian.com/yuzimawoxo

> non debian distribution

This tool is here to make a distro for SBC. Tool works and is well auto-tested with Ubuntu Jammy. Elsewhere? _It might, no warranty, no support._ Supporting all distributions out there is simply impossible, far from primary mission.

> I am pretty sure its a bug or mismaintain.

It is, but its yours as you are running not supported build host. But if you find and fix this problem on the way that it won't break primary build host distribution, we will merge it.

> This is also true for the stabe v23.11 version.

Why would that be different in supporting distribution X?

> documentation suggests 

It also suggest to run this virtualised in case you have concerns. In order to generate images via /dev/loop devices, we need privileged Docker access. We tried many things and failed, but if you know what we don't, improvements are always welcome.

0wulf, 2024-04-09:
@igorpecovnik 

Thanks for the response.
I was not aware of the new uboot parameter, where can i find more information about it? i will probably need to read the scripts.

As for the docker param, it doesnt work on my machine, it seems to be a bug related to `systemd-nspawn` containerization. 

My solution was indeed virtualization. Fully working now. I would recommend virtualizing if any compatibility problem @ the host system.

igorpecovnik, 2024-04-09:
> where can i find more information about it?

Sadly, only in the scripts / code. 

> how is the process on contrib @ documentation level and i gladly send some pullreqs

[We are trying to make it](https://forum.armbian.com/events/event/47-armbian-documentation-follow-up/) ... join if you can.

This also:
https://github.com/armbian/build/pull/6389