armbian/build #5203: Cannot build inside docker-shell -- cannot build as root but container is run as root

prahal, 2023-05-21:
### What happened?

I get an error if I try to run `./compile.sh BOARD=helios64 BRANCH=edge BUILD_ONLY="u-boot" RELEASE=bullseye KERNEL_CONFIGURE=no` inside the sehll given by `./compile.sh docker-shell`:
```
./compile.sh docker-shell
[🌱] Using prebuilt Armbian image as base for 'ubuntu-jammy' [ DOCKER_ARMBIAN_BASE_IMAGE: ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
[🌿] Docker info [ Docker 20.10.24+dfsg1 Kernel:6.1.0-7-amd64 RAM:7.527GiB CPUs:4 OS:'Debian GNU/Linux 11 (bullseye)' hostname 'cyclope' under 'Linux' - buildx:no - loop-hacks:yes static-loops:no ]
[🌱] Creating [ .dockerignore ]
[🌱] Docker launcher [ enabling all extensions looking for Docker dependencies ]
[🌱] Extension manager [ processed 20 Extension Methods calls and 64 Extension Method implementations ]
[🌱] Preparing rkdevflash host-side dependencies [ rkdevflash ]
[🌱] Creating [ Dockerfile; FROM ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
[🌱] Armbian docker image [ already exists: ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest ]
[🌱] Building [ Dockerfile via 'build' ]
[🔨]   Sending build context to Docker daemon  1.567MB
[🔨]   Step 1/7 : FROM ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest
[🔨]    ---> d5724dc15e8c
[🔨]   Step 2/7 : RUN echo "--> CACHE MISS IN DOCKERFILE: apt packages." &&  DEBIAN_FRONTEND=noninteractive apt-get -y update &&  DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends bash git psmisc uuid-runtime bc binfmt-support bison libc6-dev make dpkg-dev gcc ca-certificates ccache cpio debootstrap device-tree-compiler dialog dirmngr dosfstools dwarves fakeroot flex gawk gnupg gpg imagemagick jq kmod libbison-dev libelf-dev libfdt-dev libfile-fcntllock-perl libmpc-dev libfl-dev liblz4-tool libncurses-dev libssl-dev libusb-1.0-0-dev linux-base locales lsof ncurses-base ncurses-term ntpdate patchutils pkg-config pv qemu-user-static rsync swig u-boot-tools udev uuid-dev zlib1g-dev file tree expect colorized-logs unzip zip pigz xz-utils pbzip2 lzop zstd parted gdisk fdisk aria2 curl wget axel parallel python3-dev python3-distutils python3-setuptools python3-pip python2 python2-dev gcc-x86-64-linux-gnu gcc-aarch64-linux-gnu gcc-arm-linux-gnueabihf gcc-arm-linux-gnueabi gcc-riscv64-linux-gnu debian-ports-archive-keyring libc6-amd64-cross g++-aarch64-linux-gnu g++ btrfs-progs cryptsetup f2fs-tools nilfs-tools xfsprogs zerofree qemu-utils qemu-utils libudev-dev libusb-1.0-0-dev dh-autoreconf build-essential openssh-client gcc-arm-linux-gnueabi
[🔨]    ---> Using cache
[🔨]    ---> cf3763557b39
[🔨]   Step 3/7 : RUN sed -i 's/# en_US.UTF-8/en_US.UTF-8/' /etc/locale.gen
[🔨]    ---> Using cache
[🔨]    ---> 18deb6f499b8
[🔨]   Step 4/7 : RUN locale-gen
[🔨]    ---> Using cache
[🔨]    ---> 735d5b1695a9
[🔨]   Step 5/7 : WORKDIR /armbian
[🔨]    ---> Using cache
[🔨]    ---> 4936f256c7c0
[🔨]   Step 6/7 : ENV ARMBIAN_RUNNING_IN_CONTAINER=yes
[🔨]    ---> Using cache
[🔨]    ---> 9d26e4de9503
[🔨]   Step 7/7 : ADD . /armbian/
[🔨]    ---> Using cache
[🔨]    ---> 1a900e484c16
[🔨]   Successfully built 1a900e484c16
[🔨]   Successfully tagged armbian.local.only/armbian-build:initial
[🌱] Socket /run/user/1000/keyring/ssh [ SSH agent forwarding into Docker ]
[🌱] Launching Docker shell [ docker-shell ]
root@b6e7c22b9caf:/armbian# ./compile.sh BOARD=helios64 BRANCH=edge BUILD_ONLY="u-boot" RELEASE=bullseye KERNEL_CONFIGURE=no
[🐳|🌿] Applying cmdline param [ 'BRANCH': '(unset)' --> 'edge' early ]
[🐳|🌿] Applying cmdline param [ 'BOARD': '(unset)' --> 'helios64' early ]
[🐳|🌿] Applying cmdline param [ 'BUILD_ONLY': '(unset)' --> 'u-boot' early ]
[🐳|🌿] Applying cmdline param [ 'RELEASE': '(unset)' --> 'bullseye' early ]
[🐳|🌿] Applying cmdline param [ 'KERNEL_CONFIGURE': '(unset)' --> 'no' early ]
[🐳|💥] PROBLEM: don't run ./compile.sh as root or with sudo [ PROBLEM: don't run ./compile.sh as root or with sudo ]
[🐳|💥] Problem detected [ directly called as root ]
[🐳|💥] Exiting in 10 seconds [ Press <Ctrl-C> to abort, <Enter> to ignore and continue ]
Counting down: 9... 8... 7... 6... 5... 4... 3... 2... 1... 0... 
[🐳|💥] error! [ Exiting due to 'directly called as root'  ]
[🐳|💥] Exiting with error 43 [ at /armbian/lib/functions/logging/traps.sh:1
                exit_with_error() --> lib/functions/logging/traps.sh:1
   exit_if_countdown_not_aborted() --> lib/functions/general/countdown.sh:41
   cli_standard_relaunch_docker_or_sudo() --> lib/functions/cli/utils-cli.sh:230
     cli_standard_build_pre_run() --> lib/functions/cli/cli-build.sh:14
    armbian_cli_pre_run_command() --> lib/functions/cli/utils-cli.sh:122
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:79
                           main() --> ./compile.sh:50
 ]
root@b6e7c22b9caf:/armbian# exit
exit
[💥] Error 43 occurred in main shell [ at /home/prahal/Projects/WIP/helios64/armbian/test/lib/functions/cli/cli-docker.sh:61
                 cli_docker_run() --> lib/functions/cli/cli-docker.sh:61
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> ./compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
[🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-docker-shell-d1726ac6-8bfa-4d1e-80e9-9d87ee1294d2.log.ans ]
[🌿] Share log manually (or SHARE_LOG=yes): [ curl --data-binary @output/logs/log-docker-shell-d1726ac6-8bfa-4d1e-80e9-9d87ee1294d2.log.ans https://paste.next.armbian.com/log ]

```

### How to reproduce?

`./compile.sh docker-shell`
then inside the root docker shell:
`./compile.sh BOARD=helios64 BRANCH=edge BUILD_ONLY="u-boot" RELEASE=bullseye KERNEL_CONFIGURE=no`

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Relevant log URL

https://paste.next.armbian.com/ojixehuwuj

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rpardini, 2023-05-21:
Confirmed. `dockershell` needs rework. For now, use `ALLOW_ROOT=yes` or just press a key.

paulopacitti, 2023-10-01:
Same here, I can't start the build running from a Docker container since it's root based. I also tried the docker image provided to build images in GitHub actions, but it's not working neither

prahal, 2023-10-01:
@paulopacitti if you want to build from a docker container just omit the "docker" from the compile.sh command (the docker build is run by default in the new build system).

If you want to have a shell inside the docker build container from where to start the compile.sh command, then this issue still stands.

prahal, 2024-02-01:
Bug fixed at least with master code, likely by commit 7b81255a5f786202aaf2d983e88a710e9a0028e1 "Fix compilation within docker-shell" by Gunjan Gupta from the 1st of February of 2024.

I can `./compile.sh docker-shell` then inside `./compile.sh uboot BOARD=helios64 BRANCH=edge  RELEASE=bookworm ARTIFACT_IGNORE_CACHE=yes`.

There is a likely minor issue left that is when I exit the docker-shell I get:
```
root@a3831ca17d6e:/armbian# exit
exit
[💥] Error 130 occurred in main shell [ at /home/prahal/Projects/WIP/helios64/armbian/build/lib/functions/cli/cli-docker.sh:77
                 cli_docker_run() --> lib/functions/cli/cli-docker.sh:77
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> ./compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
```

closed 2024-02-01 by prahal

viraniac, 2024-02-03:
> There is a likely minor issue left that is when I exit the docker-shell I get:

I have seen that behavior as well. But it seems it exits with the exit code of last command. If the last executed command exited with exitcode 0, then docker shell is also exitting without error

prahal, 2024-02-09:
@viraniac confirmed this error was due to the exit code of the last command.
Exiting the docker-shell with `exit 0` avoids this error message even if the last command exit code was not zero (ie if I interrupted the ./compile.sh build from within docker-shell before quitting it, as I did when I got the previous error message).