armbian/build #5274: Problem found ...

Road-tech, 2023-06-01:
### What happened?

I run this commond in Github Action
```./compile.sh build BOARD=nanopineo BRANCH=current BUILD_MINIMAL=no RELEASE=jammy KERNEL_CONFIGURE=no```

and I got this error 
```
2023-05-31T10:41:34.6408629Z ##[group][🥑] Group artifact_is_available_in_remote_cache
2023-05-31T10:41:34.6452628Z [0m[1;30m[🐳|[0m🌱[0m] [0mGetting ORAS manifest [[0;32m ORAS manifest from ghcr.io/armbian/os/uboot-nanopineo-current:23.08.0-trunk--2022.04-Se4b6-P6b27-H0429-Ve2df-Bf484 [0m][0m
2023-05-31T10:41:34.8819683Z Error: ghcr.io/armbian/os/uboot-nanopineo-current:23.08.0-trunk--2022.04-Se4b6-P6b27-H0429-Ve2df-Bf484: not found
2023-05-31T10:41:34.8849441Z [0m[1;30m[🐳|[0m🌱[0m] [0mArtifact is not available in remote cache [[0;32m ghcr.io/armbian/os/uboot-nanopineo-current:23.08.0-trunk--2022.04-Se4b6-P6b27-H0429-Ve2df-Bf484 [0m][0m
2023-05-31T10:41:34.8854110Z Section 'artifact_is_available_in_remote_cache' took 0s to execute.
2023-05-31T10:41:34.8859791Z ##[endgroup]
```
and more error like
```
Error: ghcr.io/armbian/os/kernel-sunxi-current:23.08.0-trunk--6.1.30-Sa343-Db165-P73a0-C096bHfe66-HK01ba-V014b-B9c18: not found
Error: ghcr.io/armbian/os/armbian-firmware:23.08.0-trunk--1-SAfece-B6ae3: not found
Error: ghcr.io/armbian/os/armbian-base-files-jammy-armhf:23.08.0-trunk--jammy-1armbian1-B9e4a: not found
Error: ghcr.io/armbian/os/fake-ubuntu-advantage-tools:23.08.0-trunk--1-Bdb49: not found
Error: ghcr.io/armbian/os/armbian-config:23.08.0-trunk--1-SAed8f-Baeda: not found
```

### How to reproduce?

Command you gave to the framework!

### Branch

main (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-06-01:
With provided command line its not possible to reproduce.  You must be using build config to override certain parameters.

igorpecovnik, 2023-06-01:
Under actions script behaves differently ... Needs to be investigated.

rpardini, 2023-06-01:
@Road-tech you're getting cache misses, meaning artifacts are not found in the cache and will be rebuilt during your build.
From the log you posted, build completed successfully.
What is the actual problem here? Or is just the "Error: gchr.io... not found" lines throwing you off?

hzyitc, 2023-06-01:
Try to `export GITHUB_ACTIONS=false` to avoid updating remote cache.

In current framework, if we run in GHA, it will try to do this. We need another option to enable it instand of just check GITHUB_ACTIONS which will be set in any GH workflow.

rpardini, 2023-06-01:
> Try to `export GITHUB_ACTIONS=false` to avoid updating remote cache.
> 
> In current framework, if we run in GHA, it will try to do this. We need another option to enable it instand of just check GITHUB_ACTIONS which will be set in any GH workflow.

This is true, but only when building a specific artifact, not a full image. His problem here seems to be the fact that the ORAS tooling writes "Error: " to log directly when it can't find a manifest (cache miss). I've seen a few more people confused by this.

hzyitc, 2023-06-01:
> His problem here seems to be the fact that the ORAS tooling writes "Error: " to log directly when it can't find a manifest (cache miss). 

Yes, I replied too quickly. lol

Road-tech, 2023-06-02:
I got it, thanks

closed 2023-06-02 by Road-tech